kriswiner/MPU9250 #8: Arduino Pro Mini

themuck, 2015-01-12:
Hallo,
i have an Arduino Pro Mini, MPU-9250 Board and Nokia 5110 Display, works thes code with the Pro Mini 5V?

kriswiner, 2015-01-12:
Yes, I have seen this problem with the Pro Mini. I think there is either a memory limitation or a timing conflict. I never really understood it. Sometimes the difference between working and not working was a single Serial.print! It's not a flaw in the sketch, but a flaw in the AVR platform. This is one of the many reasons I no longer use the Pro Mini and have moved almost exclusively to the Teensy 3.1.

themuck, 2015-01-12:
OK thanks, what i want to do is to have all 3 axis in a stable oriantation form 0-360°. But i am not realy shure how i must calculate it now.

themuck, 2015-01-15:
no idea?

themuck, 2015-01-15:
OK, now i checked all and use the https://github.com/richards-tech/RTIMULib-Arduino. 
Everything works, but when i add the 5110 Display the Values are wrong...

kriswiner, 2015-01-15:
I'm not sure what kind of problems you are having but I believe the Nokia 5110 is a 3V3-only display. I wouldn't use 5 V.

themuck, 2015-01-15:
Now i think it works...