kriswiner/MPU9250 #9: Magnetometer

pxdc, 2015-03-12:
Hi kriswiner. I have been following your work with imu. a few questions:
have you already tested mpu9250 using spi? The pass-through mode is only for I2C or SPI too? 
I'm not using the pass-through mode but my yaw drift alot, i don´t know if the data of magnetometer axis are correct. Can you help me?
thanks alot

kriswiner, 2015-03-12:
I have not used SPI and the boards I sell on Tindie are not designed for using SPI; they are hardired for I2C. You can buy MPU9250 breakout boards from embeddedmasters.com that do allow use of SPI. Yes, the pass through uses I2C andnot SPI, one reason I stay with I2C. At 400 kHz, I2C is plenty fast for most applications.

Yaw drift is typically due to incomplete sensor bias calibration, especially magnetometer. I have a simple magnetometer bias calibration function in my MP9250 sketches which should give you improvement in your yaw drift if you are not already calibrating the acce/gyro and mag. 

If you are doing bias calibration and still have one degree per second yaw drift, there might be something wrong with your sensor.

pxdc, 2015-03-13:
Thanks for fast reply and answers. i'm doing this very simple configuration for Magnetometer:
{0x0D, MPU6500_I2C_MST_CTRL},   // I2C Speed 400 kHz 	
{0x20, MPU6500_USER_CTRL},      // Enable AUX
{0x0C, MPU6500_I2C_SLV0_ADDR},  //Set the I2C slave addres of AK8963 and set for write.
{AK8963_CNTL2, MPU6500_I2C_SLV0_REG}, //I2C slave 0 register address from where to begin data transfer
{0x01, MPU6500_I2C_SLV0_DO}, // Reset AK8963
{0x81, MPU6500_I2C_SLV0_CTRL},  //Enable I2C and set 1 byte

{AK8963_CNTL1, MPU6500_I2C_SLV0_REG}, //I2C slave 0 register address from where to begin data transfer
{0x12, MPU6500_I2C_SLV0_DO}, // Register value to continuous measurement in 16bit
{0x81, MPU6500_I2C_SLV0_CTRL},  //Enable I2C and set 1 byte

when i read whoIam register i read 0x48 (sometimes not read). i read de axis data and the ASA registers and aply the factory calibration:
Mag.TrueX  = Mag.X * Mag.ASAX;
Mag.TrueY  = Mag.Y * Mag.ASAY;   // uT/LSB
Mag.TrueZ  = Mag.Z * Mag.ASAZ;

but i m not sure if the data is correct, what kind of data you have? i'm getting X=300, Y=500 to 5000 and Z -3 to 15, for example. The X axis is never zero. this is correct? i must determine if the data are correct before applying any fusion algorithm. can you help me?

kriswiner, 2015-03-13:
You're doing it this way rather than just reading the mag via bypass mode since you are using SPI, right?

Are you taking proper account of theEndianess of the magnetometer data?

How are you combining the two bytes to make a 16-bit data word?

How do you have the magnetometer configured? Continuous mode or one shot mode?

I recommend you take a look at the raw data to see if it is being converted to Gauss correctly. I expect that the mag should read something like x = 200 mG, y = 200 mG, and z = 400 mG if the sensor is just sitting on the tabletop. The x or y or z might be negative depending on how you have mounted the sensor..

pxdc, 2015-03-13:
Yes, i'm using SPI. Not the bypass mode.
I know that magnetemoter data are little endian and make a 16-bit data word:
Mag.X  = Byte16(s16, ReadBuf2[1], ReadBuf2[0]); // Mag.X
Mag.Y  = Byte16(s16, ReadBuf2[3], ReadBuf2[2]); // Mag.Y
Mag.Z  = Byte16(s16, ReadBuf2[5], ReadBuf2[4]); // Mag.Z

I configured the magnetometer like this:
1º write 0x01 in CNTL2 to reset all registers
2º write 0x12 in CNTL1 to read in continuos mode 1 and set the output to 16-bit 

it is necessary any more configuration? 

The raw data are already being converted to gauss or need some convertion?

kriswiner, 2015-03-13:
The raw data need to be converted to Gauss.

pxdc, 2015-03-13:
the output 16 bit data word are in uT ou mG? i think is uT.

What value i must insert in Madwick mx, my and mz (uT or mG)?

pxdc, 2015-03-13:
sory, for me the mx, my and mz input in Madgwick is irrelevant because my algoritm normalize the axis (invsqr).i try with uT and mG and the result of normalization was the same. the yaw value with madgwick is very strange the variation isn't between 0 and 360 and have multiple "0".

kriswiner, 2015-03-13:
Here is how I scale to mGauss:

void getMres() {
switch (Mscale)
{
// Possible magnetometer scales (and their register bit settings) are:
// 14 bit resolution (0) and 16 bit resolution (1)
case MFS_14BITS:
mRes = 10.*4219./8190.; // Proper scale to return milliGauss
break;
case MFS_16BITS:
mRes = 10.*4219./32760.0; // Proper scale to return milliGauss
break;
}
Mx= rawdataX*mRes*ASAX. etc.

kriswiner, 2015-03-13:
You also have to calibrate for the mag/accel/gyro offset biases if you wish to get sensible fusion results from the Madgwick algorithm.

pxdc, 2015-03-14:
i know i must calibrate mag, accel and gyro but even without calibration, the Madgwick should not indicate yaw between 0 and 360? and its really necessary apply ares, gres and mres?

kriswiner, 2015-03-14:
If you only want quaternions then you do not need to scale the raw readings,
since as you know, they are all normalized in the fusion algorithm. Only if
you want to check that the values are sensible to you need to scale the raw
values into something meaningful by themselves.

The conversion from quaternions to heading results in yaw of +/- 180. You
can easily translate this to 0 to 360 if you want.

<https://github.com/kriswiner/MPU-9250/issues/9#issuecomment-80504956> .
<https://github.com/notifications/beacon/AGY1qvmNgb_-AQ7kECXEljyAfsnPt4M8ks5
n1Dq5gaJpZM4DtXB1.gif>

pxdc, 2015-03-16:
hi Kris. I am still having problems with yaw. i have already calibrate accel and gyro. how i can determine magbias, if this is really necessary? in your code isn't perceptible. My yaw does not behave linearly, never goes up +90 jumps to +150 goes to +180 and start decreasing from -180 to -120. Seems like have a blind zone and i don't know why (values between 90-150 and others). when i remove mag from madgwick the yaw becomes correct even with 1dps drift.

kriswiner, 2015-03-17:
Your mag bias needs to be corrected. You have to recenter your response
surface at the origin. Simply subtracting the mean x/y/z values measured
while moving the sensor in a figure eight is usually good enough. There
should be an example of hoe to do this in the GutHub repository.

If you don't at least recenter your mag bias, your sensor fusion will be
garbage in many cases. You should also consider spherizing the magnetometer
response surface also, but this is a second order error that can usually be
ignored for most applications.

Kris

<https://github.com/kriswiner/MPU-9250/issues/9#issuecomment-81983197> .
<https://github.com/notifications/beacon/AGY1qtmz6x0pO5QysQ93uBPwPeyGPlxQks5
n11s3gaJpZM4DtXB1.gif>

pxdc, 2015-03-19:
Hi Kris.
I calibrated all sensors (including hard iron) and the yaw is now very better almost no drift in static position and moves good when i rotate the axis. however remain some errors.
the yaw axis drift when i pitch or roll the sensor. in static position there is no drift..
what can it be?

thank you

kriswiner, 2015-03-19:
There are other sources of error.

I mentioned the asphericity in the magnetometer response surface. There is
also a small cross talk between axes of all MEMS sensors, typically ~1% or
less. Both of these effects result in a change in yaw when pitch or roll
angle is changed. You can calibrate out the first source of error, but not
the second.

Lastly, the Euler angle approximation is not very good at some (higher)
angles; these angles are just a derived quantity from the more fundamental
and absolute quaternion representation. But there will be error in the yaw
from this approximation.

These errors, and still others (mechanical vibrations, electrical noise,
etc), are why the best hardware sensor fusion engines quote heading error of
between 2 and 4 degrees at best.

It is very hard to do better than this.

<https://github.com/kriswiner/MPU-9250/issues/9#issuecomment-83239006> .
<https://github.com/notifications/beacon/AGY1qvPiOYL80uAtaNVMoyNy2vEXDM1Zks5
n2hWKgaJpZM4DtXB1.gif>

pxdc, 2015-03-19:
i see..probably, I do not accomplish anything better than this.

one question: have you used DMP? from what I understand , the dmp is able to fusion the 9 axes, run-time calibration and provide yaw, picth and roll axis. what means run-time calibration? means that no manual calibration is needed (as I have done)? the calibration is done automatically? why not uses this feature instead of the discussed where?

kriswiner, 2015-03-19:
DMP is limited to 6-axis sensor fusion; there is no way to include magnetometer data in the DMP.

pxdc, 2015-03-19:
i think that exist a library to do this. but what i want to know is if dmp is able to fusion the 6 axis and provide yaw, pitch and roll. and the calibration is done automatically? sorry for the questions but after read about DMP i'm tempted to try :)

kriswiner, 2015-03-20:
Yes, you should try it. But I think you will find the results are not really
any better than the open source sensor fusion.

<https://github.com/kriswiner/MPU-9250/issues/9#issuecomment-83799339> .
<https://github.com/notifications/beacon/AGY1qpjJEhUT_3SCQH9J8tnYUtF9Yp2Bks5
n21DAgaJpZM4DtXB1.gif>

djsg, 2015-04-15:
Hi kris,

Sorry, first I posted this on other thread, and found here may be more appropriate.

Based on your 9250 code, I wrote a piece of firmware for my STMF401 sensor board which is connected with a MPU9250 through SPI. The obtained Quaternion values are as showed below:

ax = 4.211426 ay = -39.550781 az = -23.498535 mg
gx = 1.854966 gy = -0.740454 gz = 0.687027 deg/s
gx = -704.388916 gy = -356.964600 gz = -345.222534 mG
temperature = 20.997005 C
q0 = 0.416777
q1 = -0.695210
q2 = -0.463045
q3 = 0.358565
Yaw, Pitch, Roll: 57.775761 6.464238 -113.452301
average rate = 493125.218750

ax = 4.211426 ay = -39.550781 az = -23.498535 mg
gx = 1.854966 gy = -0.740454 gz = 0.687027 deg/s
gx = -704.388916 gy = -356.964600 gz = -345.222534 mG
temperature = 20.997005 C
q0 = 0.416761
q1 = -0.695179
q2 = -0.463089
q3 = 0.358587
Yaw, Pitch, Roll: 57.782719 6.463478 -113.451973
average rate = 493125.218750

Comparing your result, which almost remain the same for half an hour, my results changes rather fast. Can you tell me what could be the possible causes?

FYI, below I attached the printf messages during the init stage:
x gyro bias = -2.480916
y gyro bias = 0.519084
z gyro bias = -0.267176
x accel bias = -0.005310
y accel bias = 0.032104
z accel bias = 0.026733

MPU9250 initialized for active data mode....
6500 user control register = 64
force spi mode
reset mag
mag_asa 182, 184, 171
compass who am i 72. is it 72?

magCalibration {72.0000, 184.0000, 171.0000}
AK8963 initialized for active data mode....
Accelerometer full-scale range = 2.000000 g
Gyroscope full-scale range = 250.000000 deg/s
Magnetometer resolution = 16 bits
Magnetometer ODR = 100 Hz
Accelerometer sensitivity is 16384.000000 LSB/g
Gyroscope sensitivity is 131.072006 LSB/deg/s
Magnetometer sensitivity is 0.776487 LSB/G

You mentioned about sensor calibration, in your code there is function calibrateMPU9250(), which only works on acce and gyro. How about magnometer?

hi pxdc,

from your posts, it seems to me you added something to make your original work do better job. Can you share what are those extra things?

abstruse123, 2016-05-10:
Unable to read the magnetometer AK8963 device ID.
This is the function I used.
MPU_I2C_Read(0x0C, 0x00, 1, &devID);
It is returning junk. 
Any help?

kriswiner, 2016-05-10:
Did you set the MPU9250 into passthrough mode via the INT_CFG register? Can you discover the AK8963C via an I2C scan?

abstruse123, 2016-05-11:
Yes. This is my code handling that portion:  

  MPU_I2C_Write(MPU_I2C_ADDR, MPUREG_INT_PIN_CFG, 1, 0x22);
  Delay_ms(100);
  MPU_I2C_Write(MPU_I2C_ADDR, MPUREG_INT_ENABLE, 1, 0x01);
  Delay_ms(100);
  
  MPU_I2C_Write(MPU_I2C_ADDR, MPUREG_I2C_MST_CTRL, 1, 0x00);   // Disable I2C master
  Delay_ms(100);
  MPU_I2C_Write(MPU_I2C_ADDR, MPUREG_USER_CTRL, 1, 0x00);      // Disable FIFO & I2C master
  Delay_ms(100);

and this is my code to read the device ID of AK8963
MPU_I2C_Read(AK8963_I2C_ADDR, AK8963_WIA, 1, &devID);
Delay_ms(100);

The output is ÿ instead of H (0x48)

kriswiner, 2016-05-11:
When you scan for I2C devices, does the magnetometer show up? Before writing
0x22 to INT_PIN_CFG? After? It should be 0x0C.

The output is � instead of H (0x48)

�
<https://github.com/notifications/beacon/AGY1qhqq4hkBgyDxnzV0oEdT7GIrbCaRks5
qAWBBgaJpZM4DtXB1.gif>

abstruse123, 2016-05-11:
There seems to be something wrong with my I2C scan program.
It is not detecting MPU9250 as well as AK8963.
However, in my program I am able to read the device ID of MPU 9250 but not of AK8963

kriswiner, 2016-05-11:
This should work:

This should work:

// I2C scan function

void I2Cscan()
{
// scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
    
}

<https://github.com/notifications/beacon/AGY1qmsqXQjhNhVp_cKyX3u_Div9jPBkks5
qAXc9gaJpZM4DtXB1.gif>

abstruse123, 2016-05-11:
This is targeted for Arduino, while my target is dsPIC.
So I wrote my own I2C Scanner function. But, its not detecting anything on I2C bus.

I got one doubt.. According to MPU 9250 datasheet, if an I2C master sends its 7 bit I2C address along with write bit (0)/ read bit (1), it gives an acknowledge signal. 

I could not understand how it responds by sending 0x68?

kriswiner, 2016-05-11:
I have no idea how to program a PIC.

Masu141, 2017-04-20:
Hi Kris,
I'm trying to implement an MPU9250 in my proj,
when printing output values from sensor i see magnetometer doing this:
`Mag_X: -350.86		Mag_Y: 188.51		Mag_Z: -177.05

Mag_X: -350.86		Mag_Y: 169.04		Mag_Z: -178.39

Mag_X: -350.86		Mag_Y: 169.04		Mag_Z: -178.39

**Mag_X: -1435.67		Mag_Y: -14515.51		Mag_Z: -41090.56**

Mag_X: -350.86		Mag_Y: 171.38		Mag_Z: -187.78

Mag_X: -350.86		Mag_Y: 179.94		Mag_Z: -170.34

Mag_X: -350.86		Mag_Y: 178.39		Mag_Z: -175.70`

I've followed your sample for the ProMini 8Mhz

why i have the value flipping like here? It's all OK and i've to discard those values?

Thank you very much in advance!

Masu141, 2017-04-20:
Those are raw values:
Mag_X: -146		Mag_Y: 264		Mag_Z: 12
Mag_X: -154		Mag_Y: 261		Mag_Z: 11
Mag_X: -143		Mag_Y: 256		Mag_Z: 24
Mag_X: -146		Mag_Y: 257		Mag_Z: 8
Mag_X: -160		Mag_Y: 257		Mag_Z: 18
Mag_X: 8		Mag_Y: 170		Mag_Z: 254
Mag_X: -164		Mag_Y: 257		Mag_Z: 14
Mag_X: -157		Mag_Y: 261		Mag_Z: 20
Mag_X: -148		Mag_Y: 253		Mag_Z: 15
Mag_X: -158		Mag_Y: 270		Mag_Z: 6
Mag_X: -158		Mag_Y: 263		Mag_Z: 14

and this is the converted value: (2nd time readings)
Mag_X: -720.91		Mag_Y: 328.06		Mag_Z: -71.26
Mag_X: -747.80		Mag_Y: 274.29		Mag_Z: -86.86
Mag_X: -527.35		Mag_Y: -170.18		Mag_Z: 311.88
Mag_X: -740.63		Mag_Y: 356.74		Mag_Z: -69.52
Mag_X: -753.17		Mag_Y: 433.80		Mag_Z: -123.27
Mag_X: -756.76		Mag_Y: 419.46		Mag_Z: -97.26
Mag_X: -733.46		Mag_Y: 277.88		Mag_Z: -93.79

it's incredible..

kriswiner, 2017-04-20:
How can the raw values be negative?

Masu141, 2017-04-20:
i refer to this thread:
[https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS_t3.ino](url)

Here the raw value is stored in a signed 16bit value..

kriswiner, 2017-04-20:
Look at the byte values of the register.

GijsLui, 2017-07-19:
Dear KrisWiner, 
I've used your code from 1 april 2014 for MPU9520 wich I connected with the arduino uno according as is said in your code. 
I used a logic level transistor to get everything to 3.3v

In the beginnen most of the code was working. After a while wire.endTransmission was causing errors. There is no way of fixing it.

I tried to bypass the wire.endTransmission by using a different wire.h which goes out of the freesze.

My problem is that I do not get any data back from my MPU9520 in wire.endtransmission.

So all Im left with is 0 in my matrix. 

Can you give me a pointer or help me out?

kriswiner, 2017-07-19:
Replace the Arduino UNO with a Teensy 3.2 or a Butterfly
<https://www.tindie.com/products/TleraCorp/butterfly-stm32l433-development-board/>
or Ladybug
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
.

Or try an ESP8285
<https://www.tindie.com/products/onehorse/esp8285-development-board/> even.

If you have to use an UNO, do you have I2C pullups?

GijsLui, 2017-07-19:
Dear Kris Winer,

Thnx for swift reply, we do own the pull up resistors 4.7K, but we though a logic level transistor would be better xD

I'll try with the pull ups and contact you later today or tomorow, have to get a break after 7 hours.

Thnx again and yes I'm short on time and want to use the arduino uno and improve/update hardware/software from there.

kriswiner, 2017-07-19:
You will need both pullups on the I2C bus and a logic level translator.

GijsLui, 2017-07-19:
If you are still there, do you think I could do http://www.trackimu.net/single-post/2016/06/18/Tutorial-to-Using-Arduino-with-10DOF-IMU-MPU9250BMP180 

What they do here? Could I connect it the same way or would my chip burn because I put in to much volt?

kriswiner, 2017-07-19:
> If you are still there, do you think I could do http://www.trackimu.net/
> single-post/2016/06/18/Tutorial-to-Using-Arduino-
> with-10DOF-IMU-MPU9250BMP180
>
> What they do here? Could I connect it the same way or would my chip burn
> because I put in to much volt?

MPU9250 is 3V3 only, don;t feed it any 5 V signals.

SDA/SCL both need a resistor pullup to 3V3 in order for I2C to work.

GijsLui, 2017-07-19:
Do you think only pullups on the I2C bus will suffice? without the logic level transistor?

kriswiner, 2017-07-19:
No 5 V signals to the MPU9250.

GijsLui, 2017-07-19:
I bought  https://www.tinytronics.nl/shop/nl/sensoren/magnetisch-veld/mpu-9250-accelerometer-gyroscope-magnetometer-9dof-module-3.3v-5v    

Do you think your code will work on this one? I really like to get it working to show at the hospital, we have some really cool tests in mind.

kriswiner, 2017-07-19:
> veld/mpu-9250-accelerometer-gyroscope-magnetometer-9dof-module-3.3v-5v

Should work fine, already has the i2C pullups.

GijsLui, 2017-07-19:
Kris winer

I got the MPU9250 working and tomorrow I will use the other one. I have some questions  left:

1. In your youtube movies you visuelize the measurements (Really nice) What program did you use? How did you do it? (I was thinking matlab and get the info from port 7)

2. Is it possible to get the position relatively easy? (Mathematically the error gets way to big when I want to go to orrientation, how it works outside of the theory. I don't know.

3. I deleted a section of your code using // 
After that everything works.  Could you explain if it hurts to remove this part:

if (d != 0x48)
    {
      // Communication failed, stop here
      Serial.println(F("Communication failed, abort!"));
      Serial.flush();
      abort();
    }

After this I think I do not have more questions. I will write a report about it (Including a lot of other stuf).
Would you like to see it when it's done?
I will include you in the acknowledgements, is there a specific way you want to be mentioned?

Greetings

kriswiner, 2017-07-19:
No mention is required.
Position is very difficult without large errors.

If you are getting data this should be fine.

Shambler13, 2017-10-11:
Hi,
I'm using STM32L4 with the MPU9250.

I tried to read AK8963 id after enabling By-Pass mode.
I always get  nacked when i put the @ to 0xC

Please find below my code part  :   

errorValue |= BSP_MPU9250_SetRegValue(PWR_MGMT_1,0x01); 
  errorValue |= BSP_MPU9250_SetRegValue(SMPLRT_DIV,0x04);
  errorValue |= BSP_MPU9250_SetRegValue(CONFIG,0x03);
  errorValue |= BSP_MPU9250_SetRegValue(I2C_MST_CTRL,0x00);
  HAL_Delay(100);
  errorValue |= BSP_MPU9250_SetRegValue(USER_CTRL,0x00);
  HAL_Delay(100);
  
  
  errorValue |= BSP_MPU9250_SetRegValue(INT_PIN_CFG,0x22);
  HAL_Delay(100);
  errorValue |= BSP_MPU9250_SetRegValue(INT_ENABLE,0x01);
  HAL_Delay(100);
    errorValue |= BSP_MPU9250_SetRegValue(INT_PIN_CFG,0x22);
  HAL_Delay(100);

BR,
xxx

kriswiner, 2017-10-11:
what do you see with an I2C scan?

Shambler13, 2017-10-12:
Using the HAL_I2C driver,
When i call the I2C_RequestMemoryRead(), just after config to generate the I2C start, the addr flag doesn't rise and i have nackf.

I tried to use I2C_Master of the MPU9250 (using I2S_SLV0 registers and then reading on EXT_SENS_DATA_00, i read only 0x00
please find below my configuration of MPU
 errorValue |= BSP_MPU9250_SetRegValue(PWR_MGMT_1,0x80);
  errorValue |= BSP_MPU9250_SetRegValue(PWR_MGMT_1,0x01);
  errorValue |= BSP_MPU9250_SetRegValue(PWR_MGMT_2,0x00);
  errorValue |= BSP_MPU9250_SetRegValue(FIFO_EN,0x00);
  errorValue |= BSP_MPU9250_SetRegValue(PWR_MGMT_1,0x00);
  errorValue |= BSP_MPU9250_SetRegValue(CONFIG,0x01);
  errorValue |= BSP_MPU9250_SetRegValue(SMPLRT_DIV,0x00);
  errorValue |= BSP_MPU9250_SetRegValue(GYRO_CONFIG,0x00);
 errorValue |= BSP_MPU9250_SetRegValue(ACCEL_CONFIG,0x00);
  errorValue |= BSP_MPU9250_SetRegValue(USER_CTRL, 0x20);
  errorValue |= BSP_MPU9250_SetRegValue(I2C_MST_CTRL, 0x0D); /* 400 kHz on I2C Master */

Then to read the id
  errorValue |= BSP_MPU9250_SetRegValue(I2C_SLV0_ADDR, AK8963_ID|ReadEnable);
  errorValue |= BSP_MPU9250_SetRegValue(I2C_SLV0_REG, AK8963_WIA);
  errorValue |= BSP_MPU9250_SetRegValue(I2C_SLV0_CTRL, 0x81);
  HAL_Delay(200);
  
  errorValue |= BSP_MPU9250_GetRegValue(EXT_SENS_DATA_00, value);

Note:
I use I²C as interface, and the read ID from the MPU9250 works well i get 0x71, but from the AK8963 it return always 0x00

kriswiner, 2017-10-12:
I really have no idea what you are doing with the above code, but maybe you
should try a different MPU9250 breakout?

Shambler13, 2017-10-16:
Hi Kris,

I've used the MPU9250 and i wired it with a STM32-Nucleo , i can read the AK8963 ID, and the MPU9250 correctly.

But my issue is when the two components are hardwired in one board, i cannot read anymore the AK8963 Reg ID.
I compared the two schematic (on MPU Datasheet, and the one on board, PIN1 and PIN 20 are not connected to VDDIO and ground as des:cribed in datasheet.
FurtherMore, no pullup resistors on SCL/SDA lines.

I could not find any root cause or fix this problem by software

kriswiner, 2017-10-16:
Sounds like your breakout board is poor;y designed. Try adding pullup
resistors to the SDA/SCL lines or configure them as OUTPUT_PULLUP and see
if this changes. Even if you can read the I2C addresses, you won;t get good
mag data without pin 1 connected to 3V.

Shambler13, 2017-11-02:
Hi Kris,
A question offtopic,
In the register Map, i dont see any Registers or bit controls for the DMP "See MPU9250 register Map", i find that the bit definitions are marked reserved.

I'm confused, where can i find also the advanced hardware register map. I have no idea about it.

Thanks.

arhimed383, 2022-05-04:
MPU9250
How to read the SPI magnetometer data correctly?
Not to see the chip of the AK8xxx magnetometer by SPI

kriswiner, 2022-05-04:
https://github.com/kriswiner/MPU9250brianc118

arhimed383, 2022-05-05:
> https://github.com/kriswiner/MPU9250brianc118
> […](#)

The magnetometer does not respond, I tried your code, does not give a 0x48 response to reading the register 0x00 AK8963

arhimed383, 2022-05-05:
I changed the API, but it does not work, it does not respond yes to condon 0x00 with a response of 0x48.
readak8963registers(AK8963_WHO_AM_I,1,&_buffer);
/* reads registers from the AK8963 */
void readAK8963Registers(uint8_t subAddress, uint8_t count, uint8_t* dest)
{
	// set slave 0 to the AK8963 and set for read
	writeRegister(I2C_SLV0_ADDR, AK8963_I2C_ADDR | I2C_READ_FLAG);
	// set the register to the desired AK8963 sub address
	writeRegister(I2C_SLV0_REG,subAddress);
	// enable I2C and request the bytes
	writeRegister(I2C_SLV0_CTRL,I2C_SLV0_EN | count);
	// takes some time for these registers to fill
	HAL_Delay(300);
	// read the bytes off the MPU9250 EXT_SENS_DATA registers
	readRegisters(EXT_SENS_DATA_00,count,dest);
}
* writes a byte to MPU9250 register given a register address and data */
int writeRegister(uint8_t subAddress, uint8_t data)
{
	MPU_SPI_Write(&data, subAddress, 1);
	HAL_Delay(10);

	  readRegisters(subAddress,1,_buffer);
	  /* check the read back register against the written register */
	  if(_buffer[0] == data) {
	    return 1;
	  }
	  else{
	    return -1;
	  }
}