kriswiner/MPU9250 #11: Optimal register settings for Teensy 3.1

cojohnst, 2015-05-18:
Kris, 

I've been running the EMSENSR-9250 on a Teensy 3.1 using I2C, 72MHz, and your MPU9250BasicAHRS sketch (THANK YOU BTW!). For some reason, I can only get the Mahony Algorithm to run at 1030Hz, (not 2200 has stated in your description). I was wondering the settings you used for CONFIG, SMPLRT_DIV, GYRO_CONFIG, ACCEL_CONFIG, and other relevant registers to achieve 2200 Hz? Also, for some reason when I overclock to 96MHz, my loop rate in the MPU9250BasicAHRS drops to 880Hz, which seems quite odd. 

Thanks for any help!

-Corbin

kriswiner, 2015-05-19:
Are you using the i2c_t3.h I2C library and are you running with I2C at 400 kHz? The configuration I typically use is the same one in the sketch above.

cojohnst, 2015-05-20:
Thanks! The library switch from wire.h to i2c_t3.h appears to have been the issue... FYI, at the Wire.begin function in setup, I also had to change I2C_RATE_120 to I2C_RATE_1000 to achieve Sensor Fusion (Mahony) at 2100Hz+...

I'm still slightly confused by the CONFIG and SMPLRT_DIV settings. If we set CONFIG to 0x03 (as your default code states), our Gyro sample rate is only 1KHz (divided by 1+ SMPLRT_DIV (0x04)) gives us a 200Hz gyro sample rate, yet we're looping at 2100Hz in our main code... Does this mean we're only getting new gyro data into Mahony roughly once every 10 loops? And if so, would I need to change the CONFIG/DLPF_CFG, SMPLRT_DIV, and A_DLPF_CFG settings so the gyro and accelerometer sample rates are 8KHz and 4KHz respectively in order to sample new data on each loop in our 2100Hz SFA? 

Thanks so much!

kriswiner, 2015-05-21:
You don't want to have data sample rates the same as your sensor fusion
rate. The sensor fusion algorithm is iterative and you want the data sample
rate to be four or five times smaller than the fusion rate to allow the
fusion algorithm to converge. Unless you are riding a bucking bronco, 200 Hz
data rates are fast enough so your fusion rates need to be ~1 kHz. Scale
according to the specificas of your application.

<https://github.com/kriswiner/MPU-9250/issues/11#issuecomment-103947225> .
<https://github.com/notifications/beacon/AGY1qrayRLGPIDhFbmLqn9NqP-Vl3wqOks5
oLKzWgaJpZM4EfQyY.gif>

cojohnst, 2015-05-21:
Ok. And just for the sake of discussion, say I am riding a bucking bronco... What would the fusion algorithm do if my data rates were only 2-3 times less than my sensor fusion rate? Say I set my gyro to 1KHz, accelerometers to 1KHz, and overclocked my Teensy to fuse at 2.5KHz. Would this work or would the algorithm not converge?

Thank you for your help!

kriswiner, 2015-05-21:
Well, you would have to test it, but I would suspect that these open source fusion algorithms are pretty (simple and) efficient and 2 - 3 iterations per data time stamp ought to produce good enough results for you. Make sure to set your (low-pass) bandwidths at ~10-20% of your data sample rates so the data jitter doesn't confuse your fusion algorithms. But proof is in the eating...