kriswiner/MPU9250 #12: Yaw Pitch and Roll Estimation

akhilmohan4487, 2015-05-27:
Hello Kris,

I am back with more questions.

1. In the MPU9250 algo you have derived the Euler angles from quaternion representing orinatation of sensor wrt NED frame. How did you decided you pick one rotation sequences (Tait Bryan, Euler angle) and what is the basis of selection.Depending on the selection you have different set of Y,P and R values and out of which one is correct for one case. I have read a bit on this and discussed with few of my friends,but did not understood the matter well.
2. What is sensor to segment calibration and can you give me a simple example to understand this?
3. The system calculates accelero/gyro biases and magneto hard iron biases. How do I verify whether this is enough ?

I know I have asked too many theoretical questions, but hope at least you would be able to provide some materials or the path through which I have to look .

Thanks and Regards,
Akhil.

kriswiner, 2015-05-27:
Hi Akhil,

I don't know what you mean by sensor to segment calibration. Whether simple
bias calibration is enough depends on your application. There are more
sophisticated corrections that can be applied that correct for know errors
in MEMS sensors due to cross-axis sensitivity, pc board stress, thermal
drift, etc. The simple bias calibration usually suffices for all but the
most demanding applications.

I will try to provide a better answer tomorrow but for now:

These are well-known white papers on magnetometer calibration which I hope
will be useful to you:
	
	http://www.freescale.com/files/sensors/doc/app_note/AN4246.pdf
	
	http://cache.freescale.com/files/sensors/doc/app_note/AN4248.pdf

Kris

<https://github.com/kriswiner/MPU-9250/issues/12> .
<https://github.com/notifications/beacon/AGY1qhTwRd1fyKzq6GyZndRZC0htKmFFks5
oNU-_gaJpZM4ErXP_.gif>

akhilmohan4487, 2015-05-27:
Hi,

Sensor to segment calibration - Suppose I am attaching the IMU on a solid body like human hand for analyzing joint angles, tremors and so on what I have read is we have to align our sensor to segment (hand) which might have a different co-ordinate conventions and this segment movement wrt external global can be used for obtaining parameters of interest. What my confusion is does these kind of things have to be done say if IMU is fixed on a vehicle or at present how is your code is  intended to use (application wise). I hope I am asking my doubts in the correct way. 

I will read the articles you provided by mean time. 

Thank you very much Kris :)

Regards,
Akhil

kriswiner, 2015-05-28:
Take a look here:

http://www.vectornav.com/support/library/calibration#accelerometer

<https://github.com/kriswiner/MPU-9250/issues/12#issuecomment-105834588> .
<https://github.com/notifications/beacon/AGY1qloQMaa2kwWyU-24-SzCeTh1ole_ks5
oNYLagaJpZM4ErXP_.gif>

akhilmohan4487, 2015-05-28:
Yea, I have read this before and the mathematics seems to be not fully correct and they have not described all of those. But as a starting point of calibration it helped me. 
I am trying to develop a rehabilitation game platform for people with stroke and the input to the game is IMU's. So I am not bothered absolute values, but dont want to see a change in the system response during runtime (dynamic errors which might be due to soft-iron biases, vibration etc.). So If I have a system which can calculate this dynamically at the start of experiment, I hope dynamic change would be minimal.
After reading the article you provided, I feel like in the static condition I can use tilt compensated algorithm for deriving required parameters and in dynamic, the Madgwick with low beta (the Madgwick algorithm assumes errors are equal in all axes and then I have to find this in ad-hoc manner). What is your opinion on this ?

Thanks and Regards,
Akhil

kriswiner, 2015-05-28:
I think that simple bias calibration of the sensors, coupled with 9-axis
Madgwick or Mahoney fused quaternions for absolute orientation is a good
enough solution for most purposes, including yours. The other errors people
might worry about are for absolute position tracking over long times, which
no MEMS sensor suite can do well.

<https://github.com/kriswiner/MPU-9250/issues/12#issuecomment-106173064> .
<https://github.com/notifications/beacon/AGY1qiAxn3UBep88ss9stI5P6N8-Iy1Gks5
oNpjmgaJpZM4ErXP_.gif>

akhilmohan4487, 2015-05-28:
Hello,

I also hope this might be good enough. I will do a sample test and inform you the progress.

Thanks Kris :)

Best Regards,
Akhil

akhilmohan4487, 2015-05-30:
Hello,
I was reading ISB recommendation on JCS (joint co-ordinate system) for fingers and ended up in real confusion esp in this para,
"Rotations are described using Euler angles. For a clearer interpretation of these angles it is suggested that the coordinate systems of the proximal and distal body segments are initially aligned to each other by the introduction of ‘anatomical’ orientations of these coordinate systems. The rotations of the distal coordinate system should then be described with respect to the proximal coordinate system. If both coordinate systems are aligned, the first rotation will be around one of the
common axes, the second rotation around the (rotated) axis of the moving coordinate systems, and the third rotation again around one of the rotated axes of the moving coordinate system. This last axis is preferably aligned with the longitudinal axis of the moving segment."
What I understood is,
1. The qSG derived from Madgwick have to change in to qBG (quaternion which describes orientation of body in the global frame). For this, we should know how the IMU sensor coordinaes in body co-ordinate frame (qSB). For an initial pose this is known and will be constant in an experiment.
so qBG = qSG X q'SB where ' is the conjugate

2. Did not understand how to derive the Euler angle sequence for calculating joint angles

Do you have any idea in this area ?

Thanks and Regards,
Akhil.

kriswiner, 2015-05-30:
Hi Akhil,

All I know is that Euler angles are rather easily derived from the
quaternions. So if you have the quaternions in each of the two coordinate
frames, you should be able to derive the respective Euler angles.

I prefer to work in the lab frame where everything is oriented relative to
an absolute and fixed reference frame.

Kris

1.	Did not understand how to derive the Euler angle sequence for

<https://github.com/kriswiner/MPU-9250/issues/12#issuecomment-107074049> .
<https://github.com/notifications/beacon/AGY1qu0fi513PkHSytnAQH7bLyjnVGFAks5
oOfgdgaJpZM4ErXP_.gif>

akhilmohan4487, 2015-05-30:
Hello,

Okay, I will do a bit more reading and hope to resolve the problem.

Thanks Kris.
Akhil

akhilmohan4487, 2015-06-04:
Hello Kris,

Few more doubts I have,

 1. I tried to calculate roll and pitch using acclero and yaw with compass (tilt compensated) and the results looks fine.

2. Now the same I am trying to calculate using gyro and this is not getting as I wanted,
    float angle_gyro_x = 0, angle_gyro_y = 0,angle_gyro_z = 0;
    angle_gyro_x += gx2 * 0.005;
    angle_gyro_y += gy2 * 0.005;
    angle_gyro_z += gz2 * 0.005;
If this basic thing worked I am also interested to calculate angle using Runge-Kutte integration method to avoid the jitter (expecting from gyro). Am I on right track ?

3. Also what is the role of sensitivity values given for accelro, gyro and compass like 131LSB/deg/sec

4. In your compass mRes ,
void getMres() {
  switch (Mscale)
  {
    case MFS_14BITS:
          mRes = 10.0*4219.0/8190.0;  should be 10.0*4800.0/8192.0 this right ??
          break;
    case MFS_16BITS:
          mRes = 10.0*4219.0/32760.0; should be 10.0*4800.0/32768.0 this right ??
          break;
  }
}

Thanks and Regards,
Akhil

kriswiner, 2015-06-04:
Hi Akhil,

Integrating the gyro alone is unlikely to give you anything sensible because the gyro drifts fairly rapidly. This is why people use sensor fusion so the mag can be used to correct for the gyro drift.

The sensor sensitivity simply tells you how to convert from raw counts to meaningful output like degrees per sec or gs, etc

The AK8963C spec states that the max mag output is 42.19 Gauss, even though it is rated as a 48 Gauss sensor. Maybe an e-mail to AK would clear this up.

Kris

akhilmohan4487, 2015-06-04:
Hi,
Only with a 3-axis gyro I thought we can estimate angle for short periods (1min) and this drifts with time, and to remove we obtain info from other sensors. I was reading several blogs and people have able to estimate the angle in this crude way (simple integration). I am not using interrupts, so do I have to verify the gyro actually samples at 200Hz (burst mode in FIFO only guarantee the specified fs or otherwise DRDY interrupt for check) ? (Is this is what causing error)

What I meant is not scale adj but sensitivity Invensense have given (Sensitivity Scale Factor), in the calib routine you have used this like gx2*scale/gyrosensitivity and not while directly reading the registers.

Thanks for the quick response :)
Akhil.

closed 2015-09-09 by akhilmohan4487

kriswiner, 2015-09-09:
Did you get your project to work out all right Akhil?

akhilmohan4487, 2015-09-09:
Hello Kris,

Hope you are doing good. I have developed framework using dsPIC for obtaining the info from MPU, and implemented a basic first order filters to smooth the data from MPU and secondly the Mdgwick. I have not calibrated it using an another one, and will be doing in this month, so not sure how much accurate the results could be.
When I perform the quat to euler conversion with a singularity check, it looks okay, but still not enough.
Glad that you still remember me and thanks for all the input you provided.

Best Regards,
Akhil