kriswiner/MPU9250 #13: Different Chip breakouts:  VDDI = ADO

iclevine, 2015-06-23:
Hello! This isn't an issue as much as a note:
I have one of these breakouts
[image]

and there isn't a VDDI pin so I wasn't sure how to connect the breakout to power in the proper way. I did some schematic cross-examination and figured out that it is the same as the "ADO" pin on the alternative breakout configuration (the one available on amazon.com). Just thought I'd post that so y'all wouldn't have to go to the trouble I did to figure it out.

kriswiner, 2015-06-23:
For the breakout pictured above VDDIO and VDDA should be tied together, and both are driven by 3V3 coming out of the voltage regulator. This allows 5 V into VCC on the board edge. The ADO should be the I2C address LSB. If they have ADO tied to VDDIO it should be through a pullup resistor so it can be toggled to GND to change the address.

iclevine, 2015-06-23:
I wasn't able to get it to communicate via I2C until it the "ADO" was hooked up to power. Do you have any idea why this might be?

kriswiner, 2015-06-23:
ADO is the LSB of the I2C device address, so behavior depends on the I2C
address you used. With ADO = LOW, I2C address = 0x68. When ADO= HIGH, I2C
address = 0x69. Which did you use? I bet 0x69. Try connecting ADO to GND and
change the I2C address in your code to 0x68. Should work.

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-114612673> .
<https://github.com/notifications/beacon/AGY1qvq0MJqFlGUJGawZNmplJmUoKeYFks5
oWaXHgaJpZM4FKHax.gif>

iclevine, 2015-06-24:
awesome! Thanks Kris

joshgrob, 2015-09-11:
Hi Kris,

First off thank you for the great code tutorials and being active on the forums.  It is a big help.  I want to piggy back on this thread, and will you help me troubleshoot my wiring?  I have the next version of the board mentioned above (it has the MPU-9255/GY-9255 - http://amzn.to/1NsQm3y) which I believe is the same breakout schematic other than the MPU.  Reading your last post and from the code I have setup my wiring as:
Arduino Uno                    GY-9255
+5v                                  VCC
GND                                  GND
A4       -> 4.7k resistor -> SDA
A5       -> 4.7k resistor -> SLC
GND                                 AD0

When I do this and upload the code the WHO_ AM_I lookup always returns 0xFF instead of 0x73.  I tried several other configurations (eg. w/o pullup resistors, moving AD0 to power), and still the same 0xFF.  Do you have any thoughts what is wrong?  Thank you for the time and help.

Best,
Josh

kriswiner, 2015-09-11:
Hi Josh,

There are two potential problems. One is connecting the 3V3 SDA/SCL lines of
the board to the 5 V UNO, which might be sending 5 V to the chip, which
isn't good. You probably need a voltage translator between these. The other
possibility is that you have the wrong I2C address. If you do an I2C scan,
what I2C devices show up on the bus?

Kris
Pesky Products

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-139453540> .
<https://github.com/notifications/beacon/AGY1qo5lsCkEHPxvUYldSJHmkZ6-joiRks5
owll-gaJpZM4FKHax.gif>

joshgrob, 2015-09-12:
Great suggestion on the scan, and that lead me to find bad solder connections on my part.  I finally found a device on x68 and was able to connect and get ypr readings.  My final wiring was 3V3 to VCC, AD0 on GND and no pullups on the SCL/SDA since it appears the board has them already.  Thank you for the help to get me talking to the device.

kriswiner, 2015-09-12:
Hi Josh,

Glad I could help. You might find other interesting devices to play with
here: https://www.tindie.com/stores/onehorse/.

Kris
Pesky Products

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-139790903> .
<https://github.com/notifications/beacon/AGY1qhjEHswUmTHXdvONL-5ldn8C2Womks5
oxEOqgaJpZM4FKHax.gif>

LyubomyrD, 2016-02-03:
Hello.
I need to use 0x69 address for IMU GY-9250 in my project. Do I have to make any changes in the code besides changing the address of the IMU? I connect ADO to 3.3V (its Arduino Due) and that is all, right?

kriswiner, 2016-02-03:
Should be enough.

Use an I2C scan to check.

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-178961714> .
<https://github.com/notifications/beacon/AGY1qvRQaJL6x-QcRVuAW_BCMnMmnn66ks5
pgVxSgaJpZM4FKHax.gif>

LyubomyrD, 2016-02-03:
Thank You very much for reply! I used I2C scanner and the address is indeed 0x69. 
But is the data from IMU received in the same manner? I mean, I don't need to make changes in the function which reads data from IMU, do I?

kriswiner, 2016-02-03:
Everything else should be the same...

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-178968230> .
<https://github.com/notifications/beacon/AGY1qsfSjNPFCukzQmDA7XXvbCQoMUT7ks5
pgWIfgaJpZM4FKHax.gif>

LyubomyrD, 2016-02-03:
Thanks a lot!

JanKokalj, 2016-02-04:
Hi,
I just got the same board (GY-9255) and since I dont have any experience with these things I was wondering if some tutorial/example is available somewhere? I dont find anything for this board on the web.
Thanks

kriswiner, 2016-02-04:
I don't know anything about these boarda and cannot really help you here. I
make these:

https://www.tindie.com/products/onehorse/mpu9250-teensy-31-add-on-shields/

and recommend them to your attention.

Kris

<https://github.com/kriswiner/MPU-9250/issues/13#issuecomment-180017592> .
<https://github.com/notifications/beacon/AGY1qtRaa7n5DBr6aEJLf7Dr3C7xuuWLks5
pg6AxgaJpZM4FKHax.gif>