kriswiner/MPU9250 #14: EMSENSR-MPU9250 burning up/shorting randomly (Teensy 3.1)

cojohnst, 2015-07-09:
Guys,
I've been running Teensy 3.1 connected to an EMSENSR-MPU9250 successfully, however after 2-8 hours of operation the 9250 breakout board is burning up and shorting VDDIO to GND, and then burning up my Teensy 3.1. 

I've contacted the embedded masters (mfg of EMSENSR-MPU9250) with little success. I wanted to make sure that my wiring was correct, or if anyone else was having this issue?

Teensy 3.1 -------------- EMSENSR-MPU9250
3.3V ----------------------- VDD
3.3V ----------------------- VDDIO
GND ----------------------- GND
SDA0 (PIN 18)------------ SDA
SCL0 (PIN 19)------------ SCL

I have been using the pullup resistors built into 9250 Breakout board, and haven't added any other resistors to my system. 

Again, I've been able to communicate with the 9250's successfully for anywhere from 2-8 hours, but sometime in that window I've been experiencing the short and burn-up glitch. 

Anyone have any insight or ideas? 

Thanks for the help!

-Corbin

kriswiner, 2015-07-09:
I've never run my EMSENSR-9250 board for eight hours straight, but for an hour or so I have never experience this kind of problem. How are you connecting the two? Maybe you have introduced an intermittent short? Did you solder wires between the Teensy and 9250 board or are you using clip on leads? The MPU9250 is a pretty low current device and the Teensy GPIOs can handle 25 mA so I can't understand how you could be overloading the Teensy. Are you using the 3V3 regulated output of the Teensy to power VCC and VDDIO of the EMSENSR board? Are you using the GND (not AGND) of the Teensy? Mysterious...

cojohnst, 2015-07-09:
I'm powering the Teensy using USB, and then using the wiring diagram above in my first post. All wires between the Teensy and the 9250 are soldered, and I've checked soldering with an ohm-meter to ensure no shorts. I've had the same burnout both with the 3.3V regulated output of the Teensy, and 2 diodes stepping down from 5 to 3.6V (which is a little high). I'm currently constructing a voltage divider to come down from 5V to 3.3V. 

I have been curious about GND and AGND on the Teensy? I have been using them interchangeably, but based on your response it sounds like I should only use GND? 

Thanks!

kriswiner, 2015-07-09:
AGND is a separate analog ground useful to keep analog circuits isolated from noisy digital lines like I2C. This is useful, for example, when measuring a voltage on the Pin 14 ADC. You may use AGND as GND but it offers no advantage when talking with the MPU9250, can't overload the device, but might interfere with other analog devices in the circuit noisewise. If the EMSENSR has a design flaw (which I doubt, those guys are very good designers), can you reproduce the problem with a different MPU9250? Like these: https://www.tindie.com/products/onehorse/mpu9250-teensy-31-add-on-shields/ or the cheap Chinese versions.

cornicis, 2016-02-11:
@cojohnst Did you manage to solve the issue? I connected the EMSENSR-board like you did, instead of an arduino/teensy I use a single atmega328 @ 3.3V.
I burned-up my MPU9250 as you did and my ATmega followed shortly afterwards. VDDIO and GND are shorted now and not responding...
I looked at the schematic of the tindie-board, vdd is connected there to vddio by default.

Could it be that in some strange case (on power up?) VDDIO is greater than VDD? Or are the additional pullups on the EMSENSR posing a conflict? I would try without pullups as a next step, but I've got only one board left...

cojohnst, 2016-02-11:
I was never able to isolate the original problem when connected to the Teensy, but when I really let the magic smoke out, I had 2 diodes in series dropping a 5V line to 3.6 Volts. The maximum rating on the 9250 is 3.6V, with they "typical" voltage being 2.5V for VDD and 1.8V for VDDIO. 

[image]

If you have a bad voltage regulator somewhere, or anywhere in the circuit where think its 3.3V and its actually closer to the 3.5/3.6V range, the chip will burn out and cascade backwards from there. 

As far as pull-ups, I'm using everything internal to the Teensy and the EMSENSR with no issues. Sorry that's not much help, but verify your voltages and go from there. I'm using the wiring diagram above with no issues, so I might have had bad breakout boards or regulators originally.