kriswiner/MPU9250 #15: I2C bypass enabled is creating issues

ByReaL, 2015-07-11:
I have tried mpu9250 init code you wrote and the moment i set the BYPASS_EN bit to 1 the I2C bus goes berserk
(INT_PIN_CFG, 0bxxxxxx1x) # BYPASS_EN = 1
(USER_CTRL, 0bxx0xxxxx) # I2C_MST_EN = 0
i see in the docs the below note, is there any way we can explicitly disable those pull-ups ?
 When asserted, the i2c_master interface pins(ES_CL and ES_DA) will go into ‘bypass mode’ when the i2c master interface is disabled. The pins will float high due to the internal pull-up if not enabled and the i2c master interface is disabled.

after the bypass gets enable the signal looks like it's pulled too much up or the 0 reference is totally off
[image]

nice clean bus prior to enabling the bypass
[image]

anyway any idea what is causing this and what setting must be made to properly enable the bypass.

kriswiner, 2015-07-11:
Do you have external pullup resistors on the host I2C bus?

bad
<https://cloud.githubusercontent.com/assets/3046965/8635517/a947b02a-27da-11
e5-9077-11f5b2bee1fd.png> 

good
<https://cloud.githubusercontent.com/assets/3046965/8635516/a945d12e-27da-11
e5-876d-125f289bdf3b.png> 

<https://github.com/kriswiner/MPU-9250/issues/15> .
<https://github.com/notifications/beacon/AGY1qhHs3YBL-DMHBBZpxvbs17i4xjA4ks5
ocYUigaJpZM4FWxLI.gif>

ByReaL, 2015-07-11:
I connected the scope to the AUX_CL and AUX_DA lines and i configured the 9250 in I2C master mode to read the magnetometer and all work fine and the I2C captured signal  going to the magnetometer is perfect and i can see the values in EXT_SENS_DATA_00 registers.

to answer your question:
 - i use Intel Edison Arduino that has internal pull-ups 
 - between Arduino and MPU9250 there is no external pull-up
 - on the AUX line of the MPU9250 there is no external pull-up

i tried to disable the Intel Edison internal pull-ups but no change in the signal (although i'm not sure they actually got disabled)

kriswiner, 2015-07-11:
If I remember correctly the Edison pullups are rather weak so I would simply
try adding 4K7 pullups on the host SDA/SCL lines and see if you get any
improvement. In our Edison applications we alw2ays add 4K7 pullups to th I2C
bus lines, and never depend on internal puulups.

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-120666009> .
<https://github.com/notifications/beacon/AGY1qlM7nFF_wc_L0sDOBzGfgeV5DXeOks5
ocY90gaJpZM4FWxLI.gif>

ByReaL, 2015-07-11:
i can configure them from 50K to 910ohms, i tried this also,
/available_pullstrength
2k      20k     50k     910ohms

experiment:
with external 1K pullup and bypass enabled between logic 0 and 1 there is only a 500mV difference
without it and bypass enabled there is a 1100mv difference

if i measure completely disconnected and powered off the resistance between 
AUX_CL and AUX_DA pins and VDD ( i made sure i measured pin 13 of the chip and just to be sure i measured also the system vcc to make sure there is no pull up to the 5V line either)
so i get ~90K ohms between AUX pins and VDD, this confirms that there is no external pull ups on the AUX lines. (and many megs between AUX and system 5V line)

conclusion:
based on this experiment and previous oscilloscope readings i don't think we have a weak pull-up issue, i believe we have an issue because of a too strong pull-up and there is nothing i can think off besides the 9250 internal pull-ups not being disabled.

kriswiner, 2015-07-12:
It is always possible your MPU9250 board is defective. What are you using?

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-120673248> .
<https://github.com/notifications/beacon/AGY1qjjBZCF_A3M7jVvkXNaWS-a7HFeqks5
ocaTFgaJpZM4FWxLI.gif>

ByReaL, 2015-07-13:
http://www.amazon.com/gp/product/B00OPNUO9U?psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00
and looking more careful i see it has 2 10K pull ups on the SDA/SCL lines (although i don't think those are the problem, once i have the new board i'll remove them and give it a try)

i just bought EMSENSR-MPU9250 from invensense website i'll report back once i try with that one.

kriswiner, 2015-07-13:
Looks like you might have to pull nCS high to enable I2C, or is there a
pullup on this line?

http://www.amazon.com/gp/product/B00OPNUO9U?psc=1&redirect=true&ref_=oh_aui_
detailpage_o00_s00

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-121067395> .
<https://github.com/notifications/beacon/AGY1qi2zuSyZqoM5l_6dFt-O-1auapnBks5
odCcvgaJpZM4FWxLI.gif>

jaime-rq, 2016-06-02:
Hi, 
Sorry to post on this thread after so long.
But I am wondering if you were able to fix this issue. I am having the same problem, I can read Gyro, Acc, but fail to read Mag :(

Any help it's much appreciate it :)
Thanks :)

kriswiner, 2016-06-02:
Write 0x02 to INT_CFG

(#15)

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-223404176> ,
3nks5qHzXkgaJpZM4FWxLI> .
<https://github.com/notifications/beacon/AGY1qnFDXyRLE0ZX3fYqt7YXX3bgwRDfks5
qHzXkgaJpZM4FWxLI.gif>

jaime-rq, 2016-06-02:
Thanks for replying so quickly :)
Do you mean by INT_CFG the (0x37) 55 register?
So it would be equivalent to do I2C bypass.
If I try to do the I2C bypass, the whole device becomes unresponsive, I can't do any other I2C read or Write.
I am using the latest UPM library, I am using MPU9250

Thanks :)

kriswiner, 2016-06-02:
Don't know what this is: the latest UPM library

Try:
https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

(#15)

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-223406029> ,
QWks5qHzergaJpZM4FWxLI> .
<https://github.com/notifications/beacon/AGY1qtQ514ZzI_YYqzwzpU8JQqSsP0-uks5
qHzergaJpZM4FWxLI.gif>

kriswiner, 2016-06-02:
> (#15)
>
> i wasted a lot of time on this and nothing, in the end i just drooped this
> and purchased
> https://www.bosch-sensortec.com/bst/products/all_products/bno055
>
> for me exactly that was the problem the moment i would enable that bit it
> will all go south.

Which MPU9250 breakout are you (did you) use?

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-223406621> ,
Uoks5qHzg4gaJpZM4FWxLI> .
<https://github.com/notifications/beacon/AGY1qoGQ5iYFYXc9XcypXP3k5TdgvSdWks5
qHzg4gaJpZM4FWxLI.gif>

jaime-rq, 2016-06-03:
This is the one I am trying to use: http://www.amazon.com/SeeedStudio-Grove-10DOF-Source-BOOOLE/dp/B00WA75G6Y/ref=sr_1_1?ie=UTF8&qid=1464955793&sr=8-1&keywords=grove+imu+10

I use a Intel Edison, from what I have research so far, it may be due to the pull up/down resistors that the Edison has integrated, but I am not sure, neither how to fix it. 
Still looking :)

Thanks :)

kriswiner, 2016-06-03:
I have had a lot of trouble with the Edison. If you have a Teensy 3.1 or 3.2
I would recommend you at least try to make your MPU9250 with it. I have had
no trouble using the Teensy and MPU9250 together.

(#15)

http://www.amazon.com/SeeedStudio-Grove-10DOF-Source-BOOOLE/dp/B00WA75G6Y/re
f=sr_1_1?ie=UTF8&qid=1464955793&sr=8-1&keywords=grove+imu+10

<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-223562817> ,
7Gks5qIBoJgaJpZM4FWxLI> .
<https://github.com/notifications/beacon/AGY1qneoxSmp4GqwxclpZpbeWZ-3YN5pks5
qIBoJgaJpZM4FWxLI.gif>

pierluca87, 2016-10-26:
Hello kriswiner. I have a doubt on GROVE 9 DOF MPU9250: I'm not sure if the beakout board is equipped of pull up resistors or not. Reading the DS there is a circuit aimed at performing the IIC level conversion from 5 V to 3.3 V, but I'm not sure if resistors R3 and R4 performe also the pull-up function: could you give me some informations if you know?

Thank you

[Grove - IMU 9DOF v2.0.pdf](https://github.com/kriswiner/MPU-9250/files/553202/Grove.-.IMU.9DOF.v2.0.pdf)

kriswiner, 2016-10-26:
Looks like the board has 10 K pullups.

(#15)

<https://github.com/kriswiner/MPU-9250/files/553202/Grove.-.IMU.9DOF.v2.0.pd
f> 

-
<https://github.com/kriswiner/MPU-9250/issues/15#issuecomment-256331876> ,
ot6XEraks5q30bVgaJpZM4FWxLI> .
<https://github.com/notifications/beacon/AGY1qt149tArDSVPP8iqtksixBK_Bemcks5
q30bVgaJpZM4FWxLI.gif>

A1145681, 2020-07-23:
Hi,
  Hope you are doing well. But I do encounter the same issue with MPU9250 and the official DMP library. I'm working with TI's TM4C123G evaluation boards and can successfully read device id. However, after switching to bypass mode to config the compass then switch back, I2C would be broken and the following read/write commands all failed. However, if I disable the compass then there's no issue with I2C, as bypass mode is not entered. Did you have any change to solve this issue? Really appreciate your help.

kriswiner, 2020-07-23:
Never had any trouble using the MPU9250 in bypass mode or otherwise.
However, if you use bypass mode to configure the AK8963, I would recommend
you keep bypass mode active in order to read the data from the AK8963.

A1145681, 2020-07-23:
Thanks a lot for your reply. I also tried enabling bypass mode without existing, Then I can read the compass data, but got 0xff for all the gyro/accl data. Before entering bypass mode reading gyro/accl data works fine.

kriswiner, 2020-07-23:
Sounds like an I2C software error to me...

In bypass mode, the AK8963C is accessible by the MCU. This is basically a
passive switch internal to the MPU9250. So you should still be able to
reach the MPU6500 (gyro/accel).

If you select bypass mode (I2C only) first and then do an I2C scan, do you
see both 0x0C and 0x68 on the I2C bus?

A1145681, 2020-07-23:
Thanks a lot for helping me. Actually, I did not have the I2C scan functionality implemented on my platform. But I tried to read the device ID instead, with bypass enabled. For MPU6500 I got 0xFF (not expected) while for compass, I got 0xC (expected). Seems like while turning on bypass mode, I cannot talk to the MPU6500 anymore. So that I cannot really exit the bypass mode.

kriswiner, 2020-07-23:
Then something is way wrong with your i2c api.

kriswiner, 2020-07-23:
How are you putting the MPU9250 in bypass mode?

A1145681, 2020-07-23:
There's a "mpu_set_bypass" function in InvenSense's released MPL library. You can find an example here: https://github.com/vedranMv/tm4c_mpu9250/blob/master/mpu9250/eMPL/inv_mpu.c
In that function, it firstly sets "i2c_mst_en" bit of user_ctrl register to 0, then set "bypass_en" bit of cfg register to 1

kriswiner, 2020-07-23:
Gee, I wonder if setting the I2C enable bit to zero might have something to
do with it?

A1145681, 2020-07-23:
Which I2C enable bit? You mean set "i2c_mst_en" to 1 instead of 0 in bypass mode?

kriswiner, 2020-07-23:
I am not sure why you are asking me instead of Invensense since you seem to
be using their code. But I can imagine if you set the master I2C enable bit
to 0 then the MPU6500 won't respond to I2C commands!

A1145681, 2020-07-23:
I think set "i2c_mst_en" to 0 will disable MPU9250 as the master to communicate with the compass. So in this way the host microcomtroller could talk to compass directly. But anyway, I'll check with InvernSense

lbdroid, 2021-03-03:
I don't know if its about "fake chips" or not, but, unsurprisingly, I have discovered some quality/performance issues with some of the cheap breakout boards that can be purchased from amazon or ebay, and the behavior of some parts mirrors what I've seen in this thread.

The first board -- a purple PCB labeled as "GY-91" with an accompanying BMP-280 pressure sensor: https://www.amazon.ca/DIYElectronic-MPU9250-Acceleration-Gyroscope-MPU-9250/dp/B085HCGLWN (I didn't pay that much for it, this is just a listing to the board in question). Plug this board in and two devices appear on the i2c bus at addresses 0x71 (mpu9250) and 0x76 (bmp280). If I switch it to bypass mode, then again, two devices appear on the bus, at addresses 0x0c (ak8963) and 0x76 (bmp280) --- NOTE that the mpu9250 is now GONE from the bus. In addition, if I set up its master interface and try to communicate with the ak8963 through the mpu9250 slave registers, I don't get any response from the ak8963. Something is definitely wrong there. In addition, I would like to point out that this device is NOT SAFE to connect to 3.3v circuits unless you bypass the built-in voltage regulator, since it level shifts the i2c lines between 5v (external) and 3.3v (mpu9250). I can't say if this behavior is typical for these boards, since I only have one of them.

The second board -- a blue PCB labeled "MPU-9250/6500": https://www.amazon.ca/gp/product/B07TV4L6HT -- I have a total of 4 of these boards, of which 3 work apparently correctly, and one appears to have a dead ak8963. An i2c scan shows 1 device at 0x71, or two devices at 0x0c and 0x71 in bypass mode, and I'm able to communicate with the ak8963 both in bypass mode and in master mode. Except for the dead one -- in bypass mode it still shows just 0x71, and there is no communication with the ak8963 in master mode. So like it just doesn't have an ak8963. The IC is labeled "MP92", though I suppose it was possibly mislabeled. Never know, I often think that the source of these budget breakout boards is the dumpster behind the factory that makes them, so you end up with all kinds of weird things going wrong with them.

Anyhow, the point obviously is this; its easy to get your hands on bad hardware that behaves in strange ways, so the rational first thing to do if it isn't working as expected, is to try new hardware.