kriswiner/MPU9250 #16: Accelerometer calibration

xwirth, 2015-07-16:
Hi,

for calibration of the accelerometer, you divide the bias by 8  : 
  accel_bias_reg[0] -= (accel_bias[0]/8); // Subtract calculated averaged accelerometer bias scaled to 2048 LSB/g (16 g full scale)
  accel_bias_reg[1] -= (accel_bias[1]/8);
  accel_bias_reg[2] -= (accel_bias[2]/8);

But for accelerometer, offset is for all Full Scale Range with 1024 LSB/g sensitivity, and since measurement are made in 16384 LSB/g, bias has to be divided by 16.

kriswiner, 2015-07-16:
It's more complicated than this. According to the Invensense application note on calibration, the bias is stored at 4096/LSB sensitivity IIRC. But I found that if I used this I got the wrong bias correction. So I used 2048/LSB which produced better results. This might have to do with the way I am treating the temperature bit, which I need to correct. So the point is, the scale at which the bias registers store the data has nothing to do with the scale used to collect the data. And I ended up usually just applying the bias in the main loop anyway and not using the accel bias registers. I do use the gyro bias registers though.

xwirth, 2015-07-16:
It seems that I misinterpreted your code ^^'. In fact you store the bias as if it begins from bit 0 instead of bit 1. So dividing by 8 instead of 16 is correct. 

For the temperature bit, maybe using 0xFE mask for LSB will help :
  data[1] = (accel_bias_reg[0])      & 0xFE;

kriswiner, 2015-07-16:
Yes, this is the fix.