kriswiner/MPU9250 #17: Environmental magnetometer values

RayBreslin, 2015-07-17:
Hi Kris,

I have been following your work for quite some time now and made a nice smartphone app running a orientation filter simulation. Since I am trying to consider every last tweak in calculating propper results from my MPU9250-tags and to improve my simulation the only thing I cant realy grasp is your environmental correction approach for the magnetometer.

    magbias[0] = +470.;  // User environmental x-axis correction in milliGauss, should be automatically calculated
    magbias[1] = +120.;  // User environmental x-axis correction in milliGauss
    magbias[2] = +125.;  // User environmental x-axis correction in milliGauss

From my understanding its like using the built-in magnetometer of a smartphone, taking a decent magnetometer app, doing the figure-8-calibration and afterwards reading the shown magnetometer values in a static scenario (table edge). Is that correct? 

Since you left yourself a comment there to do an automatic integration of that environmental correction, thats what I am trying to achieve now and if you like I would be happy to share my
results once it is finished.

Hope you can help me out a little!

Regards

Harald

kriswiner, 2015-07-17:
I have a pretty good autocalibration routine that I have been using for some time. I plan to write up a little description and post it to my github wiki. But the gist is that the biases hardwired into the old versions of the code are now calculated on startup with the figure eight motion, as you say, just like the smart phone. I also have some scaling to round up the response surface too. I can share the code if you want, unless you want the pleasure of devising your own!

RayBreslin, 2015-07-17:
Hey Kris,

thanks for your fast reply - its good to see that I wasnt so far off track with my thought! ;)
Of course I will look into any description or code you provide, but right now its not that urgent,
because I still got a few other calibration tweaks to attend to. 

I would contact you as soon as I am done with the current work and in case you are interested I will provide the results too. :)

Regards

Ray

kriswiner, 2015-07-20:
FYI

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

RayBreslin, 2015-07-25:
Hey Kris,

thank you very much for your article about calibrating the magnetometer automatically. Was indeed insightful and put me to restructuring my magnetometer calibration method including the sensitivity readout, in order the access the magnetometer via I2C directly using the by-pass-enable-mode.
My normal application uses the I2C master mode to hook up the magnetometer to the slave registers and makes all data accessible via the external sensor registers. Works just fine!

I cant say why I had so much problems in the past with the by-pass-enable-mode and I aint happy with the current attempt either. Getting the magnetometer sensitivity is no problem and works just like in your code. However getting different magnetometer values during the figure-8-calibration is a problem. For some reason I am only getting new magnetometer values once after I start up my microcontroller and they stay constant no matter how I turn my board around. As you might guess I am not realy getting any useful hard iron correction and no soft iron correction at all.

Regarding the magnetometer calibration I am currently on hold, since I need to get this I2C issue cleaned first. Maybe I will post some small code snippets later on and you can point me in the right direction based on your own experiences. I kind of got the feeling I am missing a small puzzle piece at the moment... Anyway I hope I can get it solved soon!

Just a sidenote - my knowledge regarding the magnetometer calibration or more like the awareness of a needed hard and soft iron correction came from the following article:

http://www.sensorsmag.com/sensors/motion-velocity-displacement/compensating-tilt-hard-iron-and-soft-iron-effects-6475

Btw. I got a few other problems I would like to discuss with you, but the question is whether I should post them all within a single issue-topic here or handle them seperatly!? (f.ex. like passing -mz instead of mz into the magdwick filter or a deviation regarding the conversion of quaternion to euler angles)

Regards 

Ray

kriswiner, 2015-07-25:
You can send me PM at onehorse@earthlink.net.

You have to make sure to read the six mag data bytes and the seventh status register bits to get proper mag data out of the AK8963C. And I am assuming you have set the mag for continuous mode not the default one shot.

RayBreslin, 2015-07-27:
That is exactly my problem that left me clueless - I am configuring the continuous mode (either 8 hz or 100 hz) but I am only getting this one-shot-sample. When my application continues and configures the mpu9250 for master-mode-usage its all back to normal...

Sending you a PM about this with more details!

RayBreslin, 2015-07-28:
Ah well I underestimated the influence of considering the status register during readout. Thats why it seemed my MPU9250 was operating in one-shot-mode even though the 8 hz or 100 hz continuous-mode was configured...

Finally I can move on to the next step - I will keep you updated! :)

kriswiner, 2015-07-28:
Glad you solved your problem!

kokhua90, 2018-12-19:
May I know how to get the below +470, +120 , +125? calculated automatically?
Please advise. Thank in advance.

magbias[0] = +470.;  // User environmental x-axis correction in milliGauss, should be automatically calculated    
 magbias[1] = +120.;  // User environmental x-axis correction in milliGauss     
magbias[2] = +125.;  // User environmental x-axis correction in milliGauss

kriswiner, 2018-12-19:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration