kriswiner/MPU9250 #18: Unstable output coming from MadgwickQuaternionUpdate

multivac61, 2015-07-29:
Hey Kris! Very inspiring project you've got going on here, this theory is awesome and your implementation and explanations are spot on.

I have been using the MPU-9250 code with my Arduino UNO connected to a standard MPU-9250 chip via i2c. The raw output coming from the sensor is stable and makes sense (isn't oscillating and without any noticeable drifting). When I route the raw input into the MahonyQuaternionUpdate function I get quite reasonable roll, pitch and yaw data, without substantial drift (although a Gimbal lock often occurs, need to find a way to fix that?). However when I plug the raw data into MadgwickQuaternionUpdate the Euler angles are very unstable, especially the yaw which tends to oscillate from -180 to 180 and then from -180 again.

Have you ever encountered this problem before or know of any reason why this could happen?

kriswiner, 2015-07-29:
This will happen if the yaw is right at 180, since the heading range is
+/-180, going from -179.9 to -180, the next increment is +180, so it's an
unstable point. What happens if you move off this point to 170? You can
always change the behavior so the range goes from 0 to 360, which is more
common.

Kris 

(#18)

<https://github.com/kriswiner/MPU-9250/issues/18> .
<https://github.com/notifications/beacon/AGY1qsE5v2GVGXBxL01vJLzcXzzaX3seks5
oiSxGgaJpZM4FiR0H.gif>

multivac61, 2015-07-31:
All good, thanks for the info. I have however decided to use quaternions as outputs instead since then I don't have to deal with Gimbal lock at all..

I have one other question that can rarely be a new issue so I'll just ask it here instead: How could I get linear acceleration from the current data (quaternions + accelerometer)? I haven't found any decent information on the net about it, any thoughts/links would be well appreciated.