kriswiner/MPU9250 #19: doubt on function getMres()

bar123, 2015-08-10:
hey kris,
it would be helpful for me if you can tell the reason for multiplying 10 with (4912/32760), i got to know that 4912 is magnetic flux density and 32760 is the range of magnetometer.....i found a formula in net HuT = H * 4912/32760 where H is raw data, but why are you multiplying by 10.....is it a constant? i am using ak8975....should i use the same formula? can you please justify?

thanks in advanc

kriswiner, 2015-08-10:
I prefer to use milliG so I converted from microTesla.

<https://github.com/kriswiner/MPU-9250/issues/19> .
<https://github.com/notifications/beacon/AGY1qrBVvPf-CyRCffSPok9a-J9RHVgTks5
omHsKgaJpZM4Folnv.gif>

bar123, 2015-08-12:
thanks a lot kris, it did help me a lot, 
i am getting very random & abrupt values for roll, pitch & yaw, i have not used the micros() function & i have not done the averaging part, i have taken deltat =1,

instead my function looks like this 
   
 MahonyQuaternionUpdate(data.gx, data.gy, data.gz, gyrData[0]*0.007*(PI/180), gyrData[1]*0.007*(PI/180), gyrData[2]*0.007*(PI/180), data.compmx*mRes, data.compmy*mRes, data.compmz*mRes);
where gx, gy & gz are gravity values wrt x, y & z
compmx,compmy & compmz are compensated values from magnetometer obtained after calibration, 
moreover when the chip is flat i am not getting 0 values for roll pitch and yaw, what could be the reason?

kriswiner, 2015-08-12:
"data.compmxmRes, data.compmymRes, data.compmz*mRes"

Looks like you are multiplying by mRes in z but not x or y. Is this what you intend?

bar123, 2015-08-13:
no sir m multiplying all compensated magnetometer values of x,y & z with mRes
i.e data.compmx*mRes, data.compmy*mRes & data.compmz*mRes

kriswiner, 2015-08-13:
Why do you have the asterisk (or multiply) on the z value? Why is is
different?

<https://github.com/kriswiner/MPU-9250/issues/19#issuecomment-130521778> .
<https://github.com/notifications/beacon/AGY1qiL_KYHMgy4XqdEc0cjMYP51nN_lks5
onAu0gaJpZM4Folnv.gif>

bar123, 2015-08-13:
i am multiplying compensated values of x , y & z with resolution of magnetometer

kriswiner, 2015-08-13:
I understand what you are intending to do, I am simply asking why in the
cases of x and y you have i.e data.compmxmRes but for z you have
data.compmz*mRes. Why "*mRes" for the z case and not "mRes" or
data.compmzmRes like in the x and y cases? 

<https://github.com/kriswiner/MPU-9250/issues/19#issuecomment-130534416> .
<https://github.com/notifications/beacon/AGY1qnsK97d968IlkVX9lDrH-Ioupyktks5
onBixgaJpZM4Folnv.gif>

bar123, 2015-08-13:
here wen i commented it came out lik dat coz i had not given space b/w compmx & mRes.....instead of compmx * mRes it got printed as compmx*mRes

kriswiner, 2015-08-13:
Well, assuming the coding is correct, what kind of accelerometer and gyro bias calibration have you done? You ought to calculate the bias at rest for these sensors and subtract it off of the scaled data. Bias errors can cause the Euler angles to be off. And when you say the mag is calibrated, do you mean you have read and applied the fused ROM data from the AK8963 or did you also calculate the magnetometer bias correction? Here is a description of what could be done to more fully calibrate the magnetometer:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration
 
If you are using my code as a template and are calibrating the sensors properly, I can't understand how you could get non-zero pitch and roll, meaning 0 +/- 1 degree, with the sensor at rest on a flat table. Can you verify that you read 1 g when the sensor is oriented with the z-axis up, the y-axis up, the x-axis up, etc. The readings should be 0, 0, 1 with the 1 g for the up axis and the 0 for the other two to within 10 mg or less when calibrated. The gyro should read ~<10 mdps when at rest on all three axes. You can verify the mag reading by comparing the expected x, y, and z magnetic fields at your location.

bar123, 2015-08-13:
i have calculated the bias at rest for accelerometer &
also i have calculated magnetometer bias correction, i have not used the fuse rom data at all

kriswiner, 2015-08-13:
You really should use the fuse ROM data, these corrections can be as much as 20%! But their lack is not the cause of the roll/pitch error. What about gyro bias correction? This can be a large effect.

bar123, 2015-08-13:
i have used the gyro values directly....i have not done gyro bias correction

kriswiner, 2015-08-13:
Hold it still and average readings. Motionless bias is then subtracted from subsequent data. Same for accel and gyro.

bar123, 2015-08-13:
so sad yaar i tried dat also..... no luck still m getting non zero values when the chip is flat.....i seriously dono wat the problem is :'(