kriswiner/MPU9250 #21: Changing the Baud Rate Doesn't Change Serial.Print Speed!

Navid360, 2015-09-11:
Thanks for the Code, It works well!
The only problem is: When I change the baud rate (Serial.begin(...)) to a new value speed of Serial.print doesnt change! I mean there is no difference between Serial.begin(115200)  and Serial.begin(9600)  while serial printing.
Would you please let me know what I can do to have higher speeds of serial.print?
Thanks
Navid

kriswiner, 2015-09-11:
Hi Navid,

It depends on which platform you are using. If you are using an Arduino,
there should be a difference when you change the baud rate. If you are using
a Teensy, there won't be. The Teensy uses the fastest rate it can
independent of what you set the baud rate to be.

Kris

Speed! (#21)

<https://github.com/kriswiner/MPU-9250/issues/21> .
<https://github.com/notifications/beacon/AGY1qgyeri0Vz_tlXChmTusKnRN19obiks5
owtX8gaJpZM4F7wu5.gif>

Navid360, 2015-09-11:
Thanks for your reply.
I am using both Arduino  Pro mini but going from 9600 to 115200 still doesn't show difference in speed of serial printing. I though it might be because of the Delays in code while changing them didn't solve the speed! any suggestion?
Thanks
Navid

markussen145, 2017-07-03:
Hi,

In order for you to increase the speed of serial printing, you need to change this condition in MPU9250BasicAHRS.ino:

(line 438) - if(delt_t > 500) 

and those places in the code where delt_t is used. That variable measures the time since the last time the LCD had been updated. If you're not using an LCD (which has restrictions around frame rate), you can get rid of that condition, or you can use it to control the rate at which you serial print (i.e., change that 500 to something like 100).

I'd suggest you to keep using it if, as in my case, you're taking that data somewhere else where you use serial interrupts. That way, you won't be constantly going through interruptions and you'll leave time for code execution.