kriswiner/MPU9250 #23: Madgwick-Filter-ODR / Filtered-Magnetometer-Data

RayBreslin, 2015-10-03:
Hey Kris,

after some nicely spent vacation time I am now back to put all my effort into my little sensor-project and I do have a few more questions I would like to discuss with you.

I pretty much got your MPU9250-template set up for a CSR1010-MPU9250-combo-board and a smartphone as receiver/mc. For some reason I am not seeing any nice and stable roll, pitch and yaw angles which freaks me out...

Ok I have to admit that through my never ending researches for calibration approaches, madgwick fixes (s-parameters, inv-sqrt, ...) and other things I might have made too many adjustments and tweaks along the way! So now its „almost“ back to the start, taking your work as baseline and take it on from there (again)...

Now to the questions:

1. How is the function-call-frequency of your loop()-routine determined? Is there a start-skript somewhere or does it get called automatically? In other words how do you get/calculate a filter ODR of 145 hz?

2. If I am sampling my sensor data at 50 hz continuously, what kind of filter ODR would you propose? Should it always be at least 200 hz? (Currently doing just 50 hz when the data is coming in, which doesnt seem to do that well...)

3. What readout-frequency you use for your magnetometer? If I saw it correctly its always just 8 hz instead of 100 hz – only due to the readout-latency?

4. How does all the calibration data from the magnetometer calibration needs to be applied to the raw sensor data?

Heres my approach:

<pre><code>scaledData[Tag.MAGNETOMETER][i] = (rawData[Tag.MAGNETOMETER][i] * magResolution * currCalibrationData.magSensitivity[i] - currCalibrationData.magBias[i] * magResolution * currCalibrationData.magSensitivity[i]) * currCalibrationData.magScale[i]; </pre></code><pre><code>Legend:
i 	~ 	x-, y-, z-axis
currCalibrationData.magSensitivity[i] = (((rawSensitivityAdjustmentValue[i] - 128.0f) * 0.5f) / 128.0f + 1.0f);
magResolution = 10.0f * 4912.0f / 32760.0f;</pre></code>

I would very much appreciate your help!

Thanks in advance

Ray

kriswiner, 2015-10-03:
Hi Ray,

The Madgwick (and Mahony) filters are iterative, so ideally we would like 4
or 5 iterations on the filter solution for every new data sample. This means
if you run at a 200 Hz data rate, you want to run the sensor fusion at ~1
kHz or so. It is possible to force this rate but I just use a Teensy 3.X at
72 MHz and I get it "for free", meaning that in my sketch I just query
whatever fusion rate I get and it turns out to be > 1 kHz, which is fine.
The way the code is set up, I loop through and check every time whether
there is new data for each sensor. I so great, I use it in the next Madgwick
call. If not, I just call Madgwick with the old data and iterate again. The
code in practice will iterate on the same data 6 or 8 times which will
produce a converged solution. This is the advantage of using fast processor.
With a 16 MHz AVR processor running the fusion at 200 Hz you will not get a
converged solution, ever, unless you run with < 50 Hz data rates, which you
can do. But if your motion is "faster" than 50 Hz, you will experience
latency. The other parameter is the bandwidth. Generally, you want the
bandwidth to be 10 - 20% of the sample rate. For 200 Hz sample rates, I
usually use 40 Hz bandwidth (low-pass filter). This smooths out transient
and jitter to be expected with a 16-bit ADC. If you run with a sample rate
of 50 Hz, your bandwidth has to be ~10 Hz, which is kind of low. Lastly, the
sensors have to be sufficiently calibrated, as you know. The calibration
algorithm you show looks OK to me. How are you collecting and averaging the
mag bias?

Kris

<https://github.com/kriswiner/MPU-9250/issues/23> .
<https://github.com/notifications/beacon/AGY1qo_6D5OLDM54sf6PVJeBGKSoQ56Mks5
o4AGZgaJpZM4GIc1_.gif>

RayBreslin, 2015-10-03:
Oh thank you for this fast response! :)

The calibration data is collected just by your example coding, but the results seem a little confusing...

Heres an overview over the past 4 calibrations I did with my sensor-board:

<pre><code>
1.: accSen: 16384.0, accBias: [-361.0 / 402.0 / 418.0], gyrSen: 131.0, gyrBias: [113.0 / 112.0 / 4.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [-23.0 / 84.0 / -352.0], magScale: [1.0050505 / 1.2134147 / 0.8468085]
2.: accSen: 16384.0, accBias: [-342.0 / 412.0 / 454.0], gyrSen: 131.0, gyrBias: [117.0 / 112.0 / 5.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [-6.0 / 84.0 / -370.0], magScale: [0.98286605 / 1.2825203 / 0.831357]
3.: accSen: 16384.0, accBias: [-339.0 / 418.0 / 427.0], gyrSen: 131.0, gyrBias: [124.0 / 112.0 / -2.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [-6.0 / 94.0 / -374.0], magScale: [1.0046729 / 1.2356322 / 0.83657587]
4.: accSen: 16384.0, accBias: [-367.0 / 402.0 / 433.0], gyrSen: 131.0, gyrBias: [122.0 / 113.0 / 5.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [-6.0 / 94.0 / -373.0], magScale: [1.0046729 / 1.2356322 / 0.83657587]
</pre></code>

The bias values of acc and gyr still need to be divided by the sensitivity-factor, while the bias value of the magnetometer needs to be multiplied with the sensitivity-factor and the mag-resolution (like in my code example above). Is that calculation for the mag-bias even correct since you simply subtract it right away - if im not mistaken?

What worries me is the insanely huge scale-factor thats no where near your results for the scale biases of 0.96, 1.01, and 1.03...

kriswiner, 2015-10-03:
Hi Ray,

This all seems OK to me. By insanely huge do you mean 1.28 instead of 1.03?

Kris

(#23)

<https://github.com/kriswiner/MPU-9250/issues/23#issuecomment-145277301> .
<https://github.com/notifications/beacon/AGY1qrRG5a4Ym3zXi2XArqQGr-8Uqzs0ks5
o4BqXgaJpZM4GIc1_.gif>

RayBreslin, 2015-10-03:
Yes that was my feeling... or am I just exaggerating? :O

kriswiner, 2015-10-03:
It's a little larger than I would have expected. Are you executing the
figure eight motion when you accumulate mag bias data?

(#23)

<https://github.com/kriswiner/MPU-9250/issues/23#issuecomment-145278941> .
<https://github.com/notifications/beacon/AGY1qpahhU0xcvsYR0z1KDF8XqBas53xks5
o4B16gaJpZM4GIc1_.gif>

RayBreslin, 2015-10-03:
Thats precisely what I am doing and even after doing another 50 testruns, the picture remains the same. For some reason the mag-scale is somehow off limits...

Could it be an error coming from the hardware-setup itself? I use a CSR1010-board with coin-cell-holder and a MPU9250-drotek board which I strapped to its back. Maybe the coin-cell or the power-consumption during the data-sampling is causing this...

A second thought might be, that I do the figure-8-movement wrong without accounting for any possible rotation...

What would you say?

PS: I can produce a small 2 min MP4-video (6-8 MB) showing you exactly the results im seeing after each rotation. Would that help?

kriswiner, 2015-10-03:
What would be better is to plot x mag vs y mag or x mag vs z mag from the figure eight motion bias accumulation data to see the response surface. You should be able to confirm the bias offset and get some idea of how spherical the response surface is. The exact motion is not important as long as you are sampling enough of the phase space.I doubt it has anything to do with your hardware set up. Are you sure you are scaling the mag data properly?

RayBreslin, 2015-10-03:
Well my magnetometer calibration is done in 2 steps:

1. CSR1010-Board:
-> Collection of the raw magnetometer data during figure-8-motion
<pre><code>// Calculation of environmental x-, y- and z-axis correction in milliGauss
while(ii < ak8963_package_clbn_count)
{
    // Read the raw data from magnetometer including the status-register
    //(status-register important, otherwise data wont be updated)
    ak8963_calibration_check[12] = i2c_read_register(I2C_AK8963_ADDR, REG_MAGN_XOUT_L, &data[0], 7);

    mag_tmp[0] = (int16) (((uint16) data[1] << 8u) | (uint16) data[0]);
    mag_tmp[1] = (int16) (((uint16) data[3] << 8u) | (uint16) data[2]);
    mag_tmp[2] = (int16) (((uint16) data[5] << 8u) | (uint16) data[4]);

    for (jj = 0; jj < 3; jj++)
    {
        if(mag_tmp[jj] > mag_max[jj])
        {
             mag_max[jj] = mag_tmp[jj];
        }// if(mag_tmp[jj] > mag_max[jj])
        if(mag_tmp[jj] < mag_min[jj])
        {
             mag_min[jj] = mag_tmp[jj];
        }// if(mag_tmp[jj] < mag_min[jj])
    }// for (jj = 0; jj < 3; jj++)

    ii++;
}// while(ii < ak8963_package_clbn_count) 
</pre></code>
-> Preparation of the calibration data before transfer to smartphone (all done in integer-format since fpu is missing)
<pre><code>// Get hard iron correction (mag bias in counts)
magn_bias[0]  = (mag_max[0] + mag_min[0])/2;
magn_bias[1]  = (mag_max[1] + mag_min[1])/2;
magn_bias[2]  = (mag_max[2] + mag_min[2])/2;
// Get soft iron correction estimate (max chord length in counts)
magn_scale[0]  = (mag_max[0] - mag_min[0])/2;
magn_scale[1]  = (mag_max[1] - mag_min[1])/2;
magn_scale[2]  = (mag_max[2] - mag_min[2])/2;
</pre></code>

-> Maybe this preparation step causes some error, since im losing the accuracy while operating only on integer!?

2. Smartphone:
--> Calculation of proper mag-scale and mag-sensitivity-factor
<pre><code>avgRawMagScale = rawMagScale[0] + rawMagScale[1] + rawMagScale[2];
avgRawMagScale /= 3.0f;
newCalibrationData.magScale[0] = avgRawMagScale / rawMagScale[0];
newCalibrationData.magScale[1] = avgRawMagScale / rawMagScale[1];
newCalibrationData.magScale[2] = avgRawMagScale / rawMagScale[2];
newCalibrationData.magSensitivity[0] = (((rawMagSensitivity[0] - 128.0f) * 0.5f) / 128.0f + 1.0f);
newCalibrationData.magSensitivity[1] = (((rawMagSensitivity[1] - 128.0f) * 0.5f) / 128.0f + 1.0f);
newCalibrationData.magSensitivity[2] = (((rawMagSensitivity[2] - 128.0f) * 0.5f) / 128.0f + 1.0f);
</pre></code>

In general im thinking that any preparation of the bias data done on the CSR1010-board is causing me a loss of accuracy.

Take a look at this uncalibrated data (zero-motion) for example:
<pre><code>acc (-0,012;0,144;0,989) / gyr (0,976;0,549;0,610) / mag (67,473;-10,496;328,366)
acc (-0,012;0,142;0,992) / gyr (1,037;0,488;0,549) / mag (47,980;-5,998;332,864)
acc (-0,012;0,143;0,992) / gyr (0,976;0,549;0,549) / mag (62,974;-17,993;338,862)
acc (-0,013;0,142;0,995) / gyr (0,976;0,366;0,610) / mag (56,977;-8,996;323,868)
acc (-0,013;0,143;0,995) / gyr (0,976;0,854;0,610) / mag (68,972;-11,995;341,861)
</pre></code>

Now with calibration data (not from same measurement - just in principle):
<pre><code>accSen: 16384.0, accBias: [-235.0 / 411.0 / 423.0], gyrSen: 131.0, gyrBias: [132.0 / 120.0 / 4.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [-23.0 / 85.0 / -345.0], magScale: [0.986532 / 1.2208333 / 0.8567251]
acc (-0,003;0,001;1,010) / gyr (-0,154;-0,001;0,030) / mag (-128,650;28,630;298,541)
acc (-0,001;-0,001;1,011) / gyr (-0,154;-0,001;0,030) / mag (-135,699;46,249;301,511)
acc (-0,004;0,001;1,008) / gyr (-0,215;0,060;-0,092) / mag (-123,363;22,023;308,938)
acc (-0,004;0,000;1,011) / gyr (-0,215;-0,001;0,030) / mag (-121,600;28,630;310,423)
acc (-0,002;0,001;1,010) / gyr (-0,154;-0,001;0,091) / mag (-123,363;35,237;314,879)
</pre></code>

The accelerometer data seems to reach your expected interval of 0,0,1 - depending on the axis - with an average deviation of +-10mg. For the gyro theres still some bigger bias error of more then +-60 mdps present - not as your expected +-10 mdps deviation...

kriswiner, 2015-10-03:
Yes, the gyro is higher than I would expect, but more worrying is the very large (factor of two) changes in your mag reading. In general I would recommend the following construction of the signed integer data:

temp[0] = (int16_t) (  ( (int16_t) data[1] << 8 )  |  data[0]);

Here's how I typicall do it:

destination[0] = ((int16_t)rawData[0] << 8) | rawData[1] ;

This works well even without the leading (int16_t) as long as the destination array is already defined as int16_t.

No need to cast the LSByte and you should use int16_t not uint16_t. Don't know if this is your problem but integer cast problems are common.

RayBreslin, 2015-10-04:
Hm... I will definitely check that out!

The uint16_t comes from the underlying i²c-function I2cRawRead() which is defined as the following:

<pre><code>/*----------------------------------------------------------------------------*
 *  I2cRawRead
 */
/*! \brief Read data from the I2C bus, ACKing received bytes and NACKing the
 *   last byte.
 *
 *  This command assumes the I2C slave device has been put into a state where
 *  it is ready to transmit bytes. The function will block until the read
 *  operation has completed, and all bytes have been read.
 *
 *  \param data     Pointer to storage for the bytes that are read.
 *  \param length   The number of bytes to read.
 *
 *  \return Status of operation
 */
/*---------------------------------------------------------------------------*/
extern sys_status I2cRawRead(uint8* data, uint16 length);
</pre></code>

So I do think doing all bit-operations with (uint16_t)-casts is correct and only in the last step when
saving into the int16_t-result-variable I do the (int16_t)-cast...

Ill keep you updated!

RayBreslin, 2015-10-05:
Hey Kris,

I have been running tests for the last 2 days now and I kind off hit a wall...

Some of my adjustments I tried to do are listed below:

- Adjustment of sensor-samplerate (actually meaning transmission-rate) from 50 Hz to 100 Hz
  --> Transmission-rate was switched back to 50 (stable) though, due to latency-errors and package-loss 
  --> Sampling-rate internally follows your example:
        - 1 kHz for accelerometer/gyro (with SMPLRT_DIV-factor 5) and 8 Hz magnetometer
- Switched from Master-Mode during normal data-readout back to Bypass-Mode (like during configuration)
- Included Madgwick-Fix and Fast-Inv-Sqrt from Diydrones-Blog
- Adapted the uint8-int16-Conversion as you proposed
- Adapted the beta-Factor from 0.6045 (your example) up to 10 (even 20 in a few cases) 
  --> 
- Increased magnetometer-sample-rate from 8 Hz to 100 Hz

Now heres the picture (taken from 4 test-runs I did earlier):

(Apologies for the amount of text here, but for some reason the upload of my txt-files failed!)

Legend:
<timestamp>;<acc_x>;<acc_y>;<acc_z>;<gyr_x>;<gyr_y>;<gyr_z>;<mag_x>;<mag_y>;<mag_z>;<pkg_id>

#1. Test
<pre><code>accSen: 16384.0, accBias: [-112.0 / 949.0 / 374.0], gyrSen: 131.0, gyrBias: [138.0 / 110.0 / 0.0], magSen: [1.1914063 / 1.203125 / 1.15625], magBias: [51.0 / 198.0 / -260.0], magScale: [0.8894646 / 2.2008548 / 0.70355195]
10,664948;-0,001;0,002;1,010;-0,139;-0,108;-0,061;-88,980;-170,720;284,196;  1
10,678642;-0,001;0,002;1,007;-0,017;-0,047;0,061;-88,980;-162,780;281,757;  2
10,714985;0,000;0,000;1,007;-0,078;-0,047;0,061;-92,158;-154,839;289,075;  3
10,717145;-0,001;0,002;1,011;-0,078;-0,047;0,000;-79,446;-154,839;281,757;  4
10,717801;0,000;0,001;1,007;-0,261;0,014;-0,061;-79,446;-202,482;284,196;  5
10,761346;0,001;0,001;1,010;-0,139;0,014;0,061;-85,802;-162,780;276,878;  6
10,762520;-0,000;0,000;1,007;-0,139;-0,108;0,061;-90,569;-127,048;278,098;  7
10,809990;-0,003;0,002;1,013;-0,139;-0,169;0,061;-93,746;-182,631;278,098;  8
10,816403;-0,000;0,000;1,012;-0,200;-0,047;0,061;-71,502;-158,809;285,416;  9
10,857050;-0,002;0,001;1,007;-0,200;0,014;-0,061;-88,980;-162,780;279,317; 10
10,863018;-0,001;-0,000;1,012;-0,139;0,014;0,061;-87,391;-182,631;282,977; 11
10,866376;0,001;0,000;1,008;-0,200;0,014;0,000;-90,569;-174,690;287,855; 12
10,905561;0,002;0,000;1,010;-0,139;0,014;0,000;-79,446;-186,601;284,196; 13
10,906882;0,000;0,002;1,008;-0,139;0,014;0,000;-95,335;-194,542;276,878; 14
10,955384;-0,002;0,000;1,005;-0,139;0,014;-0,061;-92,158;-154,839;276,878; 15
10,957602;-0,002;0,002;1,010;-0,078;-0,047;0,000;-73,090;-170,720;284,196; 16
10,958234;0,000;-0,001;1,011;-0,078;-0,047;0,061;-82,624;-138,958;281,757; 17
11,003816;-0,002;0,001;1,006;-0,017;-0,047;-0,122;-74,679;-174,690;285,416; 18
11,005399;0,000;0,001;1,009;-0,078;-0,047;0,061;-88,980;-162,780;293,954; 19
11,052965;-0,000;0,002;1,010;-0,017;0,014;0,000;-87,391;-166,750;287,855; 20
11,054173;-0,001;-0,000;1,007;-0,078;-0,047;0,122;-98,513;-186,601;284,196; 21
11,054871;-0,001;0,000;1,007;-0,078;0,075;0,000;-95,335;-154,839;284,196; 22
11,102535;-0,002;-0,000;1,009;-0,139;-0,047;-0,061;-82,624;-162,780;286,636; 23
11,103263;0,000;0,001;1,007;-0,139;0,014;0,000;-96,924;-134,988;280,537; 24
11,152168;-0,001;-0,001;1,010;-0,017;0,014;-0,061;-82,624;-170,720;281,757; 25
11,153137;-0,001;0,001;1,006;-0,078;0,014;0,000;-87,391;-142,929;282,977; 26
11,199829;0,001;0,000;1,008;-0,200;0,014;-0,061;-79,446;-154,839;284,196; 27
11,203604;-0,001;0,002;1,008;-0,078;-0,047;-0,061;-84,213;-150,869;282,977; 28
11,213362;-0,001;0,002;1,009;-0,139;-0,047;0,122;-84,213;-158,809;282,977; 29
11,247793;0,000;0,002;1,011;-0,200;-0,047;0,000;-79,446;-154,839;289,075; 30
11,249409;-0,001;0,001;1,009;-0,078;-0,047;-0,061;-95,335;-178,661;286,636; 31
11,298275;-0,000;0,001;1,009;-0,139;-0,047;0,000;-66,735;-146,899;281,757; 32
11,300446;-0,000;0,002;1,008;-0,139;-0,047;0,000;-93,746;-166,750;280,537; 33
11,301467;-0,001;0,000;1,005;-0,078;-0,047;0,000;-77,857;-150,869;290,295; 34
11,344402;-0,001;0,001;1,009;-0,139;-0,047;-0,061;-95,335;-131,018;276,878; 35
11,345439;-0,000;0,003;1,005;-0,139;-0,047;0,061;-84,213;-174,690;278,098; 36
11,393688;-0,001;0,001;1,010;-0,200;-0,047;0,061;-92,158;-146,899;286,636; 37
11,395046;0,000;0,000;1,008;-0,139;0,014;-0,061;-73,090;-138,958;286,636; 38
11,395289;-0,001;0,000;1,011;-0,139;-0,047;0,000;-73,090;-131,018;276,878; 39
11,443390;-0,001;0,000;1,009;-0,139;0,014;0,061;-69,913;-186,601;284,196; 40
11,444089;-0,001;0,001;1,007;-0,078;0,075;-0,061;-82,624;-170,720;284,196; 41
11,491633;0,000;0,000;1,009;-0,139;0,014;0,000;-76,268;-186,601;291,515; 42
11,494356;0,000;-0,000;1,009;-0,139;0,014;0,122;-88,980;-146,899;279,317; 43
11,495200;0,000;0,000;1,007;-0,139;0,014;-0,122;-79,446;-178,661;286,636; 44
[... 366 lines cut ...]
230,868652;-0,005;0,000;1,007;-0,085;-0,062;0,023;-74,556;-157,665;303,313; 45
230,915970;-0,002;0,002;1,010;-0,207;-0,062;-0,038;-68,709;-187,697;305,985; 46
230,917389;-0,004;0,001;1,009;-0,085;-0,001;0,023;-80,404;-172,681;305,985; 47
230,964645;-0,002;0,001;1,009;-0,085;-0,062;-0,160;-73,094;-153,912;307,321; 48
230,965164;-0,004;0,001;1,007;-0,207;-0,001;-0,038;-67,247;-161,419;299,304; 49
230,966782;-0,003;0,002;1,008;-0,207;-0,001;-0,038;-71,633;-142,650;292,623; 50
</pre></code>

Observations:
- Accelerometer-data looks perfect (0,0,1)-stability for all axis (also after turning or waving the sensor) with an error of +-10 mg
[... cut ...]
From my perspective all the calibrated sensor data is quite stable and resets perfectly even after heavy motion and over longterm measurements. Still when doing the quat2euler-conversion I am seeing a highly unstable yaw-angle (realy the worst of all 3), a fairly stable roll-angle and a slow-responsive pitch-angle.

Here are the links for 2 videos I made, that document my dilemma:

MPU9250-Raw-Data-Plot: http://www.vidup.de/v/qlZQI/
MPU9250-Euler-Angle-Plot: http://www.vidup.de/v/jGZdA/

Cant wait to hear your statement to that! ;)

Regards

Ray

kriswiner, 2015-10-05:
On the mag data you have printed out, is the ~300,000 z-axis number supposed in micro Gauss so it is ~300 mG? If so, then it's not too bad and about what I would expect. If you use the NED convention, you need to pass the Madgwick (or Mahony) filter the data that corresponds to the proper orientation. So if the + z-axis of the mag is pointing up (which is is IIRC), then you have to pass -Mz to the filter (since Down is - z). If you want the x-axis to define North or Up (which I usually do), the you have to pass the filter Mx on the first slot, and then either My or -My depending on which direction is East. In other words, you have to make sure the sensor axes align with your orientation convention, for all three sensors. So if the accel/gyro x-axis is defined as North and the mag x-axis points in the East direction, you have to pass the filter swapped Mx and My values to maintain consistency. This might be the cause of the inconsistencies in yaw behavior you are seeing. Although it is a bit hard for me to tell what is happening in the videos.

I hope this helps.

Kris

RayBreslin, 2015-10-06:
Awwww... I totally forgot to add a description for the videos:

(1) MPU9250-Raw-Data-Plot:
  - Accelerometer-View --> turned sensor-board around x-, y- and z-Axis just to verify each axis gets a stable (0,0,1)-behaviour
  - Gyroscope-View --> turned sensor-board in 90°-steps from 0 to 360° around each axis to verify a stable reset to (0,0,0)
  - Magnetometer-View --> some left-over from gyroscope-test, then put a simple CR2032-coin cell on top of the sensor to cause a magnetic distortion and then removed it to verify a stabilization afterwards

(2) MPU9250-Angle-Data-Plot:
- Test-pattern looks like this: 
  --> static behaviour (on edge of desk) - with heavy yaw-angle jitter
  --> turned sensor in 90°-steps around x-axis - roll- and pitch-angle very stable, but yaw-angle jitter (less then in static test)
  --> turned sensor in 90°-steps around y-axis - pitch-angle pretty stable in every step, but roll- and yaw-angle jittered heavily specially for 90°- and 270°-pitch angle
  --> turned sensor in 90°-steps around z-axis - roll- and pitch-angle stable obviously, but yaw-angle does some wierd jittering again

For orientation I use the one from the InvenSense-Datasheet for the MPU9250 (the same like you do):
<pre><code>gx = (double) scaledData[1][0] * deg2radFactor;
gy = (double) scaledData[1][1] * deg2radFactor;
gz = (double) scaledData[1][2] * deg2radFactor;
ax = (double) scaledData[0][0];
ay = (double) scaledData[0][1];
az = (double) scaledData[0][2];
mx = (double) scaledData[2][1]; // Switch y --> x
my = (double) scaledData[2][0]; // Switch x --> y
mz = (double) scaledData[2][2];

filter_iteration = 0;

while(filter_iteration < 20)
{
	orientationFilter.updateAHRSmarg(gx, gy, gz, ax, ay, az, 0.0d, 0.0d, 0.0d);
	//orientationFilter.updateAHRSmarg(gx, gy, gz, ax, ay, az, mx, my, mz);
}
</pre></code>

I will setup a testbench in Matlab now till the end of the week, to provide an easy-to-use-gui for the fast analysis of all the raw-sensor-data, quaternion-variables and euler-angles.

In the meantime im testing your mahony-filter with the same data, doing a check on the correct data-format-input of the madgwick-algorithm (f.ex. passing uT instead of mG), adding a digital-low-pass-filter for the magnetometer-data and checking all the conversion-process-chain for both raw-data and calibration-data...

Somewhere has to be the explanation, why I dont get such stable results like you described in your tests!

kriswiner, 2015-10-07:
Hi Ray,

I am assuming you changed the ordering of the accel and gyro data passed to
your marg filter and are properly accounting for same. This could mess you
up if you haven't!

Kris

(#23)

<https://github.com/kriswiner/MPU-9250/issues/23#issuecomment-145904938> .
<https://github.com/notifications/beacon/AGY1quyB5cCs4xaqGkFHkQQuzNp5RljHks5
o4-NXgaJpZM4GIc1_.gif>

RayBreslin, 2015-10-07:
Yes I did that - the one case where I followed madgwicks original coding! ;)

<pre><code>//orientationFilter.updateAHRSmarg(ax, ay, az, gx, gy, gz, 0.0d, 0.0d, 0.0d);
//orientationFilter.updateAHRSmarg(gx, gy, gz, ax, ay, az, mx, my, mz);
orientationFilter.MahonyQuaternionUpdate(gx, gy, gz, ax, ay, az, mx, my, mz);
</pre></code><pre><code>public void updateAHRSmarg(double gx, double gy, double gz, double ax, double ay, double az, double mx, double my, double mz)
public void MahonyQuaternionUpdate(double gx, double gy, double gz, double ax, double ay, double az, double mx, double my, double mz)
</pre></code>

Also took in your mahony-implementation, but that one - unchanged and getting same data like madgwick - shows absolutely unpredicted behaviour! But I dont want to open a new problem there, until I fixed the madgwick-issue entirely...

Just thought hey since I am assuming you can easily exchange madgwick and mahony in your code vice-versa with approximately the same results, why dont I try that to at least exclude any false portation error... kinda didnt went as planned!

But for now I am relying entirely on my matlab testbench and I wont do any side-analysis in the meantime.

Totally appreciate your help so far, but I suppose I got to fix my coding mess here first and provide truely investigative data, that can actually help unriddling this mystery...

Will get back to you in a few days to inform you on the progress!

PS: Just 1 more question... you do multiply your calculated magBias with the magResolution and magSensitivity first before you subtract it from your rawMagData right?

kriswiner, 2015-10-07:
Hi Ray,

Yes, it is a little hard for me to follow all the changes you have made so I
can't quite tell where the problem, if any, might lay. I hope your further
investigations provide a smoking gun to identify what the issue might be.
The mag data you showed a few days ago looks somewhat strange to me and this
might be where things are breaking down.

I make sure that whatever biases I subtract off of the magnetic field has
all the same calibrations and scaling so I am subtracting properly scaled
milligauss biases from properly scaled milligauss fields.

Kris

(#23)

<https://github.com/kriswiner/MPU-9250/issues/23#issuecomment-146091210> .
<https://github.com/notifications/beacon/AGY1qr2itEk32loONBcMLJI5CI20XGUMks5
o5LWGgaJpZM4GIc1_.gif>

RayBreslin, 2015-10-07:
Im sorry that I caused this much confusion...

Already thought about getting myself a teensy 3.2-board where I can attach one of my MPU9250 to, which then allows me to run your scetches MPU9250BasicAHRS.ino or MPU9250BasicAHRS_t3.ino entirely unchanged (hopefully). Just to get some kinda similar baseline or reference!

In general I felt that my approach wasnt too bad so far, since roll and pitch behaved pretty nicely just now the yaw... 

Before I can draw any more conclusions or ask the right questions I got to improve my knowledge on basic sensor-functionality too, since right now I couldnt say in detail which sensor takes what part in actually getting to euler angles!

kriswiner, 2015-10-07:
Don't be sorry... this is a complicated area and it takes a while to get
your arms around it. I've been at it for almost two years and still don't
know enough!

I'll take a look at that article this weekend.
Kris

(#23)

<https://github.com/kriswiner/MPU-9250/issues/23#issuecomment-146097009> .
<https://github.com/notifications/beacon/AGY1ql0uwb2lphgVZ7YB7xqZP4NHBeb9ks5
o5LuFgaJpZM4GIc1_.gif>