kriswiner/MPU9250 #26: When the IMU rotates about one axis, angles about other two axes change too?

britam, 2015-10-08:
Hi, i modify and run the code of the sketch of kriswiner, and i have the problem what when i rotate the IMU (i use the MPU9250 and Arduino Pro mini 3.3 volts 8 Mhz) about one axis  angles obout the other two change too. HOW CAN I SOLVE THAT????? and what happend that to me?

thaks!!!!

kriswiner, 2015-10-09:
This can happen if your fusion rate is small compared to your sample data rate (likely with the 8 MHz Pro Mini) and/or your sensors are not properly bias calibrated. See the MPU6050 Wiki for more info about these.

britam, 2015-10-09:
i'm gonna change my bad redacted question, indeed the angles seems good calculated, i was thinking what each angles are represented independent of the other two angles. the general function i see in the program is what when i move my protoboard in a horizontal plane, the yaw angle change correctly, but the other two angles (pitch and roll) dont change (that is what i want), but when i rotate in the other planes where i only wish change only pitch, or only roll .... all the angles change. so, my question is: that is the real behavior of the sketch ..... Thanks for your fast answer and for your code :)

kriswiner, 2015-10-09:
If you rotate about only one axis at a time, you should see only changes in
that one parameter, i.e., pitch or roll. If you have a compound angle then
rotations will appear to change all the values. In other words, if you point
up at 45 degrees and then only rotate about the pitch angle, you will see
heading change too. This is expected and a result of the Euler angle
approximation to orientation.

other two axes change too? (#26)

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-146919701> .
<https://github.com/notifications/beacon/AGY1qlcs79RHarST7AdaRVlltkjTCx52ks5
o5-I0gaJpZM4GLv6x.gif>

britam, 2015-10-09:
Ok, i understand now..... but i need what the three angles moves in a independent way, maybe the transformation is not good enough for what i looking for.
I know what quaternions angles are the absolute format, how can i transform them to a format what belongs to me calculate the movement of a body in the space with three angles what move in an independent way like the navigation angles or Tait–Bryan angles

kriswiner, 2015-10-09:
Euler angles are Tait-Bryan angles. The problem is these angles are
appropriate for spherical surfaces, not Euclidean space.

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-146948292> .
<https://github.com/notifications/beacon/AGY1qskNxW5PsCFNdQ1yk5P0_cuP47Q5ks5
o5_fngaJpZM4GLv6x.gif>

britam, 2015-10-09:
understood, thanks!!!! ;) if i make an advance with my proyect i'll post it here.

britam, 2015-10-29:
Hi Kris, 
Finally, i now can read the three angles properly. But my last problem is what when i use the madgwick algorithm the angles are updated slowly.
What parameters i must change to make the filter works properly, or  what filter you recomend to me use the mahoney or the madgwick.
Regards!!!!

kriswiner, 2015-10-30:
What did you have to do to get the angles to behave? Some others report this
behavior and I see it myself often, did you change the code somehow?

You may use either Madgwick or Mahony. The best way to get faster update
speeds is to increase the speed of your controller. Are you using a Teensy?
If so you can run it at 96 MHz.

Kris

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-152338922> .
<https://github.com/notifications/beacon/AGY1qsP-7dCa6IhPnXzYmjMve5Q_sWxmks5
pAoy6gaJpZM4GLv6x.gif>

yerzhik, 2016-01-21:
Having the same issue. When sensor is rotated along Z axis while being prependicular to the floor (Z axis is prependicular to floor) then its perfectly rotating and angles showing correct values.
But when I start rotating along x or y axis then other two axis starts rotting too. 
So the question is were you able to resolve this problem? What is the reason?

kriswiner, 2016-01-21:
I always have the same question when people ask me about this: how are you
calibrating your sensors? In particular, how are you calibrating your
magnetometer?

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-173448569> .
<https://github.com/notifications/beacon/AGY1qhWU6CQdUoKrtBtzCjaRM6p8ELXPks5
pcFa7gaJpZM4GLv6x.gif>

yerzhik, 2016-01-21:
For acceleration I estimate the bias and set it. There is a DMP code for that (motion application library from invensense)
And I set up the DMP to calibrate the GYRO. Which does it (as it's written) for 8 seconds and the MPU should be set so that pins are up or down (initial position). Does that makes sense?

kriswiner, 2016-01-21:
How are you calibrating the mag? If it is uncalibrated you will just get
nonsense results.

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-173483744> .
<https://github.com/notifications/beacon/AGY1qjwp95IQNn9AjeuNmL4RwAAesF0Dks5
pcIGOgaJpZM4GLv6x.gif>

yerzhik, 2016-01-22:
I'm not, but from Invensense documentation on offset registers I see only Accel/Gyro offset registers (for setting up bias). But how would I tell DMP to use offsets for magnetometer?

kriswiner, 2016-01-22:
I don't know, but this is one of many reasons I do not use the DMP.

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-173823904> .
<https://github.com/notifications/beacon/AGY1qjdh9mOpQyktQj4uSgbZfdL073NJks5
pccM0gaJpZM4GLv6x.gif>

yerzhik, 2016-01-26:
I found out that the issue was in that the gyroscope was still calibrating
while I started rotating it. Waiting for 15 seconds makes much better
results. But issue now happens only if I rotate only along one axis until
90 degree. Is it related with euler angles or it shouldn't be like that at
all? Getting euler angles from quaternions and didn't expect that to happen.

kriswiner, 2016-01-26:
You should expect some cross talk between yaw , pitch and roll. It's hard to
get away from unless you specifically correct for it.

https://github.com/notifications/beacon/AGY1qjdh9mOpQyktQj4uSgbZfdL073NJks5

<https://github.com/kriswiner/MPU-9250/issues/26#issuecomment-174839621> .
<https://github.com/notifications/beacon/AGY1qgNP4uK_ZqQwxPltdCGAHkczpl2Sks5
pdvt1gaJpZM4GLv6x.gif>

yerzhik, 2016-01-26:
Thank you, I think its solved for me now

JuanPabloGomez, 2016-11-01:
I would like to answer Kris Winer's question because I also experienced changes in yaw/heading with changes in the pitch/roll.  

> What did you have to do to get the angles to behave? Some others report this
behavior and I see it myself often, did you change the code somehow?<

I believe the problem happens when sending the mx, my and mz raw values to the Madgwick/Mahoney quaternion function. In my MPU9150, the accelerometer/gyro and magnetometer have difference axis. 
X in the accel corresponds to Y in the magnetometer. 
Y in the accel corresponds to X in the magnetomer. 
Finally, Z corresponds to -Z. This last one is important because the code I have seen accounts for the changes in X and Y but not for the change in direction in Z, as observed below: 

`MadgwickQuaternionUpdate(ax, ay, az, gx * PI / 180.0f, gy * PI / 180.0f, gz * PI / 180.0f,  my,  mx, mz);`

As you can see, the my and mx are correctly swapped, but there is no change in the sign of mz. The correct code would look like this: 

`MadgwickQuaternionUpdate(ax, ay, az, gx * PI / 180.0f, gy * PI / 180.0f, gz * PI / 180.0f,  my,  mx,  **-mz**);`

This solved the problem for me. Hope it helps others.

jenextech, 2018-01-09:
Hello,
     I need to know about declination-angle,which is useful in measuring YAW.
Any formula to find that angle for particular location?

Thanks in advanced.

kriswiner, 2018-01-09:
This is measured not calculated. Look it up for your location.

jenextech, 2018-01-11:
Okay.
      I have checked it for my location.
      It looks like as i have attached picture.

[image]

     This is in different formate as we put declination angle's value in code.
Can you tell me that from above image what should declination angle instead of we have used in code (yaw   -= 23.45f; // Declination at Danville, California is 13 degrees 48 minutes and 47 seconds on 2014-04-04)

Please,confirm this one.

kriswiner, 2018-01-11:
0.49 degrees?

jenextech, 2018-01-12:
Hell Kris,
  When i tried to get declination angle,it shows as like image.

So,what should i put value of it instead of 23.45.

kriswiner, 2018-01-12:
The declination is a propery of your location on Earth, not the sensor. For
your area, 0.49 degrees....

jenextech, 2018-01-12:
Okay.
    Thanks for fast response.
   So,now i will check all data by considering declination angle as 0.49.
   Right?

liutianxin0722, 2020-09-22:
> Hi Kris,
> Finally, i now can read the three angles properly. But my last problem is what when i use the madgwick algorithm the angles are updated slowly.
> What parameters i must change to make the filter works properly, or  what filter you recomend to me use the mahoney or the madgwick.
> Regards!!!!

Hello britam！
I meet the same problem that when I rotate about X axis，the values in Y axis also change. I use Mahoney algorithm to obtain eular angles from a 6-axis MPU 6050 sensor.
How did you solve the problem? 
Thank you

kriswiner, 2020-09-22:
You need to send the sensor data to the filters as NED, so aN, aE, aD, gN,
gE, gD to get consistent results.