kriswiner/MPU9250 #27: Hey Kris,

dtrain04, 2015-11-01:


dtrain04, 2015-11-01:
Hey Kris,

I am pretty new to all this and I just want to see the calibrated data on the serial monitor and not the screen because I do not own it.  Is there a simple way to edit the code so that I can just see the output on the monitor or should I invest.  This will eventually be an input into matlab.  Also, I have an Uno so will this code still work? I am pretty new to arduino prototyping and prototyping in general.  Thanks for the help!

kriswiner, 2015-11-01:
You can comment out the display commands if you just want serial monitor
output. As long as you don't fry your MPU9250 with 5 V (use a logic level
translator or use a breakout that does this for you) this code should work.
You might have to change a few lines to conform to the I2C wire library
protocol.

Kris

<https://github.com/kriswiner/MPU-9250/issues/27#issuecomment-152851920> .
<https://github.com/notifications/beacon/AGY1qpq-A8JIgMxDoPUCOzBMy2j2vmYsks5
pBlfmgaJpZM4GZvIy.gif>

dtrain04, 2015-11-01:
Okay. So instead of commenting everything, I should change them to serial.print instead so I can still view everything? And changing a few lines in the for the wire protocol? You mean for transmission purposes to the serial monitor?  Also, should I use serial resistors with my UNO to the SCA and SCL pins? Sorry that Im asking silly questions. I am just running out of time! Thanks again Kris for the help!

kriswiner, 2015-11-01:
If you are not using a Nokia 5110 display, you don't need the lines to
display data to it.

You can just use Serial.ptint to spew data to the serial monitor with
teraterm or whatever terminal emulator you are using, or just the Arduino
IDE, which has a serial terminal. The I2C pullups depend on the breakout
board you are using. If you have them on the board, you don't need to add
any, if you have them elsewhere on the I2C bus, you also do not need to add
any, and you might have them on the UNO, I don't know. When you measure the
voltage on the SDA line, what do you read? Should be either 3V3 or 5 V.

No current limiting resistors are necessary between board SDA/SCL and UNO
A4/A5.

<https://github.com/kriswiner/MPU-9250/issues/27#issuecomment-152861823> .
<https://github.com/notifications/beacon/AGY1qvvWjjRRkPnSuN_1rpumrzZ0hYEUks5
pBnEJgaJpZM4GZvIy.gif>

dtrain04, 2015-11-01:
I burnt my last chip so I have to pull one together and get working on the code. I'll get back to you tonight if I have any problems! Thanks!

dtrain04, 2015-11-02:
yeah those resistors fried my board and everything is working now!  Apparently, the resistors caused the current to go backwards.  Thanks for your help!

dtrain04, 2015-11-08:
Hey Kris, to further our discussion previously I had a few extra questions.  One, is the accelerometer data filtered or should I develop my own kalman filter based on the gyroscope and accelerometer to get position data. I know that the madgwick algorthim is included for euler angles but nothing there based on position. Any thoughts?

Secondly, this is a wireless sensor application and I want to use wake up protocols, one to save battery and two, to prevent degradation of components.  Should I use the master I2C register to put the sensor to sleep or how would I approach this?

Thanks Kris!

kriswiner, 2015-11-08:
Yes, if you want position data from the IMU you will have to perform the
integrations yourself; most if not all existing sensor fusion solutions are
intended for absolute orientation not position.

You can always put the accel and or gyro in low power modes. You can turn
off the gyro and put the accel in a 1 Hz low power mode and have it wake up
the other sensors when it detects significant motion (user defined).

Kris

<https://github.com/kriswiner/MPU-9250/issues/27#issuecomment-154767113> .
<https://github.com/notifications/beacon/AGY1qvrSh5Y6h-3MRmjBqFiYfKv2Su7vks5
pDoyngaJpZM4GZvIy.gif>

dtrain04, 2015-11-08:
Which registers do i need to access to configure the power modes?  Did your Madgwick algorithm include a kalman filter of sorts? I noticed that you use a filter with the sensitivity and include the trim values from manufacturer.  What kind of filter did you use or did you just modify the values based on the self test and sensitivity?

dtrain04, 2015-11-08:
To be honest, when the mpu sensor is sitting on a table top, I am getting yaw, pitch ,and roll values to be -17, -32, and very random negative values for the roll from -5 to -15. Thoughts on what could be causing this?

dtrain04, 2015-11-08:
If you would like, I can post the changed arduino code and matlab code and see if you find something interesting.

bvanharen, 2015-12-09:
Hello, 

I'm just getting started with this code and must say it works great so far! I'm using it with a nodeMCU0.9 for now. 

I'm trying to get about the same thing dtrain04 is trying to get, position data. I know how to perform the integrations but am not very experienced in getting an absolute acceleration with respect to the ground from all orientation data I'm getting. 

My goal is to make a height estimator for a quadcopter, which should also work if the quadcopter is flying up at an angle, or if for example the sensor is mounted at an angle. 

Have you figured out how to get the position or absolute acceleration dtrain04?

Regards, 
Bob

kriswiner, 2015-12-09:
Hi Bob,

Why don't you use the altimeter on the board. If you use a simple Kalman filter to fuse the z-acceleration with the pressure-inferred height, you should be able to obtain a drift-corrected altitude estimation. Even without the accel fusion, the pressure sensor can discriminate altitude with ~13 cm resolution. This will be more accurate than accel integration alone. Best would be a combination of pressure-derived altitude with drift corrected by referring to the accel to eliminate changes in pressure due to wind or weather fluctuations, similar to the way magnetic anomaly detection works by checking with the accel to see if mag change are really due to motion, etc.

In any case (and for dtrain4) the three sensors need to have adequate bias calibration to recenter the response surfaces to the origin and in the case of the magnetometer, some kind of spherizing of the response surface is also usually necessary to get good heading estimation. If your pitch and roll are not stable to ~1 degree then you are doing something wrong. Heading should be accurate to at least 10 degrees, and with good sensors and a good sensor fusion engine this can be reduced to under 2 degrees. See: https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/.

Kris

bvanharen, 2015-12-10:
Hi Kris,

Thanks for your answer, it’s given me some new ideas and more insight.
I’ll think this stuff all over, thanks so far!

Bob

westKazhi, 2018-08-13:
Hi
I have a problem whilst using the MPU sensor. The magnetometer gives wrong deflection whenever the 'bot approaches some magnetic environment like metal furniture and stuff. the sensor works fine in open environment.
Thanks
Niv

kriswiner, 2018-08-13:
This is called magnetic anomaly detection.

westKazhi, 2018-08-15:
Hey Kris
How do I rectify this anomaly? Can you give some insight as to what to do.
Thanks
Niv

kriswiner, 2018-08-15:
Don't know, you'll have tof igure this out for yourself. But note that the
accel and mag are there to correct the gyro, and that the accel does not
respond to mag anomalies, so you can use this to check on the validity of
mag swings.

MitalPattani, 2018-08-22:
Won't the anomaly be rectified in sensor fusion? Isn't that the point?

kriswiner, 2018-08-22:
No, this is why we have to calibrate the magnetometer (and other sensors).
The magnetometer has to reflect real changes detected in a more or less
fixed Earth field. This provides the absolute in absolute orientation
estimation. If the field also depends on external source that are not fixed
then you won't get a correct solution.

You have to develop an algorithm such that the MCU can detect when the
field is real and when anomolous and either weght the mag field changes
less or discount them until the field returns to the natural one.

Plain sensor fusion doesn;t do this, but I suppose you are right in a way,
that this will be done in an enhanced sensor fusion that detects and
corrects for magnetic anomalies. The Madgwick/Mahony algorithms do not.

MitalPattani, 2018-08-23:
What if I eliminate the mag? I mean if I were to use only accel and gyro for a motion sensing device do you think it will be more accurate?

kriswiner, 2018-08-23:
If you don't need absolute orientation estimation sure. I mean your heading
will drift a lot...

MitalPattani, 2018-08-24:
Alright makes sense