kriswiner/MPU9250 #28: MPU-9250 and SPI code

bennyber01, 2015-11-07:
/* MPU9250 Basic Example Code
 by: Kris Winer
 date: April 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on Arduino.
 
 Hardware setup:
 MPU9250 Breakout --------- Arduino
 VCC ---------------------- 5V
 SCLK --------------------- 13
 SDI ----------------------- 11
 SDO ----------------------- 12
 NSC ----------------------- 10
 GND ---------------------- GND
 
 Note: The MPU9250 and SPI
 */
#include <SPI.h>

#define NCS_PIN   10
#define MOSI_PIN  11
#define MISO_PIN  12
#define SCK_PIN   13

const int NCS = NCS_PIN;

uint8_t SPI_read_register( uint8_t reg )
{
  digitalWrite( NCS, LOW);
  SPI.transfer( reg | 0x80 ); // reg | 0x80 to denote read
  uint8_t read_value = SPI.transfer( 0x00 ); // write 8-bits zero
  digitalWrite( NCS, HIGH);
  return read_value;
}

/************************************************************************/

void SPI_write_register( uint8_t reg, uint8_t value )
{
  digitalWrite( NCS, LOW);
  SPI.transfer( reg );
  SPI.transfer( value );
  digitalWrite( NCS, HIGH);
}

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define WHO_AM_I_AK8963  0x00 // should return 0x48
#define INFO             0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L	 0x03  // data
#define AK8963_XOUT_H	 0x04
#define AK8963_YOUT_L	 0x05
#define AK8963_YOUT_H	 0x06
#define AK8963_ZOUT_L	 0x07
#define AK8963_ZOUT_H	 0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
[... 1063 lines cut ...]
void writeByte(uint8_t address, uint8_t subAddress, uint8_t data)
{
  SPI_write_register(  subAddress, data );
}

 uint8_t readByte(uint8_t address, uint8_t subAddress)
{
	uint8_t data; // `data` will store the register data	 
        data = SPI_read_register( subAddress );
	return data;                             // Return data read from slave register
}

void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  

    digitalWrite( NCS, LOW);
  for (int i = 0; i < count; ++i)
  {
  uint8_t reg = subAddress + i;
    SPI.transfer( reg | 0x80 ); // reg | 0x80 to denote read
    uint8_t read_value = SPI.transfer( 0x00 ); // write 8-bits zero
    dest[i]  = read_value;
  }
    digitalWrite( NCS, HIGH);
}

natd89, 2016-08-31:
I had a question about your _readbytes_ function. when you store a value in read_value, you send a byte of zeros through SPI.transfer.  Are you using SPI.transfer(0x00) just to get a value back, or does transferring 0x00 to the IMU reset the register?

kriswiner, 2016-08-31:
Which sketch/file are you referring to? I usually use I2C for my
applications, I think I forked another repository that uses SPI in order to
someday make this work, but I haven't got around to it.

<https://github.com/kriswiner/MPU-9250/issues/28#issuecomment-243881650> ,
3PCZSaAks5qldyvgaJpZM4GeAI8> .
<https://github.com/notifications/beacon/AGY1qt9K8Nz1-kOkHT4C3wvnQl7Zn3Sgks5
qldyvgaJpZM4GeAI8.gif>

natd89, 2016-08-31:
I'm referring to the one titled "MPU-9250 and SPI code #28". The _readbytes_ function is at the very bottom of the sketch.

kriswiner, 2016-08-31:
No idea, this is not my code (at least not some of it), someone has hacked
it and posted it here apparently.

<https://github.com/kriswiner/MPU-9250/issues/28#issuecomment-243889423> ,
qXwHpT7ks5qleL4gaJpZM4GeAI8> .
<https://github.com/notifications/beacon/AGY1qk8K24CO5hsr6ueGvpQpuhB1bDRWks5
qleL4gaJpZM4GeAI8.gif>

brianc118, 2016-11-28:
@natd89 For interfacing the MPU-9250 with Arduino I suggest you use https://github.com/brianc118/MPU9250/.

And yes, SPI.transfer() is reading 1 byte of data *but* also sending one byte of data to the slave concurrently. SPI is a bidirectional protocol. i.e. for every byte sent, one byte is received.

Someone should close this issue.