kriswiner/MPU9250 #29: Angles changes when I move.

alex934, 2015-11-09:
Hey Kris, thank you for sharing your code.
Well I'm using a µC cortex M3 at 77MHz.
I sample IMU at 50Hz and fusion at 1000Hz using Madwick algo.
Firstly I calibrated my imu and my compass with your code:

void magcalMPU9250(float * dest1, float * dest2) 
 {
 uint16_t ii = 0, sample_count = 0;
 int32_t mag_bias[3] = {0, 0, 0}, mag_scale[3] = {0, 0, 0};
 int16_t mag_max[3] = {0xF000, 0xF000, 0xF000}, mag_min[3] = {0x7FFF, 0x7FFF, 0x7FFF}, mag_temp[3] = {0, 0, 0};

 CyDelay(4000);

 sample_count = 128;
 for(ii = 0; ii < sample_count; ii++) {
 readMagData(mag_temp);  // Read the mag data   
int jj = 0;
 for (jj = 0; jj < 3; jj++) {
  if(mag_temp[jj] > mag_max[jj]) mag_max[jj] = mag_temp[jj];
  if(mag_temp[jj] < mag_min[jj]) mag_min[jj] = mag_temp[jj];
 }
 CyDelay(12);  // at 8 Hz ODR, new mag data is available every 125 ms
 }

// Get hard iron correction
 mag_bias[0]  = (mag_max[0] + mag_min[0])/2;  // get average x mag bias in counts
 mag_bias[1]  = (mag_max[1] + mag_min[1])/2;  // get average y mag bias in counts
 mag_bias[2]  = (mag_max[2] + mag_min[2])/2;  // get average z mag bias in counts

 dest1[0] = (float) mag_bias[0]*mRes*magCalibration[0];  // save mag biases in G for main program
 dest1[1] = (float) mag_bias[1]*mRes*magCalibration[1];   
 dest1[2] = (float) mag_bias[2]*mRes*magCalibration[2];  

// Get soft iron correction estimate
 mag_scale[0]  = (mag_max[0] - mag_min[0])/2;  // get average x axis max chord length in counts
 mag_scale[1]  = (mag_max[1] - mag_min[1])/2;  // get average y axis max chord length in counts
 mag_scale[2]  = (mag_max[2] - mag_min[2])/2;  // get average z axis max chord length in counts

 float avg_rad = mag_scale[0] + mag_scale[1] + mag_scale[2];
 avg_rad /= 3.0;

 dest2[0] = avg_rad/((float)mag_scale[0]);
 dest2[1] = avg_rad/((float)mag_scale[1]);
 dest2[2] = avg_rad/((float)mag_scale[2]);

 }

We agree that dest1 is the magbias table and dest2 is magCalibration table to calculate mx,my and mz right?
    mx = (float)magCount[0]*mRes*dest1[0] - dest2[0]; 
    my = (float)magCount[1]*mRes*dest1[1] - dest2[1];  
    mz = (float)magCount[2]*mRes*dest1[2] - dest2[2];

well,
the code works but if I rotate on one axe, I got the good angle,  but angles on other axes change too. As you said here: ''fusion rate is small compared to your sample data rate (likely with the 8 MHz Pro Mini) and/or your sensors are not properly bias calibrated''
https://github.com/kriswiner/MPU-9250/issues/26

But in my case 50Hz IMUcompare to 1000Hz fusion seems enough, and my IMU is calibrate.

So if you have any suggestions?

Thank you very much

kriswiner, 2015-11-10:
I'm not really sure what to say. Mild cross axis mixing is expected for
Euler angles; it is not the best approximation. If the accel, gyro, and mag
are all bias calibrated the mixing should only be apparent at very high
angles.

The factory mag calibration is read from the AK8963 and is what I think I
called mag calibration. This captures the different axes sensitivity
measured at the factory. The bias calibration (dest1) is just the offset on
each axis from the origin. Dest2 is a poor man's attempt to capture the soft
iron calibration required. It just rescales the data to equalize the
response on all three axes. The result should be a nearly spherical response
surface, You can check this by plotting Mx vs My and Mx vs Mz, etc. In
rprinciple, the same scaling and bias calibration sould be done for the
accel and gyro, but usually bias calibration is enough. That is, accel
should read 0, 0, 1 g and gyro 0, 0, 0 when the sensor is flat and at rest.

Well I'm using a �C cortex M3 at 77MHz.

#26 <https://github.com/kriswiner/MPU-9250/issues/26> 

<https://github.com/kriswiner/MPU-9250/issues/29> .
<https://github.com/notifications/beacon/AGY1qkz1LMuXXERBTsPt5e-zuhvCX6c8ks5
pEM16gaJpZM4Gew0P.gif>

alex934, 2015-11-10:
Thank you very much kris,
well I changed my fusion to 500Hz and as you can see my calibration seems good:
[image]

Actually you right the mixing is apparent when angle is > 80° so for my application it's enough.
Now the datas seems good but it really depends on the parameters, what would you advise for the Ascale,Gscale and Mscale? and how find the value of beta

Thanks a lot

kriswiner, 2015-11-10:
I usually use 2 g, 250 dps, and 48 G (fixed for AK mag) accel, gyro, and mag fullscale ranges, but this is just for wearable applications,. If you put the sensor on a vehicle, you would likely need 8 or 16 g for accel and 2000 dps for gyro. These are typical values for drones, for example.

You should not have heading drift of more than one or two degrees when motionless if you have a properly calibrated mag.

xxnz, 2015-11-25:
Hi Guys,

Just a simple question, after getting the hard & soft irons (dest 1 & 2), how can I implement them in the main loop??

Thanks.

kriswiner, 2015-11-25:
There should be an example of this in the sketch, but for the mag the hard
iron (offset bias) is simply subtracted from the scaled mag data, the soft
iron is a scale factor so the bias corrected mag data is multiplied by the
soft iron correction.

<https://github.com/kriswiner/MPU-9250/issues/29#issuecomment-159613235> .
<https://github.com/notifications/beacon/AGY1qhl6n4a2ZBz9Mb84fWd-RYWlVn21ks5
pJbTAgaJpZM4Gew0P.gif>

xxnz, 2015-11-25:
Hi Kris,

Thank you for your fast response! 
Pretty new to this whole imu area, and I was wondering if I have to do calibrations for accelerometer/gyro as well or your nice code has already done it in the self test part? So does that mean I shouldn't move the imu at the start up?

Thank you!

kriswiner, 2015-11-25:
The self test is a separate test from the calibrations, but yes, if the code
is doing accel and gyro calibrations, it expects the sensor is motionless,
otherwise you will get poor results.

Kris

<https://github.com/kriswiner/MPU-9250/issues/29#issuecomment-159659855> .
<https://github.com/notifications/beacon/AGY1qi6HIXWdaaCBsiCvup3lGkbGIfu2ks5
pJdZjgaJpZM4Gew0P.gif>