kriswiner/MPU9250 #33: Accelerometer/ Trinket

dtrain04, 2015-11-29:
Hey Kris,

I currently using the MPU-9250 with the 3V pro trinket with the pins properly connected and interrupt connected to 3 (since pin 2 doesnt exist).  I am getting a weird output... It performed accurately on the arduino uno but when I tried to use the trinket, the Ax and only Ax reading stay constant.  Idk if this is a voltage issue or something but the values go constant.

[image]

Any thoughts on what could be causing this?

dtrain04, 2015-11-29:
The first row is Ax, the second row is Ay and, the rest follow with the tenth row being deltat. for clarification

kriswiner, 2015-11-29:
This is pretty odd behavior. You can look at the individual bytes being
read. This is often the result of an integer type difference between
microcontrollers. If it works elsewhere you can discount the hardware and
code as being wrong, likely there is some kind of error in treating the raw
counts read or raw counts conversion to scaled data.

<https://github.com/kriswiner/MPU-9250/issues/33#issuecomment-160362317> .
<https://github.com/notifications/beacon/AGY1qg2vuEZ1Gc0ArgsqM4C_ZFa5oZnPks5
pKl_PgaJpZM4Gq7Zu.gif>

dtrain04, 2015-11-29:
See I thought the same thing too but all of the conversions are the same among all three accelerometer readings.... Ill  try dig around a bit and let you know if I come up with anything.  Wanted to see if you could think of something. Thanks!

dtrain04, 2015-11-29:
So after going through the code, I wanted you to know that I don't use the scale until I analyze the raw data in matlab.  Also, I wanted to ask why in your code the readAccelData only accesses the ACCEL_XOUT_H register.  What about Y and Z?  Should I try accessing the Y or Z registers instead? Thanks!

dtrain04, 2015-11-29:
Okay so there may be another issue... It worked fine on the arduino uno because I was connected serially through usb but not now Im connected via bluetooth to my desktop where all the processing is being done in matlab. Could this be a problem with the buffer? Its odd that the first number in a NaN.. What do you think? I am using the HC-05 module to a bluetooth dongle.

kriswiner, 2015-11-29:
Because I read six bytes and the x, y, and z are contiguous in register
memory. More efficient than reading just two bytes at a time. In fact, one
could read all 20 bytes (accel, gyr, mag, temp) of data at the same time,
even more efficient, but I like some modularity.

<https://github.com/kriswiner/MPU-9250/issues/33#issuecomment-160430098> .
<https://github.com/notifications/beacon/AGY1qkZwoQnGa8SIokRwAa4oHLp5n7nWks5
pKx4DgaJpZM4Gq7Zu.gif>

kriswiner, 2015-11-29:
I don't have much experience with transferring IMU data over BT but this
could be a problem. It is often the case that the first few bytes have to be
discarded. Why don't you try sending some garbage pre-bytes and see if the
actual data makes it through.

<https://github.com/kriswiner/MPU-9250/issues/33#issuecomment-160432665> .
<https://github.com/notifications/beacon/AGY1qhzq7595NCsaCkCzwbwii7hZByTMks5
pKynUgaJpZM4Gq7Zu.gif>

dtrain04, 2015-11-30:
SOLUTION:

For anybody out there using the HC-05 module and Kris's code with matlab, the fscanf function does not properly read the data incoming for some reason.  I had to delimit all of my outputs and use the raw data bits.. Below is my solution in matlab:

if nineDOF.BytesAvailable >0
            reading = char(fread(nineDOF,nineDOF.BytesAvailable)'); %read all bytes available
    %fscanf(nineDOF,'%s')
            
            data = [data reading]
            
end

This is in a while loop to keep retrieving the raw data from the arduino trinket via HC-05.
When the user is done collection data, transform the data array to a string using the sscanf function and then turning it into a double using the ',' delimiter.
data_string = sscanf(data,'%s');
    data_double = str2double(strsplit(data_string,','));

This gave me the correct outputs for the MPU-9250. I hope this helps anybody with similar problems...