kriswiner/MPU9250 #34: accel and magnetometer orientation

alexx600, 2015-11-30:
Hello! I have a few question to your MPU9250BasicAHRS.ino file. 

// Sensors x (y) -axis of the accelerometer is aligned with the y (x) -axis of the magnetometer; // The z-axis magnetometer (+ Down) is opposite to the z-axis (+ up) of accelerometer and gyro! // We have to make some allowance for this orientationmismatch in feeding the output to the quaternion filter. // For the MPU-9250, we have chosen a magnetic rotation sensor That keeps the forwards along the x-axis just like // in the LSM9DS0 sensor. This rotation can be modified is allow any convenient orientation convention. // This is ok to aircraft orientation standards! // Pass rate gyro as rad / s
// Madgwick Quaternion Update (ax, ay, az, GX * PI / 180.0f, gy * PI / 180.0f, g * PI / 180.0f, we, x, m); Mahony Quaternion Update (ax, ay, az, GX * PI / 180.0f, gy * PI / 180.0f, g * PI / 180.0f, we, x, m);

1.Why you did not use -mz instead of mz? I  mean
MahonyQuaternionUpdate (ax, ay, az, GX * PI / 180.0f, gy * PI / 180.0f, g * PI / 180.0f, we mx -mz);
Why in your code mz is positive?
2. Is it necessary that calculate mx,my,mz is gauss? In the filter is normalized it anyway? Or I am wrong?

kriswiner, 2015-11-30:
You do not need to scale the mag data, it is normalized.

<https://github.com/kriswiner/MPU-9250/issues/34> .
<https://github.com/notifications/beacon/AGY1qktPhk5Qy79NbPJjDBJ1Eh8t7i4Wks5
pLI22gaJpZM4GrqVa.gif>

alexx600, 2015-11-30:
OKAY.
All the time and do not understand why you do not use -mz? Could you tell something about it?

I am using microprocessor stm32f401 on the board evaluation discovery. And I have a huge problem with magnetometer. Axles what I assume are presented in the picture below
 http://obrazki.elektroda.pl/9993576100_1448821919.jpg
If you wanted to have a look. What kind of parameters should call the function?
MahonyQuaternionUpdate (ax, ay, az, gx, gy, g, x, we, -mz); ?

kriswiner, 2015-11-30:
This looks like North-West-Up, whereas Mahony and Madgwick usually assume North-East-Down convention. Put x facing North, which axis is pointing East? This is what you put into the Mahiny function for y. Which axis is pointing Down? This is what you put into z. So maybe you would put in x, -y, -z? Also, unless your accel/gyro and especially  mag is properly calibrated, you will get garbage from the filters.

alexx600, 2015-12-01:
I am confused all the time.
http://obrazki.elektroda.pl/5888817700_1448883307.jpg this is mpu9250.
Accelerometer and Gyroscope is North-West-Up
Magnetometer is North-East-Down

if Madgwick is North-East-Down why You use 

MahonyQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);

Why you use North-West-Up to accelerometer and gyroscope data?

kriswiner, 2015-12-01:
I am lazy, better to correct the orientation to make it consistent.

<https://github.com/kriswiner/MPU-9250/issues/34#issuecomment-160938607> .
<https://github.com/notifications/beacon/AGY1qqU5o4OftRf6dHyhiMcfHQNxFdN2ks5
pLXmsgaJpZM4GrqVa.gif>

alexx600, 2015-12-03:
Which MPU9250 IC
http://www.lucidarme.me/wp-content/uploads/2015/01/AxisOrientation.png
Your code looks like
MadgwickQuaternionUpdate(ax,..., my, mx, mz);

Which BNO-055 IC
http://api.ning.com/files/4Epf-T61UHP41R5kTSFUGZGG6Y5QUaIJsZX7p6NHlqqaLunj1GGg5sE1roq0ssP4yS7Sp*7jmI2vr8P5N561hYwSloZoWx76/BNO055.jpg
Your code looks like
MadgwickQuaternionUpdate(ax,..., my, mx, mz);

In MPU9250 magnetometer Z axis is negated, and code is not modifed.
In BNO-055 magnetometer Z axis is normal, and code is not modifed.
I do not understand why?

As you said Madgwick filter is NED, futher information about optimalization could be found there
http://stackoverflow.com/questions/26416738/madgwick-imu-algorithm-simulate-on-iphone

But when i watch this video https://www.youtube.com/watch?v=BXsGWoOMtmU

Top is NORTH-WEST-UP
Bottom is NORTH-EAST-DOWN

How it shuld be ?

mattisback, 2017-03-09:
I'm also finding this greatly confusing.  Could me being in the southern hemisphere be causing the algorithm to get confused?

kriswiner, 2017-03-09:
I doubt it. More likely there is something wrong with the implementation.