kriswiner/MPU9250 #35: GPS+AHRS with Xadow Mbed (mag calib)

whatnick, 2015-12-01:
I have been following your work for a while and I have created a GPS- AHRS stack using modules from the Xadow family. You can see my mbed code here: https://developer.mbed.org/users/whatnick/code/Xadow_MPU9150AHRS/ . I was wondering if it would be possible to add the magnetometer initial calibration code to the mbed library. FYI , Xadow M0 runs the sensor fusion loop at 500Hz.

kriswiner, 2015-12-01:
Hi Tisham,

Everything I put on github is open source; are you proposing to add
something from my repository for MPU9150 mag calibration on mbed? This is
fine with me. The most recent (complete) mag calibration function is
actually in the  MPU9250 code.

Kris

<https://github.com/kriswiner/MPU-9250/issues/35> .
<https://github.com/notifications/beacon/AGY1qrcP_td5SlkkpwfOzNN1wejCp9FEks5
pLPJRgaJpZM4Gr732.gif>

whatnick, 2015-12-01:
Yes I was proposing to port the calibration code to this library you have on mbed - https://developer.mbed.org/users/onehorse/code/MPU9150AHRS/ .

whatnick, 2015-12-01:
Could you point me to the .ino file with calibration code in it.

whatnick, 2015-12-01:
Looks like the sample loop is here: magcalMPU9250 . I will port this over to mbed.

kriswiner, 2015-12-01:
See:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

You can also read about some magnetometer calibration studies I did here:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration

Kris

<https://github.com/kriswiner/MPU-9250/issues/35#issuecomment-160828765> .
<https://github.com/notifications/beacon/AGY1qumImC-dnKUs5Vw3XAUTka0tbQhGks5
pLP5rgaJpZM4Gr732.gif>

kriswiner, 2015-12-01:
Let me know if I can help...

<https://github.com/kriswiner/MPU-9250/issues/35#issuecomment-160832721> .
<https://github.com/notifications/beacon/AGY1qsta0CUAzwrHpPE5IB2OjGTmfu75ks5
pLQPqgaJpZM4Gr732.gif>

whatnick, 2015-12-01:
Hi Kris I have update the MPU9150 with the calibration loop here - https://developer.mbed.org/users/whatnick/code/Xadow_MPU9150AHRS/file/46b5acee9472/MPU9150.h . You could add it to your main repo for other Mbed users.