kriswiner/MPU9250 #36: Using MPU9250BasicAHRS with Arduino

CJRud, 2015-12-11:
Hi there,

In your description you say that this code is an Arduino Sketch. I'm fairly new to coding with Arduino and would love to use your libraries/ code to get fused sensor data coming out of my MPU9250 to be logged onto an SD card. 

For the time being, I'm taking one step at a time and just want to see the fused sensor values in my serial monitor without any SD card functionality. I am using just an **Arduino Uno R3 and MPU9250 breakout from Drotek**, see link below, but **no Adafruit display that is described in the "MPU9250BasicAHRS.ino" file description**. When I run the "MPU9250BasicAHRS.ino" there are obvious problems associated with the fact that I don't have the display libraries/ LCD attached. It won't compile. The MPU9250 I have is on the default I2C bus address 0x68 and is recognised when running an I2C scanner.

**I'd like all the functionality you described** in the starting comments: "Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Addition of 9 DoF sensor fusion using open source Madgwick and Mahony filter algorithms", just **without the LCD display**. **How do I do this?**

Could you give any advice (quick step-by-step perhaps) or adapted code that can get this functionality using what I have? I assume it isn't as simple as commenting/ deleting all the calls to the LCD display/libraries? Hopefully your experience can help me out here!

Secondly, I'd like to sample my accel, gyro and mag at a fairly low **25 Hz**. Looking through the code I found a number of comments talking about setting sample rates for each accel, gyro, mag and temp sensors but was confused as to exactly how to do this. There's a lot of code here and I don't want to change something somewhere and have it mess up everything else. **How do I do this?**

If you also have any comments about the feasibility of saving these values to an SD card at 25Hz then I'd be happy to hear them. I'm using the Adafruit datalogging SD card shield, see link below.  I tried a while ago to save from an MPU6050 using DMP to an SD card and the FIFO buffer didn't like it, it kept overflowing and everything was very slow. I resorted then to saving only raw sensor values, which worked very well, but I really want to use the fused values for this latest project.

Any advice welcomed! Thank you in advance, looking forward to hearing your comments.

Best,
Chris

Drotek sensor (assembled for I2C):
http://www.drotek.com/shop/en/home/421-mpu9250-gyro-accelerometer-magnetometer.html?search_query=mpu9250&results=3

Adafruit SD shield:
https://www.adafruit.com/products/1141

kriswiner, 2015-12-11:
Hi Chris,

If you are not using the display, then yes remove all of the reference to
it.

Also, the I2C functions are set up to use the i2C_t3.h library for the
Teensy (which I highly recommend). If you want to use Wire.h on the UNO you
will need to make some modifications on the wire.begin and wire.readByte,
etc functions. The Wire.h versions are commented out in those function,
shouldn't be too hard to figure out.

If you want to understand how to change the data rates you are going to have
to read the data sheet; I would recommend this anyway if you are going to
use this sensor well.

Once you get these issues sorted out, the code will produce fused yaw,
pitch, and roll as well as quaternions for you to spew to the serial
monitor. After that, writing to an SD card is easy.

Kris

http://wwwdrotekcom/shop/en/home/421-mpu9250-gyro-accelerometer-magnetometer
html?search_query=mpu9250&results=3

https://wwwadafruitcom/products/1141 

-
<https://github.com/kriswiner/MPU-9250/issues/36> .
<https://github.com/notifications/beacon/AGY1qq00IyLFsk0i6wvq2Z8ABv3t2cQoks5
pOvBKgaJpZM4Gzs4a.gif>

andresespinosapc, 2016-10-10:
@CJRud could you make it work? I also want to use an Arduino UNO, just because I already have it. I don't care if it could work better, I just want it to work.

@kriswiner I don't understand why should we modify the Wire.h if you specify in the `MPU9250_MS5637_AHRS_t3.ino` file that the "Sketch runs on the 3.3 V 8 MHz Pro Mini". If it works for that Arduino, why does it not work for the Arduino UNO?

kriswiner, 2016-10-10:
The _t3 means it uses the Teensy-specific i2c_t3.h wire library.

<https://github.com/kriswiner/MPU-9250/issues/36#issuecomment-252755101> ,
cQkX_pSks5qyrJqgaJpZM4Gzs4a> .
<https://github.com/notifications/beacon/AGY1qsN2NOPFBBuq2uOJ7E-ICZz5reHdks5
qyrJqgaJpZM4Gzs4a.gif>