kriswiner/MPU9250 #37: Multiple I2C with MPU9250

gawsalyan, 2016-01-23:
Hi...
How can I change the code to connect at least 2 MPU9250 sensor to STM32F401RE to different I2C buses

kriswiner, 2016-01-23:
Each device can be 0x68 or 0c69 depending on whether the SDO pin id LOW or
HIGH, respectively. So the strategy to read multiple MPU9250 on the same bus
is to set them all LOW except for the one device you want to read, its
address will be 0x69. Just keep toggling all of the devices until you have
read them all, then repeat. Or you can use an I2C multiplexer like the
TCA9548A available at Tindie.com.

<https://github.com/kriswiner/MPU-9250/issues/37> .
<https://github.com/notifications/beacon/AGY1qsCJAuoYp-Oyjh6vdAg_6XDabRO5ks5
pcwzMgaJpZM4HK3Pp.gif>

gawsalyan, 2016-01-23:
Thanks a lot Robert,
But I need to connect sensors with every i2c master of STM32F401RE and I couldn't able to achieve high speed, i.e.. I have reached only about 200+ Hz with your code, but I need at least 350+ fused data.
How can I change the MPU9150/9250 library to support that.

kriswiner, 2016-01-23:
My pleasure Wilbur,

What do you mean by speed? Speed of the sensor fusion? With Madgwick or
Mahony? I was able to easily get 1 kHz fusion rates using an STM32F4 Nucleo
board.

If you mean sample data rate, then you just have to change the input
parameter at the beginning of the sketch to a higher rate. The accel and
gyro supports rates at or above 1 kHz.

I still do not understand what you are asking...

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-174198226> .
<https://github.com/notifications/beacon/AGY1qnvZCttkz7qr4GOByGQ2U2pU7nbmks5
pc6I1gaJpZM4HK3Pp.gif>

gawsalyan, 2016-01-23:
I am accessing mpu9150 sensor using my Stm32f401 I2C with ur library.
..only change I did was..I changed the serial out to virtual com port...but
i could only able to achieve about 200 quaternion output...how can I
increase it to above 350.

Thank u

kriswiner, 2016-01-23:
The problem with the MPU9150 is the mag, which must be read with a one-time
only write/read, not a continuous 100 Hz read like the MPU9250, so the rate
is limited by the time it takes to write to the mag to tell it to prepare a
sample, and then read the sample data. I am surprised you can even get to
200 Hz fusion rates. It takes ~10 ms for the AK8975 mag to write/read the
data.

Switch to the MPU9250 and you should achieve rates above 1 kHz if desired.

https://github.com/notifications/beacon/AGY1qnvZCttkz7qr4GOByGQ2U2pU7nbmks5

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-174210747> .
<https://github.com/notifications/beacon/AGY1qgcvsZzW1JtEw6lqa9lA05XwE-CUks5
pc77agaJpZM4HK3Pp.gif>

gawsalyan, 2016-01-24:
Really thank you for your advice and valuable time.

But I tried with MPU9250 also.. I couldn't achieve high data rate.
And How can I use other I2c master of the STM32F4 board to connect other
sensors.
​

kriswiner, 2016-01-24:
There must be something wrong with your setup; I am able to get greater than 1 KHz fusion rates with a Cortex M4 rocessor.

gawsalyan, 2016-01-24:
I connected my MPU9250 SCL and SDA to PB_8 and PB_9 VCC and GND to 3.3V
annd GND.
And I only commented the coding which is respect to lcd displays only...my
output is on the hyper terminal so, I use SERIAL_TX and SERIAL_RX as my
Serial out.
Is there anything I can DO..
and what about multiple I2C master

gawsalyan, 2016-01-24:
Sorry ..it was my fault...it's not the data rate. ..but when I try to print
it through serial port pc. it's serial out rate is low. Is there anyway to
increase the terminal output..or is it possible to wifi shield and increase
the output from STM32F401 to PC.
On 24 Jan 2016 07:47, "Siva Gawsalyan" <gawsalyan@gmail.com> wrote:

> I connected my MPU9250 SCL and SDA to PB_8 and PB_9 VCC and GND to 3.3V
> annd GND.
> And I only commented the coding which is respect to lcd displays only...my
> output is on the hyper terminal so, I use [...]

kriswiner, 2016-01-24:
I don't know, but why not try an LCD display like the Nokia 5110, the update
rate is much faster than serial.

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-174298740> .
<https://github.com/notifications/beacon/AGY1qh50cH6ilZgkF_jqoG0PctcmHPONks5
pdMnFgaJpZM4HK3Pp.gif>

gawsalyan, 2016-01-25:
I am developing a Java based platform also to view the motion..that's why I
need to do that...
Can u give me suggestion on using multiple sensors with different i2c.?

kriswiner, 2016-01-25:
I thought I already described how to do this. You can also use one of these:

https://www.tindie.com/products/onehorse/tca9548a-i2c-multiplexer/

https://github.com/notifications/beacon/AGY1qh50cH6ilZgkF_jqoG0PctcmHPONks5

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-174364858> .
<https://github.com/notifications/beacon/AGY1qnsx_lmb1fpt2FHfcrJ9bDIN5E0Gks5
pdW4-gaJpZM4HK3Pp.gif>

gawsalyan, 2016-01-25:
i am able to achieve that using higher baudrate 921600.

is it possible instanciate two MPU9250 in the mbed environment to use it
via second i2c master?

YanJenHuang, 2016-04-09:
Hi Kris,
I am connecting two MPU9250 using I2C, and I pulled AD0 to low or high to change the address 0x68, 0x69.
I have an issue here:
When I pull AD0 to low, the mpu9250 address is 0x68, but I set the address to 0x69 purposely, I cannot read accelerator or gyro raw data, it make sense, but when I am reading the mag raw data, I get the mag raw data!!
So I am confused what should do then I can read the individual mag raw data from two sensor?

kriswiner, 2016-04-09:
Each MPU9250 has an internal switch to allow mag acces from the host. Just
turn this switch off for the MPU9250 with I2C address 0x68 and you can read
the mag from the MPU9250 with address 0x69.

// Configure Interrupts and Bypass Enable
// Set interrupt pin active high, push-pull, hold interrupt pin level HIGH
until interrupt cleared,
// clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional chips

  // can join the I2C bus and all can be controlled by the Arduino as master
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22); 

set INT_PIN_CFG to 0x20 to turn off switch.   

<https://github.com/notifications/beacon/AGY1qjo3NNoG2xnq5xyzXUWmgpkgYjHAks5
p12GigaJpZM4HK3Pp.gif>

YanJenHuang, 2016-04-11:
Thanks Kris~
I will try it. =)

KhArtNJava, 2016-11-23:
@kriswiner , I still can't understand, if we will change MPU-9250 address by AD0, that described in datasheet like 
[image]
it will change only MPU9250_ADDRESS, but what about AK8963_ADDRESS ?
What will be happened with AK8963_ADDRESS, if AD0 will be HIGH (will be 1)?
Will mag AK8963 be still accessible by AK8963_ADDRESS=0x0C, when AD0 = 1? Or, when AD0 = 1 mag will be disabled? If it will be disabled, where is info about it in MPU-9250 datasheet?
[MPU-9250-Datasheet.pdf](https://github.com/kriswiner/MPU-9250/files/608940/MPU-9250-Datasheet.pdf)

KhArtNJava, 2016-11-23:
What will be happened with AK8963 address, when AD0=1? Will it be changed, or not?
If not, how to read mag data from multiple MPU-9250, if many of AK8963 will have same address?
Thank you!

KhArtNJava, 2016-11-23:
Looks, I understood it... I've correctly understand your prev post, when I wrote this ))).

kriswiner, 2016-11-23:
OK, then did you get it to work?

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-262562220> ,
syCmBA-ks5rBGh9gaJpZM4HK3Pp> .
<https://github.com/notifications/beacon/AGY1qpi4Ul1Dk_MyTah89WjBeRzY2KVcks5
rBGh9gaJpZM4HK3Pp.gif>

kriswiner, 2016-11-23:
When you change the MPU9250 ADO this does not change the I2C address of the AK8963C. The correct way to handle multiple MPU9250's on a single I2C bus is described here:

https://github.com/kriswiner/MPU-9250/issues/86

but you can also do this:
Each MPU9250 has an internal switch to allow mag acces from the host. Just turn this switch off for the MPU9250 with I2C address 0x68 and you can read the mag from the MPU9250 with address 0x69. // Configure Interrupts and Bypass Enable // Set interrupt pin active high, push-pull, hold interrupt pin level HIGH until interrupt cleared, // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional chips // can join the I2C bus and all can be controlled by the Arduino as master writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22); set INT_PIN_CFG to 0x20 to turn off switch.

KhArtNJava, 2016-11-23:
No, now I still learning datasheet and Register map.
I think, I will do like
1) Set AD0 = 1 for all MPU;
2) In cycle, set AD0=0 for each MPU and try to set INT_PIN_CFG to 0x20 and power down mag with AK8963_CNTL
3) In cycle, before poll sensors, I plan to set INT_PIN_CFG to 0x22 and power up mag with AK8963_CNTL, after that read sensors data, set INT_PIN_CFG to 0x20 and power down mag with AK8963_CNTL.

Ie, 
1) on init - power off mags in all MPU modules, set INT_PIN_CFG to 0x20.
2) Before reading data, enable BYPASS_EN and power up mag in needed MPU module, read data from sensors on this module, power off mag in this MPU module and disable BYPASS_EN for this module.
3) Go to next MPU module and repeat from step 2. Do this in infinite loop. 

Is it correct idea?

kriswiner, 2016-11-23:
You should be able to leave the AK8963 on all the time and just use INT_CFG
to open and close the internal I2C switch to enable or disable access to the
AK8963.

But using the slave registers is the better way to do it.

<https://github.com/kriswiner/MPU-9250/issues/37#issuecomment-262573054> ,
iF767jmks5rBHGCgaJpZM4HK3Pp> .
<https://github.com/notifications/beacon/AGY1qnziqYAXuw_dfmAsrY9GCHDhq3p3ks5
rBHGCgaJpZM4HK3Pp.gif>