kriswiner/MPU9250 #38: Register setup error?

TimBergel, 2016-01-29:
Hi Kris,

I'm using your MPU9250 code with a Teensy 3.2 and it has been a vast help to me, thank you very much for making it available. However, there seems to be a flaw in the code that sets up some of the configuration registers. In initMPU9250() you have code intended to read, modify and write back register data thus:

// Set gyroscope full scale range
// Range selects FS_SEL and AFS_SEL are 0 - 3, so 2-bit values are left-shifted into positions 4:3
uint8_t c = readByte(MPU9250_ADDRESS, GYRO_CONFIG);
writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c & ~0x02); // Clear Fchoice bits [1:0] 
writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c & ~0x18); // Clear AFS bits [4:3]
writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c | Gscale << 3); // Set full scale range for the gyro

but surely this is not correct? Each time you write to the register you overwrite whatever you wrote before, so the fchoice selection is unchanged and the full-scale selects are or-ed with the previous values (I found this when I tried to use different scale values). I think the code needs to be something like:

uint8_t c = readByte(MPU9250_ADDRESS, GYRO_CONFIG);
c &= ~0x02; // Clear Fchoice bits [1:0] (but I think it should be ~0x03)
c &= ~0x18; // Clear AFS bits [4:3]
c |= Gscale << 3; // Set full scale range for the gyro
writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c); // Write back changed data

The same issue applies to setting up ACCEL_CONFIG and ACCEL_CONFIG2.

My apologies if I should have submitted changed code to the repository instead of bothering you - I'm very new to GitHub.

kriswiner, 2016-01-29:
I think you are correct. I will make these changes. Thanks,

Kris

<https://github.com/kriswiner/MPU-9250/issues/38> .
<https://github.com/notifications/beacon/AGY1quU4O8trq6kVYmszpwYUxfmOiwqVks5
pe55AgaJpZM4HPaTU.gif>

davidthings, 2016-03-19:
Possibly 

c &= ~0x02; // Clear Fchoice bits 1:0

Ought to be

c &= ~0x03; // Clear Fchoice bits 1:0

Since 0x02 = 0010b and 0x03 = 0011b

TimBergel, 2016-03-19:
Errgh ... yes, you are of course quite correct davidthings, can't imagine what I was thinking of