kriswiner/MPU9250 #39: How can I get the bias of Acc and Gyro?

ijunglee, 2016-02-05:
Hello, I would like to ask that how to get the bias of Acc and Gyro? I'm not really sure for that......
I let the IMU be flat on the table with Z-axis toward up, and then I get the average value of 1000 records of Ax, Ay, Az. The value should be 0, 0, 1, but the values I got are -0.2, -0.06, 0.96. Is the bias of Acc 0.2, 0.06, 0.04? So I have to add 0.2, 0.06, 0.04 to all of values of Ax, Ay and Az. Is this right?
Can I use the method to get the bias of Gyro, too? I think it should be 0, 0, 0 deg/sec when it is flat on the table.
Thank you very much!

kriswiner, 2016-02-05:
What you describe is basically correct I just don't know if the numbers you
are quoting are millig or counts or what. As long as they are consistent
this procedure should work well. The bias values must be subtracted from the
subsequent data. The same procedure can work for the gyro too. Magnetometer
is different. You can read about a successful strategy here:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration

Kris

<https://github.com/kriswiner/MPU-9250/issues/39> .
<https://github.com/notifications/beacon/AGY1qmDNcpWazZV4bJxjIH4eiwmwxSnkks5
phCZFgaJpZM4HUBaj.gif>

ijunglee, 2016-02-05:
Thanks for your reply and strategy sharing!
The numbers I quoted are g (unit of accelerometer). The correct value when the z-axis toward up is (Ax, Ay, Ax) = (0,0,1). 
By the way, did you use the magnetometer of MPU-9250? I use the LSM9DS0 now and want to change to MPU-9250. What's the difference between the two IMUs? I found that the Magnetometer Scale of MPU-9250 can not be changed, but LSM9DS0 can. Does it affect the result when I use Madgwick AHRS filter? Thank you very much!

kriswiner, 2016-02-05:
There is a comparison of these two IMUs at the github repository I linked
to.

<https://github.com/kriswiner/MPU-9250/issues/39#issuecomment-180236797> .
<https://github.com/notifications/beacon/AGY1qhd6wdPhZNVXQSYsOjNh7j6XpTDyks5
phEiYgaJpZM4HUBaj.gif>