kriswiner/MPU9250 #42: calibration of magnetometer

vreijs, 2016-02-13:
Hello kriswiner,
I read you article: https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration and I am wondering about a few things:
. do you know if in the Android OS this calibration is included or does the App writers need to implement this?
. if it is part of the OS, is this 'calibration' movement one need to do in some way recognised by the OS? And when recognised, it performs the calibration (both for hard and soft irons?).

I am asking this as I have some problems with measuring azi/alt.
I am using a Samsung Tab2 7.0 GT-P3110 Android 4.2.2
Regardless of the App I am using (e.g. SunSurveyor or Measure Angle Apps), the results of azimuth and altitude measurements depend on the rotation of the device. So when pointing towards an object (Moon/Sun) with known azi/alt with the camera, the azimuth and altitude measurements changes when the device is rotated (see attached plot).
[image]

The precision both in azimuth and altitude is quite good (0.5 deg, by measuring the azi/alt several times using the same rotation of the device), which is close to the one of a normal compass/clinometer. But the accuracy is not that good! 
<see for definition of the word accuracy and precision here: https://en.wikipedia.org/wiki/Accuracy_and_precision >

Attached is a plot of the measured azi and alt (using SunSurveyor App,
but other Apps [such as Measure Angle] have same behavior).
I turned the device 90 degrees (starting with landscape, portrait, the
-landscape and -portrait) and measured each time the az/alt of the
Moon. You also see the average of these measurements and the ephemeris location
of the Moon. So the average looks ok-ish compared to
the ephemeris values calculated for the Moon.

I have now been measuring the Moon and Sun many times and I see every
time the similar behavior and accuracy spread (although same variation
can happen). Again the precision stays within the 0.5 degrees when doing
multiple measured at the same rotation.
By the way I am away from iron objects (external or on body) and also
away from electricity cables, etc. I also do the so called
'calibration' of the magnetometer (I did several different calibration
movements). That calibration has some effect, but normally the spread stays the same. 
I also compared it with a normal compass and clinometer, which gives values comparable to the ephemeris.

If you see that plot, do you think this is something in the OS, the sensor, the device and/or the App?
Calculating azi/alt is of course not an easy thing. In the device or Operating systems things are being done of course, but I am wondering why I see this large inaccuracy (certainly in azi [up to 20 degrees], and smaller in alt [up to 7 degrees])?

Here is another article that described the issues around the magnetometer
(and also how to overcome them [partly]). I am wondering if the
calibration inside the device and/or OS is done properly?
http://www.nxp.com/files/sensors/doc/app_note/AN4246.pdf

Any ideas/hint you see in this? 

Thanks for any feedback.

All the best,

Victor

vreijs, 2016-02-13:
here are ta few other sessions of such rotations (also different Apps)
[image]
[image]
[image]
[image]

kriswiner, 2016-02-13:
Hi Victor,

This is a very interesting application of attitude estimation! I hadn't
though about using it for astrometry.

I don't know the details of the tools you are using but I have some thoughts
and a suggestion.

First of all, as you note, precision is not very useful if the accuracy is
poor. Since sensor fusion to obtain absolute orientation (attitude
estimation) is well-understood I must assume the problem is in the
underlying sensors data. I have ample evidence that even with excellent
sensor fusion filtering, the accuracy can be poor (OK, meaning 7 - 10
degrees in heading or attitude) if the underlying sensors are not very good.
Even hard iron and limited soft iron calibration can't always cure a
misbehaving magnetometer, and for some sensors nothing can help.

You can see examples of this here:

https://github.com/kriswiner/MPU-6050/wiki/Hardware-Sensor-Fusion-Solutions

 and here:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/E.-Typical-Resul
ts-Using-the-SENtral

 
The data plots show variations in azimuth and altitude of ~4 degrees
standard deviation. This is not terribly bad, in fact I would say this is
typical for even pretty good sensor fusion solutions out there like the
MAX21000. Do you know which sensors are being used in the Samsung product? 

The best orientation solution I have found is the EM7180 coupled with the
MPU9250 which can provide 1 degree accuracy in pitch and roll and 2 degree
accuracy in heading (for AHRS navigation). Not sure what this would
translate to for astrometry.

So in short, the problem is IMHO not the apps, the fusion algorithms, or the
calibration (this can always be better but can't fix poor sensors) but the
underlying sensor data and there is nothing you can do about that except
change sensors.

Kris

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration and I am wondering about a few things:

image18
<https://cloud.githubusercontent.com/assets/17218070/13027981/3599099c-d259-
11e5-9ce8-4d2930c4545c.gif> 

http://www.nxp.com/files/sensors/doc/app_note/AN4246.pdf

<https://github.com/kriswiner/MPU-9250/issues/42> .
<https://github.com/notifications/beacon/AGY1qj5F8eb0XZsN4fRaS0VIuGaQFcsKks5
pjy5JgaJpZM4HZo42.gif>

vreijs, 2016-02-13:
Thanks Kris for your very valuable feedback.
The stdev I see is 3 deg for the alt and around 8 deg for the azimuth.
I think there would be no real difference for AHRS navigation and
astrometry, IMHO.

I find it so interesting that the spread of the points is quite
reproducable (not fully, but still)...  So I was nt thinking about a
statistical error but more a systematic error. And according you this could
be very well in the sensor. Correct?

I plan to have a chat with the Samsung support desk, I will see what info I
can get out of them (like which sensor they use in the Tab2). The
AndroSensor App does not provide any info on type of magnetometer.

So I need to have a device with EM7180 coupled with the MPU925. Do you know
something like that (in android/phone form)? Or do I need to use a Arduino
or Raspberry?

Thanks again for your help.

All the best,

Victor

On 13 February 2016 at 17:23, Kris Winer <notifications@github.com> wrote:

> Hi Victor,
>
> This is a very interesting application of attitude estimation! I hadn't
> though about using it for astrometry.
>
> I don't know the details of the tools you are using but I have some [...]

vreijs, 2016-02-13:
Hello Kris, Used CPU-Z App to determine the sensors and the magnetic senor
is MS-3E (YAS530) and the orientation sensor is MS-x.

Is this familiar?

All the best,

Victor

On 13 February 2016 at 17:54, Victor Reijs <web.victor.reijs@gmail.com>
wrote:

> Thanks Kris for your very valuable feedback.
> The stdev I see is 3 deg for the alt and around 8 deg for the azimuth.
> I think there would be no real difference for AHRS navigation and
> astrometry, [...]

kriswiner, 2016-02-13:
Hi Victor,

I use standard deviation but you are correct, the errors I see are rather
systematic. Especially in the Bosch sensors but also in the ST ones. The
heading precision is very good (I think his corresponds to the azimuth here)
but when estimating ninety-degree turns the heading estimation can be off by
ten degrees, reproducibly (I see the same in my cell phone). This is not
sensor jitter but rather a systematic bias that in principle could be
removed, but simple response surface recentering (hard iron calibration) and
axes scaling (poor man's soft iron correction) does not correct for it.

So yes, I would recommend an EM7180+MPU9250 solution. You can solder a
breakout board
(https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/)
with this solution to your favorite microcontroller. I have designed
integrated solutions for UAVs that have a microcontroller and EM7180+MPU9250
on board (with motor drivers and boost converters that might not be useful
to you). Here is an example:

https://forum.pjrc.com/threads/32985-Teensy-Flight-Controller

The advantage is a simple Arduino IDE can be used to write control software;
I have all the code necessary to get absolute orientation from this kind of
device.

I have a similar device using an nRF51822 BLE and ESP8266 wifi-enabled
controller.

I am interested in the astrometry application. Do you think there is a
commercial market for such a device? I mean like an instrumented finder
scope which reads out absolute alt-azi on a display, etc. Maybe we could
collaborate on development of such a device?

Kris

https://github.com/kriswiner/MPU-6050/wiki/Hardware-Sensor-Fusion-Solutions
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/E.-Typical-Resul
https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
https://cloud.githubusercontent.com/assets/17218070/13027981/3599099c-d259-

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183713975> .
<https://github.com/notifications/beacon/AGY1qjzJla60mpztwc6DbriGSukmFdCoks5
pj2VhgaJpZM4HZo42.gif>

kriswiner, 2016-02-13:
No, MS probably means Measurement Specialties, but I am not familiar with
the numbers. I'll poke around and see if I can find more info with this as a
start.

https://github.com/kriswiner/MPU-6050/wiki/Hardware-Sensor-Fusion-Solutions
https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/E.-Typical-Resul
https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
https://cloud.githubusercontent.com/assets/17218070/13027981/3599099c-d259-

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183716381> .
<https://github.com/notifications/beacon/AGY1qv04GAaQSaQm2LJhsQxOqW-scAibks5
pj2gGgaJpZM4HZo42.gif>

vreijs, 2016-02-13:
Hello Kris,

On 13 February 2016 at 18:16, Kris Winer <notifications@github.com> wrote:

> I use standard deviation but you are correct, the errors I see are rather
> systematic. Especially in the Bosch sensors but also in the ST ones. The
> heading precision is very good (I think his [...]

Great that you see similar behavior, that means that sensors react in a
similar way.
The 1 or 2 degrees you mention with the EM7180+MPU9250 is that the accuracy
or the precision? So does it have systematic erratic behavior or is it more
Gaussian errors?

> So yes, I would recommend an EM7180+MPU9250 solution. You can solder a
> breakout board
> (https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/
> )
> with this solution to your favorite [...]

In the past I was quite into soldering and experimenting (build my own
oscilloscope in the tube times and moving to the ZX81 times: build my own
speech synthesizer and software; all in Z80 assembler;-), but that is some
40 years ago. I envy the present times with all these options and
controllability...

> I am interested in the astrometry application. Do you think there is a
> commercial market for such a device?

Don't know. I know there was once (some 5 years ago?) a device that would
name stars by looking into it and pointing to part of the sky. Don't know
the name anymore (was build in USA), I think at that time the accuracy was
not great due to the sensors.  But with a 1 or 2 degrees precision (and
accuracy smaller than that), that would change things.
But a commercial market, I don't know! I am working in the archaeoastronomy
environment:
http://www.archaeocosmology.org/eng/archaeocosmology.htm
and
http://www.archaeocosmology.org/eng/aarde.htm
(even got an MA degree in this early last year;-)

But it is quiet difficult to make profit in that environment. Ease of use
is very important and that is what I like about this SunSurveyor App
(Android). It has augmented reality and the paths of the Moon and Sun
projected on the camera pictures (incl. equinox, and standstill events). So
nice in the field. The only problem is the accuracy of the device sensors
(you can't blame the App, although I have asked them to be able to offset
the azi and alt)...

> I mean like an instrumented finder
> scope which reads out absolute alt-azi on a display, etc. Maybe we could
> collaborate on development of such a device?
>

Any way, I am always interested in cooperating in this kind of projects, as
I think that they are valuable (not only for archaeoastronomy). For sure I
am willing to provide theoretical background and testing experience. I am
willing to distribute/advertise in my environment. But again, commercially
I don't know if it is viable (but perhaps I am now throwing in my own
glasses;--)...

Thanks again for your interest.

All the best,

Victor

vreijs, 2016-02-13:
See here:
http://download.yamaha.com/api/asset/file/?language=ja&site=jp.yamaha.com&asset_id=46264

On 13 February 2016 at 18:19, Kris Winer <notifications@github.com> wrote:

> No, MS probably means Measurement Specialties, but I am not familiar with
> the numbers. I'll poke around and see if I can find more info with this as
> a
> start.
>

kriswiner, 2016-02-13:
This is the accuracy. The precision is more like a tenth of a degree or less
depending on the mode.

The link I sent you has some results using this sensor fusion solution; you
can judge for yourself the quality of the output.

Kris

http://www.archaeocosmology.org/eng/archaeocosmology.htm
http://www.archaeocosmology.org/eng/aarde.htm

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183725768> .
<https://github.com/notifications/beacon/AGY1qncOWLvtBQtrBkqJbL1V9AZyxChBks5
pj3UMgaJpZM4HZo42.gif>

vreijs, 2016-02-13:
Hello Kris,

Thanks indeed for that like. I think my precision is a little bit
differently defined;-) So I have a camera picture and there is this (green)
cross in the middle of the picture. Now I get the Moon/Sun to this center
of this cross and then read the azi/alt. So my precision is including my
ability to oriented the cross on the moon/sun (and of course my hands are
not that stable, but similar to using an ordinary compass and
clinometer;-). But the device on a tripod would be better;-) (like I do
with my theodolite).

Attached is a screen shot (in the 4th quadrant of the cross you can see a
tiny piece of the Moon's crescent). The large Moon and the path is the
augmented reality of SunSurveyor.

Interested if we could make something accurate!

All the best,

Victor

On 13 February 2016 at 19:11, Kris Winer <notifications@github.com> wrote:

> This is the accuracy. The precision is more like a tenth of a degree or
> less
> depending on the mode.
>
> The link I sent you has some results using this sensor fusion solution; you
> can judge for yourself the quality of the output.
>

kriswiner, 2016-02-13:
Hi Victor,

Didn't get any attachment on the last e-mail.

Accuracy here is in absolute orientation. For navigation heading is the hard
part, and two degree heading accuracy is superb for a low-cost, small
footprint device.

Accuracy means when the device says it is pointed North, it is less than 2
degrees from True North, everytime, no matter the pitch and roll.

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183729817> .
<https://github.com/notifications/beacon/AGY1qtK_-gmgbOfKiNlN8m-Gzq1BjdQkks5
pj3vAgaJpZM4HZo42.gif>

vreijs, 2016-02-13:
I added the picture on the github site:
https://cloud.githubusercontent.com/assets/17218070/13029622/29c54132-d289-11e5-8d52-177abd3872a2.jpg

Indeed 2 degrees sounds good (4 times better then what I now have). A
normal compass is close to this also, might be somewhat smaller).
I assume the 2 degrees is the 68% (1 sigma) definition (assuming it is
normally distributed).

All the best,

Victor

kriswiner, 2016-02-13:
We haven't dome super precise characterization yet. I think the 2 sigma
accuracy is less than 2 degrees, but we still need to measure this. I wonder
if a small battery powered device affixed to a finder scope would give
sufficiently accurate tracking with this solution to be interesting?

https://cloud.githubusercontent.com/assets/17218070/13029622/29c54132-d289-1
1e5-8d52-177abd3872a2.jpg

https://github.com/notifications/beacon/AGY1qtK_-gmgbOfKiNlN8m-Gzq1BjdQkks5

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183749628> .
<https://github.com/notifications/beacon/AGY1qhVFGDRv79QT1C7lG23t1NzVns2Yks5
pj4tUgaJpZM4HZo42.gif>

vreijs, 2016-02-13:
Hello Kris,

On 13 February 2016 at 21:04, Kris Winer <notifications@github.com> wrote:

> We haven't dome super precise characterization yet. I think the 2 sigma
> accuracy is less than 2 degrees, but we still need to measure this.

Ok, so it looks good.

> I wonder
> if a small battery powered device affixed to a finder scope would give
> sufficiently accurate tracking with this solution to be interesting?
>

So you need a GPS, camera, a display and then overlay the location and/or
name of the celestial object. The camera needs to be light sensitive and
the display needs to have a good resolution. Of course it does not have to
be that accurate (due to the sensor's accuracy of around 1 or 2 degrees (1
or 2 sigma)).

For my applications: determining where the sun/moon (or other celestial
object) set/rise/culminates, etc) aka its path is; teh device can work at
the level of a normal phone camera/screen (aka using an App like
SunSurveyor, but with a better sensor).  Google Sky Map is another App
(although it does not augment [yet] the seen sky
http://google-sky-map.en.softonic.com/android ).

I found it again: https://en.wikipedia.org/wiki/SkyScout .  That web page
also mentioned the advance of the tablet/ipad... Knowing your competitor (
http://www.telescopes.com/blogs/helpful-information/18931204-celestron-skyscout-review
) is always good.

All the best,

Victor

kriswiner, 2016-02-14:
I had in mind a simple finder scope with a small electronic package
attached. The electronic package would report the altitude and azimuth where
pointed. If this were mounted on a simple tripod, the user could record
altitude, azimuth and time and get data to track any celestial object. If
the device was equipped with a button, the user could simply push it to
record these data and the data could be stored on the device for later
retrieval or could be immediately sent via BE or wifi to a host computer or
smartphone. The electronic device would be a module that could be attached
to a (most, any?) finder scope. The total cost would be tens of dollars with
the accuracy discussed previously.

I don't see any use for a camera or GPS in this use case.

Kris

http://google-sky-map.en.softonic.com/android ).

http://www.telescopes.com/blogs/helpful-information/18931204-celestron-skysc
out-review

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183765524> .
<https://github.com/notifications/beacon/AGY1qsY-U_nhBBj8tLezv4JF5axBInmtks5
pj6OwgaJpZM4HZo42.gif>

vreijs, 2016-02-14:
Indeed for that idea no camera is needed. To be able to determine what
celestial object is in the direction one needs to know the latitude,
longitude and time (so a GPS). All the best,

Victor

P.S. Do you know what magnetometer sensor is used in the iPads? It seems
that these devices are more accurate in the azimuth...

kriswiner, 2016-02-14:
I suppose a separate GPS would do. The device I have in mind would have a
very good clock (2 ppm) on board which could be calibrated against a GPS
time pulse. The idea is to get real time tracking information on a celestial
object onto a laptop or smartphone.

I don't know what is used on the iPhone. Generally smartphone sensors are
optimized for small size and ow power, not accuracy.

http://www.telescopes.com/blogs/helpful-information/18931204-celestron-skysc
https://github.com/notifications/beacon/AGY1qsY-U_nhBBj8tLezv4JF5axBInmtks5

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-183873184> .
<https://github.com/notifications/beacon/AGY1qqj0IMfcUsyGJpKLT2QC9jVZ1G6dks5
pkFN1gaJpZM4HZo42.gif>

vreijs, 2016-02-14:
Hello Kris,

The time does not need to be that accurate (seconds level is ok), but
knowing longitude and latitude are essential.
Here you see the SkyScout mounted on a telesope. SckyScope are offered
around 80$ on ebay.
All the best,

Victor

On 14 February 2016 at 18:32, Kris Winer <notifications@github.com> wrote:

> I suppose a separate GPS would do. The device I have in mind would have a
> very good clock (2 ppm) on board which could be calibrated against a GPS
> time pulse. The idea is to get real time tracking information on a
> celestial
> object onto a laptop or smartphone.
>

vreijs, 2016-02-17:
Hello Kris,
In the datasheet of the MPU-9205 I don't see the heading accuracy (
http://store.invensense.com/datasheets/invensense/MPU9250REV1.0.pdf ). Do
you know the specs for this (I know what you measured). For the Project
Tango Tablet development kit the BMX055 has a 3 deg heading accuracy. For
the Samsung Tab2 with YAS530 has a 5 deg sensitivity.

Would be great to see the official accuracy of the MPU 9205. Thanks.

All the best,

Victor

kriswiner, 2016-02-17:
Hi Victor,

Heading accuracy really doesn't apply to a 9 DoF sensor, that's why
Invensense doesn't quote it. The "accuracy" quoted for the BMX055 and YAS530
are more likely what I would call jitter or Gaussian error. I have tested
the BMX055 and I can tell you the heading accuracy is more like 7 or 10
degrees RMS.

Using the EM7180 sensor hub with it's sensor fusion filtering and the
MPU9250 sensor data as input to the sensor hub I can routinely get 2 degree
or better heading accuracy, with a heading jitter of a tenth of a degree or
less.

This is a far superior absolute orientation solution to any that can be
achieved with either of the mags you mention.

Kris

http://www.telescopes.com/blogs/helpful-information/18931204-celestron-skysc
https://github.com/notifications/beacon/AGY1qsY-U_nhBBj8tLezv4JF5axBInmtks5
https://github.com/notifications/beacon/AGY1qqj0IMfcUsyGJpKLT2QC9jVZ1G6dks5

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-185145473> .
<https://github.com/notifications/beacon/AGY1qnv3LdNxmFIVq2ShSFkEI0QFsNxDks5
plERogaJpZM4HZo42.gif>

vreijs, 2016-02-17:
Thanks Kris,

On 17 February 2016 at 16:22, Kris Winer <notifications@github.com> wrote:

> Heading accuracy really doesn't apply to a 9 DoF sensor, that's why
> Invensense doesn't quote it. The "accuracy" quoted for the BMX055 and
> YAS530
> are more likely what I would call jitter or Gaussian [...]

I see value range of 24 deg (min to max). And this results (assuming the 24
degrees is twice 3 sigma) in a 4 deg (1sigma) which is close to the 5
degrees specified in YAS530.I got data from a friend who has an iPad and he
sees a range of around 10 degrees (so more or less half what my Samsung
does).
It is always difficult to determine what people mean with
accuracy/precision (I always use in my evaluations the 1 sigma value).

> Using the EM7180 sensor hub with it's sensor fusion filtering and the
> MPU9250 sensor data as input to the sensor hub I can routinely get 2 degree
> or better heading accuracy, with a heading jitter of a tenth of a degree or
> less.

So I hope that your 2 degrees means a reduction of a factor of around 3 to
4 compared to what I see in the YAS530.... I really would need to use my
workflow on the MPU-9250 to to be sure we compare like with like.

Do you know which tablets/phones have this sensor? I have also asked this
question to Invensense (no answer back yet).

All the best,

Victor

vreijs, 2016-02-17:
Seeing this datasheet (that include the AK8963, which is 'part' of the
MPU-9250:
http://www.invensense.com/products/motion-tracking/9-axis/mpu-9250/ )
http://www.pnicorp.com/wp-content/uploads/Sentral-MandM-Technical-Datasheet_rG.pdf
the heading accuracy is 2 deg (general description and section 2.1).

On 17 February 2016 at 16:35, Victor Reijs <web.victor.reijs@gmail.com>
wrotr
> Using the EM7180 sensor hub with it's sensor fusion filtering and the
>
>
> So I hope that your 2 degrees means a reduction of a factor of around 3 to
> 4 compared to what I see in the YAS530.... I really [...]

vreijs, 2016-02-17:
How can I reach you privately? Can I send a private message on GitHub? I got feedback from Invensense Tech Support on my query of heading accuracy and phone/tablet.

kriswiner, 2016-02-17:
PM me at onehorse@earthlink.net

vreijs, 2016-02-23:
Hello Kris,

Did more measurements and the picture is perhaps becoming clearer. The
range of azimuths observed for each set of measurements is more a sinus
(but with an argument double the azimuth). This 'range of azimuth' observed
shows up when rotating the device (and keeping the heading direction
towards Moon or Sun).
I think it is not related to a possible mounting error of the camera (that
would not behave in that way).

This dependency of double the azimuth is interesting.(this could be due to
squaring a sinus/cosinus). So where is this coming from?
I have the idea it is not relating to the App (Geosurvey, Measure Angle and
SunSurveyor seem to behave similarly), so it must the code that determines
the azi/alt from all the sensor data.

Interesting and luckily it looks to converge to something that could be
explainable.

Are you able to perform similar type of accuracy measurements with your
device? So I have a heading to a far away object (at least 200m) and
measured the azi/alt; rotate the device 90 deg (keeping heading the same)
and again measure azi/alt and keep going until back in the first
orientation.
The azi/alt should stay the same regardless of the rotation of the device.
But in my case the azimuth varies considerable (the range), as shown in the
graph and thus depending on the avg azimuth.

All the best,

Victor

vreijs, 2016-02-23:
Hello Kris,

<I first thought that my measurements has a periodicity of double 'avg azi', but that was because of the formula I used in my evaluation, which resulted in an abs function (and that becomes a double 'avg.azi' function)>

So reevaluated my measurements. If I take the difference of azimuths between the landscape (0 deg rot) and -landscape (180 deg rot) measurements, I get the following picture:
[image]
Why this sinus behavior is seen within landscape rotation, I still don't know...

If I take the difference of azimuth for the portrait (90deg rot) and -portrait (270deg rot), I get:
[image]
This is more or less a flat graph (so no dependency of the azimuth range and the avg. azimuth when using portrait mode).

So when using portrait mode it looks that the systematic error is more random (1 sigma=~5 deg),...

Still something to find out (why this sinus-behavior?).

All the best,

Victor

kriswiner, 2016-02-23:
Hi Victor,

Your results here are pretty typical of the average sensor fusion solution available in a smartphone. The roll and pitch (corresponding to the altitude) is generally good to within a couple of degrees but the heading (corresponding to the azimuth) can be off by 7 or 8 degrees RMS or standard deviation. This is why we developed the Ultimate Sensor Fusion Solution I mentioned, where we can routinely get 1 degree accuracy in altitude and 2 degree accuracy in azimuth. You simply cannot get this performance with any commercial smartphone implementation to my knowledge.

The reason has to do mostly with the sensors themselves, such that with even the best calibration and sensor fusion filtering the limits to azimuth accuracy remain at 5 degrees (at best) but more likely 10 degrees. The root cause is non-spherical response surfaces of one or more sensors that cannot be corrected with the available calibration. It is possible to calibrate them beyond what is available natively on the tablet but there will remain an inherent error in many of the sensors chosen for their low-power, small size, and low cost in modern smart devices and tablets. In other words, the sensors are not chosen for their excellent performance.

vreijs, 2016-02-23:
Hello Kris,
Can you provide information about the accuracy/precision of your device? I
am really interested to see comparable data.  Remember that using my tablet
I get a precision of 0.5 deg (and I repeated that precision some 200
times). So it is the measurement methodology one use to determine accuracy.
If wanted I can provide guidance about the measurement methodology. I am
sure that it can be better (and I also really hope so!), and that is why I
ask if you can provide numbers for my measurement method (same heading and
rotate the device 90 degrees) .I just want a good methodology to define
accuracy and precision.

All the best,

Victor

P.S. I think that my analysis is regardless of App and/or Android/IPad. But
I will now concentrate on the Iphone/IPad (although I don't have such a
device).

kriswiner, 2016-02-24:
Hi Victor,

You can see the results of simple testing here:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/E.-Typical-Results-Using-the-SENtral

Kris

vreijs, 2016-02-24:
Hello Kris, In that link  you rotate the device while rotating along the
z-axis
Correct? I am rotating along the heading axis. So the heading alt/az should
not change, as the heading is always he same. but the device is rotated in
steps of 90 deg.
Do you know how I measure it?

kriswiner, 2016-02-24:
Hi Victor,

I do rotate about the z-axis mostly, but at the end of the first test, I
rotate in pitch by 45 degrees, then yaw by 45 degrees to show that the
heading remains more or less fixed. This show the orientation solution is
mostly free of cross axis effects. Tablets will fail this test.

Kris

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/E.-Typical-Resul
ts-Using-the-SENtral

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-188104581> .
<https://github.com/notifications/beacon/AGY1qnj2gGCQJDxgCSNouhQ8o1Yw6AUSks5
pnUfrgaJpZM4HZo42.gif>

vreijs, 2016-02-24:
You did this yaw of 45 by hand. What if you have a stable surface say a
slanted table (at 45 degrees). And you put your device there and then turn
it in steps of 90 degrees around the heading axis. I am interested what you
get there.
I will also do that with my tablet (now I used the Sun/Moon), but using a
slanted surface is much more stable of course (my hand held variation gives
some 0.5 degree precision). But I will be interested to see what the
accuracy will be (now only find a slated outside surface without iron parts
in it).

Thanks for your patience.

All the best,

Victor

vreijs, 2016-02-25:
Will do this experiment tomorrow (I devised the setup in my garden;-). I
think the sinus behavior I see between measured azimuth and real azimuth,
could be very well linked to were I started my original quest: to
understand soft iron compensation. The sensor/MD you are using compensates
for hard and soft iron effect. I am really wondering if my tablet is doing
the soft iron one (i asked this several weeks ago, but never got a good
answer). I think it is not doing that.
What I now see, is very well related to that effect I think. I now can
reduce my workflow and repeat these measurements also for the Air2, iPhone5
and iPhone 6s plus to see how/what they do.

I wish I could program my own hard/soft iron compensation methods on my
Android table... I also have the idea that my tablet ('s sensor) does
calibration constantly; and thus it gets uncalibrated when doing erratic
movements.

All the best,

Victor

On 24 February 2016 at 08:55, Victor Reijs <web.victor.reijs@gmail.com>
wrote:

> You did this yaw of 45 by hand. What if you have a stable surface say a
> slanted table (at 45 degrees). And you put your device there and then turn
> it in steps of 90 degrees around the heading axis. [...]

kriswiner, 2016-02-25:
The math for hard iron is pretty simple, and you can even make a crude correction for soft iron. There is an algorithm published as part of the mag calibration article I wrote and whose link I forwarded to you. How to code this into an Android device I do not know...

vreijs, 2016-02-25:
I have this link already some time, which I thought was quite overarching:
http://www.nxp.com/files/sensors/doc/app_note/AN4246.pdf
I always though that this was included in Android, but it seems to be more
a part of the motion device aggregator (if present in the device).

vreijs, 2016-02-27:
Hello Kris,
Just looked at turning my device 90 degrees, and like you said there is
considerable change between these readings: a standard deviation of around
6 degrees around the 90 degrees. So close to the stand. dev seen in my
measurements when rotating.
I assume this is still due to the lacl of soft iron compensation.

All the best,

Victor

P.S. still need to get my hands on Air2, 6splus and 5 (but might be over a
few weeks due to a lot of travel on my side).

kriswiner, 2016-02-27:
Partly, although many mags have relatively incurable bias in their response
surface. Meaning that the response surface shape changes depending on
orientation making it very difficult to compensate for even with good soft
iron algorithms.

The technical term for this behavior is "crappy sensors".

Kris

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-189700753> .
<https://github.com/notifications/beacon/AGY1qg57tYWwVix0DMiRCW4tlOYq2OReks5
poeZTgaJpZM4HZo42.gif>

vreijs, 2016-02-28:
I wish yur sensor/MD processor combination was implemented in some
phone/tablet.

On 27 February 2016 at 19:10, Kris Winer <notifications@github.com> wrote:

> The technical term for this behavior is "crappy sensors".
>

kriswiner, 2016-02-28:
It's in the Sony smart watch.

But my suggestion was to design a purpose-built device for doing what you
want to do. Phones and tablets are convenient, yes, but they will never do
any one thing extremely well, they do many things well enough.

If you have interest in this please let me know.

Kris

<https://github.com/kriswiner/MPU-9250/issues/42#issuecomment-189819766> .
<https://github.com/notifications/beacon/AGY1qhM1fy3llk5wJynthYrWzKfKP_9eks5
poqm1gaJpZM4HZo42.gif>

vreijs, 2016-02-28:
For me the augmented reality is essential (so projecting path of celestial
object) on reality. Like sunsurveyor http://www.sunsurveyor.com/