kriswiner/MPU9250 #43: Ground plane under the chip

empyre, 2016-02-27:
Hello kriswiner,

I find all your MPU-9250 board were designed without ground plane under the chip. In datasheet of MPU-9150, the Assembly Precautions section point out

>  The exposed center die pad(e-pad or thermal pad) is not required for heat sinking, and should not be soldered to the PCB.

But it also point out 

>  For best performance, design a ground plane under the e-pad to reduce PCB signal noise from the board on which the gyro device is mounted. If the gyro device is stacked under an adjacent PCB board, design a ground plane directly above the gyro device to shield active signals from the adjacent PCB board. 

Does it mean a ground plane under the e-pad is necessary but should not be soldered to the e-pad?

I recently contacted Invensense, they said the same design rules are still applicable to MPU-9250. Your mini MPU9250 Teensy 3.1 add-on shields are directly mounted under Teensy board. If without a ground plane between Teensy and MPU9250, would the performance of gyro in MPU9250 affected by Teensy board?

Thanks

kriswiner, 2016-02-27:
Any copper underneath the QFN package is undesirable for best magnetometer
performance. In general for all MEMS sensors, avoid copper directly beneath
the package (on the top layer). This means no pad underneath the chip. If
you don't care about the mag performance, you may put a GND layer on the
bottom-most layer but I would not recommend it. The performance of the mag
and gyro is best with no copper on any layer directly beneath the QFN
package outline.

You do want to make sure your pcb design has good ground planes around the
chip and that the power ports are properly bypasses with good quality 100 nF
16 V capacitors. Route I2C or SPI traces away (i.e., bottom of board if
possible) from the analog power pins (on top layer), keep vias away from the
QFN package as well as buttons, switches, mounting holes, etc.

Best results are obtained with a four layer board design with a dedicated
ground and power plane in the middle two layers, with signal layers on top
and bottom with as much GND copper pour on top and bottom as you can muster.

See: 

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/B.-SENtral-Refer
ence-Design

for an example of good design practice using the MPU9250.

Kris
Pesky Products

<https://github.com/kriswiner/MPU-9250/issues/43> .
<https://github.com/notifications/beacon/AGY1qnl3u2JLNXxeR2DtTscVwyKnpCCbks5
poOXvgaJpZM4HkTZi.gif>

empyre, 2016-02-27:
Thanks for the suggestions.

closed 2016-03-02 by empyre

khalil50, 2017-09-27:
[image]
Hi kriswiner !! 

i have routed an mpu9250 circuit on pcb board!! But i can't get connectoin with the mpu 9250. this my circuit's schematic and my board . can you help ) 
[image]

kriswiner, 2017-09-27:
Are you saying you designed and assembled this circuit and it doesn't work?

There are a lot of things wrong here.

What CAD program are you using?

khalil50, 2017-10-02:
Hi kriswiner,
I changed my schematic and the pcb, could you please tell me if there is an error ?
[image]
[image]

khalil50, 2017-10-02:
I have change again the pcb with the recommandation of the MPU9250 datasheet,
Could you comment this please,

Thank you for your help
[image]

kriswiner, 2017-10-02:
Better:

Pin 20 needs to be connected to GND IIRC.

Pin 22 does not require a pullup, connect directly to 3V3.

Pin 11 does not require a pull down, connect directly to GND.

No copper underneath the package on any layer, do not include a ground pad,
no traces of any kind below the package on any layer.

Move cap closer to pin 15, connect pin to 3V3 on other side of cap. In
other words, the pin should see the cap not 3V3 trace like pin 8.

Use 10 mil trace for pin 10.

khalil50, 2017-10-03:
Thank you so much kris for your response,
Do you mean pin 13 intead of pin 15 ? Because in the datasheet the pin 15 is NC.

kriswiner, 2017-10-03:
Yes, pin 13, the one with the cap.