kriswiner/MPU9250 #44: When IMU moves in 2-D Plane (YZ), accelerometer measures acceleration in third axis (X axis) too?

mohit5037, 2016-02-27:
I am moving my IMU in YZ plane but it accelerometer is measuring considerable acceleration in X axis as well which very closely follows acceleration curve of Y axis.

I am drawing a semi-circle in YZ plane with y-axis as diameter of semi-circle.

Here is the gravity free acceleration curves of all 3 axes:

[image]

You can see that acceleration in x-axis is of considerable magnitude, so I can't treat it as a noise.

Please help as I am stuck with this problem since 10 days.

mohit5037, 2016-02-27:
I am using MPU 9250 with Arduino Uno Board.

kriswiner, 2016-02-27:
Did you calibrate your accelerometer to at least remove the offset bias? How
did you do it?

measures acceleration in third axis (X axis) too? (#44)

<https://github.com/kriswiner/MPU-9250/issues/44#issuecomment-189651354> .
<https://github.com/notifications/beacon/AGY1qgDgj-rrzFE5F2jTeaCWXTVd47Vkks5
poamfgaJpZM4Hkf1B.gif>

mohit5037, 2016-02-27:
Hi kris,

Yes I did calibrate my accelerometer by uploading the raw sensor data code to board and taking mean of 1000 values after 30-40 secs when it was stabilised.

Then I called the setAcceloffset() functions with those mean values to set accelero and gyro offsets. Then I again checked mean and they were close to zero except Z which was close to 16384.

Thanks

kriswiner, 2016-02-27:
Close to zero should mean a few milliG on x and y. Even then, one expects
some cross axis contamination at about 1% or maybe less. I haven't seen this
kind of behavior but I usually do not look at just acceleration.

One thing, is the MPU9250 mounted at the center of your board? Because
rotations about some radius even without any linear acceleration will show
up as a linear acceleration component.

<https://github.com/kriswiner/MPU-9250/issues/44#issuecomment-189697512> .
<https://github.com/notifications/beacon/AGY1qtAAU7z22nztK0NZqOTEo-5VkAjmks5
pod4lgaJpZM4Hkf1B.gif>

mohit5037, 2016-02-28:
Hi Kris,

Please refer to the image below to get an idea of the movement. I am keeping the platform on which sensor is pasted parallel to ground always. So ideally there is no movement along any of the axes.

[image]

kriswiner, 2016-02-28:
I am not sure I know how to help you. If the sensor has a large cross-axis
component to the response there is nothing you can do in software, and your
only other choices are to try a different MPU9250 or a different sensor
altogether. You might try turning off the mag and gyro if you only want
acceleration. These also couple to the accelerometer but usually the effect
is tiny.

board
<https://cloud.githubusercontent.com/assets/8141787/13379290/aa817790-de47-1
1e5-8641-65de2d741789.png> 

-
<https://github.com/kriswiner/MPU-9250/issues/44#issuecomment-189864294> .
<https://github.com/notifications/beacon/AGY1qtlBn1Cw3IEZVb8PAP5QBj1q3x31ks5
pouQmgaJpZM4Hkf1B.gif>

mohit5037, 2016-02-29:
Hi Kris,

I know it is very difficult to understand the problem theoretically but still I really appreciate your support. By the way here is my code in case you want to have look, I have just removed the display code and changed the range of acceleration.

/* MPU9250 Basic Example Code
 by: Kris Winer
 date: April 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 10k resistors are on the EMSENSR-9250 breakout board.
 
 Hardware setup:
 MPU9250 Breakout --------- Arduino
 VDD ---------------------- 3.3V
 VDDI --------------------- 3.3V
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND
 
 Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 
 Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
 */
#include <SPI.h>
#include <Wire.h>   
#include <Adafruit_GFX.h>
//#include <Adafruit_PCD8544.h>

// Using NOKIA 5110 monochrome 84 x 48 pixel display
// pin 9 - Serial clock out (SCLK)
// pin 8 - Serial data out (DIN)
// pin 7 - Data/Command select (D/C)
// pin 5 - LCD chip select (CS)
// pin 6 - LCD reset (RST)
//Adafruit_PCD8544 display = Adafruit_PCD8544(9, 8, 7, 5, 6);

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define WHO_AM_I_AK8963  0x00 // should return 0x48
#define INFO             0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L	 0x03  // data
#define AK8963_XOUT_H	 0x04
#define AK8963_YOUT_L	 0x05
#define AK8963_YOUT_H	 0x06
#define AK8963_ZOUT_L	 0x07
#define AK8963_ZOUT_H	 0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
#define AK8963_ASAZ      0x12  // Fuse ROM z-axis sensitivity adjustment value

#define SELF_TEST_X_GYRO 0x00                  
#define SELF_TEST_Y_GYRO 0x01                                                                          
#define SELF_TEST_Z_GYRO 0x02

[... 903 lines cut ...]
	Wire.write(data);                 // Put data in Tx buffer
	Wire.endTransmission();           // Send the Tx buffer
}

        uint8_t readByte(uint8_t address, uint8_t subAddress)
{
	uint8_t data; // `data` will store the register data	 
	Wire.beginTransmission(address);         // Initialize the Tx buffer
	Wire.write(subAddress);	                 // Put slave register address in Tx buffer
	Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
	Wire.requestFrom(address, (uint8_t) 1);  // Read one byte from slave register address 
	data = Wire.read();                      // Fill Rx buffer with result
	return data;                             // Return data read from slave register
}

        void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
	Wire.beginTransmission(address);   // Initialize the Tx buffer
	Wire.write(subAddress);            // Put slave register address in Tx buffer
	Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
	uint8_t i = 0;
        Wire.requestFrom(address, count);  // Read bytes from slave register address 
	while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
}

kriswiner, 2016-02-29:
I noticed you have kept the magnetometer biases I measured for my particular MPU9250 unit. These are most unlikely to be the proper ones for yours. If you are only using the accelerometer and have turned off the magnetometer, then this is not an issue. If you are doing some kind of sensor fusion you will almost be guaranteed to get a bad result unless you substitute your own offset biases into the code.

mohit5037, 2016-03-03:
Thanks Kris, I will calibrate my magneto as per your magneto calibration tutorial.