kriswiner/MPU9250 #45: Quaternions get unstable when Pitch approaches +90 or -90?

mohit5037, 2016-03-03:
Hi Kris,

I am trying to get quaternions from the MadgwickQuaternionUpdate Function. I am not converting them into euler angles as I know they suffer from gimble lock problem.

So at the start I am getting nice quaternions like [1,0,0,0] but when I increase the pitch of my sensor to 90 degree I am expecting to get quaternions like [0.707, 0, 0.707, 0] but I am not getting them instead random quaternions start developing in yaw and roll axes.

I don't know why this is happening, pls help in you have any idea about this. Thanks

kriswiner, 2016-03-04:
Are your sensors calibrated?

gjwo, 2016-11-30:
I think this may be similar to issue #93 which I have just raised.