kriswiner/MPU9250 #46: Joint angle calculation

ijunglee, 2016-03-07:
I want to attach the 9-axis imu on finger to calculate the joint angles. I have used  Madgwick Algorithm and got the quaternion of imu after each update. I know the quaternion can show the attitude of each sensor's rotation without gimbal lock, but how can I calculate and show the angle between two imu? When my hand back is toward up, I can simply use the difference between euler angles of two imu.(roll of imu1 - roll of imu2) However, when my hand back toward left or right, I can not get the correct angle. Do you have any ideas about solving this problem? Thank you very much.

kriswiner, 2016-03-07:
You can use the Euler angles but they won't be correct unless you properly
calibrate your sensor first.

How did you perform the calibration?

<https://github.com/kriswiner/MPU-9250/issues/46> .
<https://github.com/notifications/beacon/AGY1qq1SvGmSnduUtnOfi5w6w55bj7T6ks5
pq_MQgaJpZM4Hqnz0.gif>

ijunglee, 2016-03-08:
For magnetometer, I rotated and moved the sensor as 8-shape to find the min-max value in each axis. After getting the min-max values, I average them as the offset in each axis.

For accelerometer and  gyroscope, I put the imu on the table(with Z-axis toward up), and average every input (maybe receive 1000 records). For accelerometer, I remove (0g, 0g, 1g) to get the offset in each axis because this is the ideal value for acceleration when Z-axis is toward up. For gyroscope, I think when the imu is stable, the value should be (0, 0, 0) rad/s. So the average values are offset for gyroscope. (Is this correct?)

This is what I perform for calibration, I am not sure the method is correct. Please give me advice, thanks for a lot!

kriswiner, 2016-03-08:
Yes, this is correct. Can you verify that the magnetic response surface is
centered at the origin after calibration? Plot Mx vs Mz and Mx vs My. Are
the plots a spherical cloud or do you have a lot of ellipsoidal variation?

To get adequate Madgwick sensor fusion the algorithm has to converge for ach
data set. This means the algorithm should be running at four or five times
the rate of the gyro sample rate. So if you choose a gyro sampl rate of 200
Hz with 40 Hz bandwidth (a good choice) your processor should be able to run
the sensor fusion algorithm at 1 kHz or so.

What processor are you using?

<https://github.com/kriswiner/MPU-9250/issues/46#issuecomment-193567775> .
<https://github.com/notifications/beacon/AGY1qhlE4JKjzd1udZBF05Q_emSlxCDTks5
prOHlgaJpZM4Hqnz0.gif>

ijunglee, 2016-03-08:
1. Thanks for your response. I will try to verify the magnetic data.

2. I use the MSP430F5438A. How do I choose the sampling rate and bandwidth of the acc, gyro and mag? I used the LSM9DS0 sensor for the same project before. And I use the sampling rate 800 Hz for accelerometer, 760 Hz for gyroscope, and 100 Hz for magnetometer. However, I am not sure how to choose the proper bandwidth.

Thanks for your kindly help. I look forward to your reply.

kriswiner, 2016-03-08:
These can all be chosen in the MPU9250 initialization function.

<https://github.com/kriswiner/MPU-9250/issues/46#issuecomment-193579671> .
<https://github.com/notifications/beacon/AGY1qkVSO8PLV-pJ2Kd4p7TaFkIocK-Pks5
prOgqgaJpZM4Hqnz0.gif>

ijunglee, 2016-03-08:
Thanks for your reply. I will try it.
And what are the proper sampling rate and bandwidth of acc and mag?

kriswiner, 2016-03-08:
200/40 Hz for accel, 100 Hz for the mag although 8 Hz should work well too.

<https://github.com/kriswiner/MPU-9250/issues/46#issuecomment-193589422> .
<https://github.com/notifications/beacon/AGY1qnjW7UHTc3hkENObXM2_c-BkqpdCks5
prO8vgaJpZM4Hqnz0.gif>

ijunglee, 2016-03-08:
Thanks for your response. I will try it.
Do you have any idea about how to get the joint angle between 2 sensors using euler angle? Thank you very much.

kriswiner, 2016-03-08:
This is trivial if in the same plane, otherwise requires some
straightforward trigonometry.

<https://github.com/kriswiner/MPU-9250/issues/46#issuecomment-193621730> .
<https://github.com/notifications/beacon/AGY1qlCo-aqYxPtCLfIQhFwLwT5igIzwks5
prRMbgaJpZM4Hqnz0.gif>

ijunglee, 2016-03-16:
Hello, this is the before and after of my magnetometer calibration. I found that the data form more like a sphere after calibration. Is the result correct?
figures up are Mx vs Mz and Mx vs My without calibration, below are after calibration
[MagCal.pdf](https://github.com/kriswiner/MPU-9250/files/175552/MagCal.pdf)

ijunglee, 2016-03-16:
Sorry, I think the result is much better.
[image]

kriswiner, 2016-03-16:
Yes, this is much better. Make sure the initial bias values are 0x8000 for max and 0x7FFF for min.

thisum, 2018-12-09:
@ijunglee, were you able to calculate the joint angle with quaternions or euler angles?

ijunglee, 2018-12-09:
> @ijunglee, were you able to calculate the joint angle with quaternions or euler angles?

I use quaternions to calculate the inner product of two quaternions (as vectors) to get the joint angle. You can get a good result if your magnetometer is well-calibrated. But I am not sure if the method is correct, maybe we can have more discussion.

I also try the function [Quaternion.Angle](https://docs.unity3d.com/ScriptReference/Quaternion.Angle.html) in Unity 3D...I think it also uses the similar method.

thisum, 2018-12-09:
I tested the calibration as described above and seems working fine. But the problem I'm having is to get the joint angle. I used `q_relative = q_shin*qt.q_inv(q_thigh)` to get the relative quaternion of the leg. And my plan was to calculate Euler angles from that. But it's not the correct way. Still trying to figure out the appropriate way.

Inner product is cross product?

kriswiner, 2018-12-09:
Why not simply use the Euler angle differences?

thisum, 2018-12-09:
One problem is the gimbal lock. For example, I'm trying to measure the angle of the knee with respect to the thigh IMU. Let's say the knee angle is 90 degrees. while having the angle fixed, I move the leg in the 3D space, even lying on the ground. So all these times, the knee angle is 90.

anziboy, 2019-05-06:
Hi, @ijunglee, I am having the same problem for when I'm trying to measure the absolute knee angle using two BNO055 adafruit sensors on arduino. I have fully calibrated both of the sensors and I'm getting the pitch, yaw and roll angles, however my angle values are incorrect. Any thoughts or tips on how I can get the correct ones? Thank you so much ,

best wishes, Anza

HesamB97, 2024-10-10:
Hi @kriswiner @ijunglee 
I hope you are doing well

I have learned a lot from the discussion you had. I am currently working on my PhD project, and part of my objective is to calculate the ankle joint angle. For this purpose, I used two IMUs attached to the foot and shin. Both sensors are commercial devices from WitMotion. Since I am relatively new to this field, I relied on the accelerometer calibration option in the company’s app for data collection, and due to electromagnetic interference, I only used the gyroscope and accelerometer. To obtain drift-free angles for both sensors, which I tried to align in the same plane and parallel, I used the Madgwick filter. By subtracting the Euler angles obtained from the two sensors, I calculated the joint angle. These angles appeared to have a logical trend with correct upper and lower limits, so I did not question them. However, recently, upon reviewing the concept of Euler angles, I realized that these angles are defined with respect to an inertial frame. As a result, I started to doubt whether subtracting the Euler angles from the two sensors yields angles in the correct planes, or if the resulting profile is only coincidentally correct. For instance, if the subject to whom the sensors are attached changes direction during motion, am I still able to extract the correct knee joint angles? I would like to hear your opinion on this matter.

Best Regards
Hesam

kriswiner, 2024-10-10:
Best approach would be to construct a jointed mechanical fixture where the
exact joint angles can be independently measured and check this against
your measurement method...

I would think that with just the accel and gyro you would have unacceptable
heading drift that would make such measurements inaccurate after a very
short time. I think you need to add a magnetometer and use absolute
orientation estimation for any kind of accuracy.

This <https://www.tindie.com/products/onehorse/usfsmax-module-and-carrier/>
is what we have developed and used for such measurements.

HesamB97, 2024-10-11:
@kriswiner 
Thank you very much for your quick response.
I will make a fixture as a gold standard solution as you suggested. Just as the last question, even if I use a 9-axis IMU to estimate the orientation, is that correct to calculate the hinge joint angle based on the subtraction of deriven Euler angles of 2 IMUs?
BW
Hesam

kriswiner, 2024-10-11:
Generally, subtracting the two sets of Euler angles is not going to give
you the proper result. To find the angle between two vectors in 3-space you
need to use either the dot or cross products...this is elementary
<https://www.wikihow.com/Find-the-Angle-Between-Two-Vectors>
 trigonometry....