kriswiner/MPU9250 #47: Cannot connect to MPU-9250. It returns 0x73 instead of 0x71

Poley09, 2016-03-07:
So everytime i try to connect, it prints this message - 

MPU9250 I AM 73 I should be 71
Could not connect to MPU9250: 0x73

I have completed i2c scans and received raw data from my sensor before so i'm not sure what i'm doing wrong. Thanks in advance

kriswiner, 2016-03-07:
Change the check to 0x73 and see if the sensor data appears. It's likely you
changed a register address or have some other typo there.

0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47> .
<https://github.com/notifications/beacon/AGY1qknCKI7YEQekWKGRA3uH69f5888sks5
prGqNgaJpZM4HrByj.gif>

Poley09, 2016-03-07:
it now displays 

MPU9250 I AM 73 I should be 73
Could not connect to MPU9250: 0x73

Poley09, 2016-03-07:
https://github.com/Poley09/9250/tree/master/_9250_code

thats what im using, it should be the exact same without the display.

kriswiner, 2016-03-07:
It checks that the byte read in the register is what is expected. Change the
expected value to 0x73 not the display value.

of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-193386458> .
<https://github.com/notifications/beacon/AGY1qioCCYnYeX8ddg5gK8c9N4vF09W7ks5
prHBLgaJpZM4HrByj.gif>

kriswiner, 2016-03-07:
// Read the WHO_AM_I register, this is a good test of communication

 

 

 

byte c = readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250); // Read WHO_AM_I
register for MPU-9250

 

 

Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX);
Serial.print(" I should be "); Serial.println(0x73, HEX);

 

 

 

 

 

if (c == 0x71) // WHO_AM_I should always be 0x68

 

 

{ 

 

 

Serial.println("MPU9250 is online...");

 

 

 

Change this to 0x73; what MPU9250 module are you using?

of 0x71 (#47)

 

https://github.com/Poley09/9250/tree/master/_9250_code

 

 

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-193388886> .
<https://github.com/notifications/beacon/AGY1qhZx1ae3FrYo0mxA9P4Dk0ER079gks5
prHIJgaJpZM4HrByj.gif>

Poley09, 2016-03-07:
Yep, sorry I was being stupid and looking in the complete wrong part, it all works now thank you

Poley09, 2016-03-07:
I was wondering if you could help me with another problem. when perfectly still at rest, i get values around yaw = 128.5 pitch = 3.2 and roll = -1/1 how would i go about getting them to 0 on rest? and im wanting them to control pan/tilt servos with yaw and pitch. However the yaw doesnt seem to be changing when moving it in any direction not just the y axis. i dont know whether there is a mistake somewhere as my pitch and roll seem to be mixed up aswell. thanks

kriswiner, 2016-03-07:
How did you calibrate your sensors?

of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-193496665> .
<https://github.com/notifications/beacon/AGY1qnESWBJ4sWGu-RDKwUuO2bPvLpOQks5
prK_UgaJpZM4HrByj.gif>

Poley09, 2016-03-07:
not correctly im guessing, im not too experienced with these sensors. do you have an email address i could speak to you on? or send one to matthewpole@hotmail.co.uk and ill reply back to that? please and thank you.

kriswiner, 2016-03-08:
You can PM me at onehorse@earthlink.net.

Take a look here:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration

Kris

of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-193513754> .
<https://github.com/notifications/beacon/AGY1ql7igXZVR7IMUusb_H-XgFAJdFM7ks5
prLttgaJpZM4HrByj.gif>

whatnick, 2016-05-03:
I am having a similar issue. I can communicate with the board. But I get 0xA0 as the who am i value. What can change this ?

kriswiner, 2016-05-03:
There is something very wrong if you are not reading 0x71. Can you verify with an I2C scanner that you are getting an ACK from the MPU9250? You should see 3V3 on the SDA/SCL lines and finf I2C device 0x68 or 0x69 with an I2C scan.

whatnick, 2016-05-03:
Hi Kris,

Thanks for your response. I do get an Ack from 0x68. Which indicates
something else is wrong with the chip or it is a counterfeit. I will try
another from the batch then get more from a different sipplier.

kriswiner, 2016-05-04:
It could be a bad board but if you are getting an I2C ACK this is unlikely;
it could still be that you have something wrong with the I2C read software,
or you are not supplying the proper voltage to the chip. Can you check these
things? Can you write to a control register and read it back to verify that
you are able to write?

What do you get when you read the whoami register? 0xFF?

0x73 instead of 0x71 (#47)

<https://github.com/notifications/beacon/AGY1qui4JfwoafegUzLKTgQ-Hb2CkfU-ks5
p99fzgaJpZM4HrByj.gif>

whatnick, 2016-05-04:
I get 0xA0. I am using your MPU9250 demo code. Will check voltages.

kriswiner, 2016-05-04:
I think you are reading the wrong register. Try reading some other
registers, after you write to them

0x73 instead of 0x71 (#47)

https://github.com/notifications/beacon/AGY1qui4JfwoafegUzLKTgQ-Hb2CkfU-ks5

<https://github.com/notifications/beacon/AGY1qgp3Nxj9hjvCzWgrPPmKk9a-m3Yuks5
p-DI0gaJpZM4HrByj.gif>

whatnick, 2016-05-04:
Hi,

Which register will be good control register test case. This was my first QFN soldering attempt and there might be dry joints etc. I will continue testing. Thanks for the assistance.

kriswiner, 2016-05-04:
Make sure you set Power1 to 0x00 and then 0x01 to wake from sleep and set the clock. You could try to write to the CONFIG register. Write 0x00, then read, then write 0x03, then read. You could easily have cold solder joints which prevent power from getting to the ASIC even if you are getting an I2C ACK. Verify your code with a known good sensor. Check your solder joints for contact with the QFN side pads.

YanJenHuang, 2016-05-26:
Hi, I also read the 0x73 in my WhoAmI register.
Is different MPU9250 sensor has unique value? Can I change this value myself?

kriswiner, 2016-05-26:
The WHO_AM_I register is read only, you cannot change the MPU9250 value, but
you can change the value you compare with in the sketch to 0x73.

0x73 instead of 0x71 (#47)

<https://github.com/notifications/beacon/AGY1qhiwuN8ycuGRxQkVTFmqu872-_Udks5
qFUTSgaJpZM4HrByj.gif>

guyl44, 2016-07-02:
Hi, I've been getting a similar error - the WHO_AM_I register returns "0" when I set the AD0 to 0, and "0xFF" when I set AD0 to 1. Now, I ran the I2C scanner and the device is at 0x68 - meaning, AD0 should be 0, right?
Anyway, no one else seems to have gotten "0" as the return value so I'm kinda stuck and would appreciate your help.

kriswiner, 2016-07-02:
It sounds like there is a problem with your i2c read function or with integer types. Please tell me more about your set up, which MCU, which mpu9259 breakout and what does the i2c read code look like and maybe I can help.

guyl44, 2016-07-02:
Thanks!

OK, so I'm using Teensy 3.2, and that Chinese purple breakout board
everyone seems to have (
http://www.play-zone.ch/en/mpu-9250-accelerometer-gyro-kompass.html).

And regarding the I2C scan - it just says "device found at address 0x68".
Seems to be recognized.

One thing I changed from the default code is the interrupt pin - I'm
connecting it to pin 22 on my Teensy, so I changed the code accordingly,
but I don't think that's the culprit, ha.

Thanks again.

kriswiner, 2016-07-02:
Are you pulling NCS HIGH?

0x73 instead of 0x71 (#47)

http://www.play-zone.ch/en/mpu-9250-accelerometer-gyro-kompass.html).

etks5qRnJLgaJpZM4HrByj>

-
<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-230105413> ,
sFks5qRnqcgaJpZM4HrByj> .
<https://github.com/notifications/beacon/AGY1qiT6FfH11nPUbUCVwDtb6NI1M29nks5
qRnqcgaJpZM4HrByj.gif>

guyl44, 2016-07-02:
It's not connected to anything, so I guess... not? Oops.
My connections are VCC (3.3v), GND, SCL & SDA and then the INT pin. I haven't connected anything else because all of the codes I've seen only require these five.

kriswiner, 2016-07-02:
You have to pull NCS HIGH to put the MPU into I2C mode. Try it. And read the
data sheet.

0x73 instead of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-230106545> ,
4fks5qRoBNgaJpZM4HrByj> .
<https://github.com/notifications/beacon/AGY1qqik1NlR5TQ9VgsUrZKI2pSxDHmtks5
qRoBNgaJpZM4HrByj.gif>

guyl44, 2016-07-02:
Will do. Thanks a bunch for the help!

ValooFX, 2016-08-22:
0x73 comes from the MPU9255 in case anyone else is wondering.

kriswiner, 2016-08-22:
I have asked Invensense about this and they haven't replied. How do you know
this, is it in the MPU9255 data sheet? I'll check...

0x73 instead of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-241498251> ,
tkIAeAiks5qieTggaJpZM4HrByj> .
<https://github.com/notifications/beacon/AGY1qmAnogxFtAGEa2VYdzIffIj1T0dmks5
qieTggaJpZM4HrByj.gif>

ValooFX, 2016-08-22:
i found it in the register map document (http://stanford.edu/class/ee267/misc/MPU-9255-Register-Map.pdf) after i ran into the same issue (and i actually purchased the 9255 without realizing it)

kriswiner, 2016-08-22:
Thanks, I just read the register map. Apparently the MPU9250 I bought on
aliexpress.com were MPU9255 too!

Kris

0x73 instead of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-241542597> ,
AsK6K65ks5qigjggaJpZM4HrByj> .
<https://github.com/notifications/beacon/AGY1qr-uIRFRK7GT5zFX6DTAx6YHwZGAks5
qigjggaJpZM4HrByj.gif>

seandepagnier, 2016-08-29:
I have also 0x73 chip.  Is there a difference in the registers?  What is the difference?

kriswiner, 2016-08-29:
MPU9255 has WHO_AM_I of 0x73, there is some extra capability in the ASIC
that allows some additional gesture control but otherwise this chip is
identical to the MPU9250.

0x73 instead of 0x71 (#47)

<https://github.com/kriswiner/MPU-9250/issues/47#issuecomment-243008303> ,
iGJGwqUks5qkiL0gaJpZM4HrByj> .
<https://github.com/notifications/beacon/AGY1qqSO4wxFnhsIyFg7iagUZJuDSmdvks5
qkiL0gaJpZM4HrByj.gif>

ibudnikov, 2017-05-26:
Hello, Kris!
I have same problem as guyl44 -- WHO_AM_I returning 0.
I have GY-91 (pink one) and Arduino UNO R3. Connections:
VCC - 5V
Gnd - Gnd
SCL - SCL
SDA - SDA
Tried to connect 3.3V-NCS -- with no effect.
I2C scan finds 2 devices with addresses 0x68 and 0x76
Tried to launch with old library for MPU6050 -- all works (without magnetometer, obviously), WHO_AM_I returns 56. Do you have any ideas what's wrong?
P.S.: SCL and SDA levels are ~4V

kriswiner, 2017-05-26:
If you are able to get an ACK on I2C address 0x68 you should be able to
read the WHO_AM_I address. Sounds like your I2C library isn't working then.

Start with a simple sketch to do an I2S scan and then read the WHO_AM_I
register only. What do you get? If still 0x00, post the complete (very
short) code here and we will take a look.

ibudnikov, 2017-05-26:
(Created new issue, I thought, it would be more comfortable)
`#include <Wire.h>
//#include "MPU6050.h"
#include "MPU9250.h"
//MPU6050 accelgyro;
MPU9250 accelgyro;
void setup()
{
  Wire.begin();
  byte c = accelgyro.getDeviceID();
  Serial.begin(9600);
  Serial.println(c);
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}`
If I use MPU6050.h -- I get 2 addresses on scanner and WHO_AM_I register returns 56
If I use MPU9250.h (by Snowda) I get same 2 addresses, but WHO_AM_I responds 0.
Some parts of code in library MPU9250:
`#define MPU9250_DEFAULT_ADDRESS 0xD1
#define MPU9250_MAG_ADDRESS 0x0C`
`MPU9250::MPU9250() {
    _address = MPU9250_DEFAULT_ADDRESS;
    _mag_address = MPU9250_MAG_ADDRESS;
}
...
uint8_t MPU9250::getDeviceID(void) {
    _whoami = readRegister(MPU9250_WHO_AM_I);
    return _whoami;
}
...
uint8_t MPU9250::readRegister(const uint8_t register_addr) {
    //call sensor by address
    //call registers
    uint8_t data = 0;

    Wire.beginTransmission(_address); 
    Wire.write(register_addr); 
    Wire.endTransmission(); 

    Wire.requestFrom((int)_address, 1);

    while(Wire.available()) {
        data = Wire.read();    // receive a byte as character
    }

    return data; //return the data returned from the register
}`

kriswiner, 2017-05-26:
Where is this defined?

MPU9250_WHO_AM_I

ibudnikov, 2017-05-27:
#define MPU9250_WHO_AM_I 0x75
0x75=117 -- correct register, I think.

kriswiner, 2017-05-27:
Well, if you get an ACK on 0x68 and you are simply reading register 0x75 and getting zero return there are two possibilities:

1) your MPU9250 breakout is garbage

2) you I2C read code is flawed.

ibudnikov, 2017-05-27:
But it works with MPU6050 library:
`#define MPU6050_ADDRESS_AD0_LOW     0x68 // address pin low (GND), default for InvenSense evaluation board
#define MPU6050_ADDRESS_AD0_HIGH    0x69 // address pin high (VCC)
#define MPU6050_DEFAULT_ADDRESS     MPU6050_ADDRESS_AD0_LOW
#define MPU6050_RA_WHO_AM_I         0x75
#define MPU6050_WHO_AM_I_BIT        6
#define MPU6050_WHO_AM_I_LENGTH     6
MPU6050::MPU6050() {
    devAddr = MPU6050_DEFAULT_ADDRESS;
}
uint8_t MPU6050::getDeviceID() {
    I2Cdev::readBits(devAddr, MPU6050_RA_WHO_AM_I, MPU6050_WHO_AM_I_BIT, MPU6050_WHO_AM_I_LENGTH, buffer);
    return buffer[0];
}`
It returns 56. Raw measurments from accels&gyros also makes sense -- but I can't read mag values due to hardware difference.

kriswiner, 2017-05-27:
There is something wrong with your I2C implementation then

ibudnikov, 2017-05-29:
You were right. I've changed underlying Wire.h library to I2Cdev, made necessary changes to my code, and everything is working now.

ibudnikov, 2017-05-29:
One more question -- this time about your code:)
Lines 407-409:
`magbias[0] = +470.;  // User environmental x-axis correction in milliGauss
 magbias[1] = +120.;  // User environmental y-axis correction in milliGauss
magbias[2] = +125.; // User environmental z-axis correction in milliGauss`
Why this numbers? It's not related to magnetic inclination and declination, if I understand correctly.

kriswiner, 2017-05-29:
These are for my specific MPU9250. You have to calibrate your own.

ibudnikov, 2017-05-29:
Ok, thanks.

GerardSD, 2018-04-04:
Kriswiner i need you help!
I am programming the mpu9250 with the mpu432 in CodeComposer. At the moment I just want to be able to read the WHO-I-AM register. Do you have an example to help me?
Thanks

kriswiner, 2018-04-04:
There are plenty in the repository...

carbonadam, 2018-07-25:
IS your code going to work on with the MPU9255 or are there other problems I might run into.  Ive had some of the code running but the Yaw values are inconstant, for example seem to want to return to around 90.

kriswiner, 2018-07-25:
Should work on MPU9255. Did you calibrate the sensors? Are you feeding the
sensor data into the Madgwick filter in NED order?

amod8, 2019-06-14:
If anyone could help me resolve this error.

> inv-mpu6050-i2c 1-0068: invalid whoami 0x40 expected 0x71 (MPU9250)

 I am stuck since 3 days. Please follow this stackoverflow link for the complete question:
[https://stackoverflow.com/questions/56578146/how-to-fix-mpu9250-configuration-issue-in-device-tree](url)

kriswiner, 2019-06-14:
Your I2C is either not connected properly or it is not working properly.

amod8, 2019-06-14:
I have already configured pressure sensor and I/O expander with I2c. Also I will share my schematics
[image]

kriswiner, 2019-06-14:
OK, when you do an I2C scan, which devices show up? Is it possible the
other device also have I2C address 0c68?

amod8, 2019-06-14:
I2c scan shows this , I have already configured other devices connected to i2c, only this imu is showing above mentioned error.
  
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- UU UU UU -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- 76 --

kriswiner, 2019-06-14:
And did you try it with a different MPU9250 breakout board?

amod8, 2019-06-14:
Nope, we are working on a custom hardware board designed by some professional company. so depending on the hardware faults would be our last option which I think will be opting in a few days if no software faults are detected.

kriswiner, 2019-06-14:
Do you have I2C ports exposed? You could simply connect another board to
the main board and see if the MCU and software can properly detect the
WHO_AM_I.

Does this occur on all of your "professionally" designed boards?

Can't the "professionals" figure it out?

amod8, 2019-06-14:
Since I am completely newbie to the embedded linux, I am doubting if there is any software faults from my side, that is why I am searching for this particular error just to be sure myself. If nothing happens, I have seek their help. And yes, thankyou for your suggestions kriswiner, i will definitely follow it and update if any solution comes up.

AleDanielGuerrero, 2020-03-26:
hi kris i am new and i would like to know why it is important to check the connectivity of mpu 9250 and AK8963. Since previously I was testing a code in which I did not do this check and obtained the sensor readings without any problem.

kriswiner, 2020-03-26:
This is not important to do just good practice to check all communication
paths as part of the setup and initiation. If your program works for you
without doing this then there is no problem.

[10 more replies]