kriswiner/MPU9250 #48: Magnetometer calibration

alex934, 2016-03-07:
Hi Kris,
I'm trying to run the AHRS with the new NUCLEO cortex M7.
The fusion rate is around 8000Hz.
But I'm still not able to get any results usable.
I calibrated the magnetometer with your magCalibration+ magScale, but I don't understand when I plot my sphere, the resultat is not well centred as you can see in attachment. I'm in 8Hz magneto rate.
So It results that the pitch and roll seems correct but the yaw is never correct. During the calibration you said to wawe the device in a figure eight until done, this means that the MPU9250 should lay flat on the table? 
At the begining I thought It was my µC which was not enough fast for the fusion, but now with the cortexM7 it's couldn't be that, so what do you suggest?
It's hard to believe this algo could be run for a camera stabilization, when there is an acceleration on axe X, Y or Z the angles changes or swing, is that the same for you?

Thank you very much Kris

kriswiner, 2016-03-07:
When you calibrate the gyro and accel, the sensor should be flat and
motionless on a table. But when you calibrate the mag, it should be waved
around in a figure eight motion so the sensor can sample as much of the 3D
response surface as possible. Your mag is not calibrated. When you plot the
response surface it should be centered on the origin and reasonably
spherical.

At the begining I thought It was my �C which was not enough fast for the

<https://github.com/kriswiner/MPU-9250/issues/48> .
<https://github.com/notifications/beacon/AGY1qmc4RgqQpIjz7dW__yY5HRauZ56Wks5
prJHjgaJpZM4HrK-x.gif>

alex934, 2016-03-07:
Thank you very much for your quick answer,
after calibration, when you display the sphere you move the sensor on all axis?
My missunderstanding is that during calibration we move only on X and Y axis, but after that to display the sphere I turn my sensors in all orientation.

Alex

kriswiner, 2016-03-07:
You must expose the mag to the entire 3D response surface during
calibration, otherwise the Euler angles will be wrong, especially Yaw or
heading.

<https://github.com/kriswiner/MPU-9250/issues/48#issuecomment-193486079> .
<https://github.com/notifications/beacon/AGY1qnN7kLgMEkaRd4R2cxft3OWdk1EGks5
prKeBgaJpZM4HrK-x.gif>

alex934, 2016-03-21:
Hi Kris,
Finally I got a calibrated magnetometer:
[image]

the fusion works well, but when I translate on a axe, my angle change (due to acceleration of the acceleromter), Do you know how can I neutralize the effect of accelerometers in the angle estimation?

Thx

kriswiner, 2016-03-22:
Did you also remove the gyro and accel offset bias? It also looks like there is residual scale bias in the mag response (+/- 200 on x, y versus +/- 300 on z)

alex934, 2016-03-22:
What do you mean?
I let the calibration for the gyro and accel:
`    mpu9250.calibrateMPU9250(gyroBias, accelBias); // Calibrate gyro and accelerometers, load biases in bias registers  `

What the link between  gyro and accel offset bias and the mag response?
Thx

kriswiner, 2016-03-22:
There is a link between offset bias and Euler angle response.

<https://github.com/notifications/beacon/AGY1qmIN8baXiDVl-wvYjg5hATEfuwOlks5
pv_3SgaJpZM4HrK-x.gif>

alex934, 2016-03-22:
So what do you recommend?

Luhung300897, 2017-09-22:
Hi Alex,
can you tell me how to plot the magnetometer calibration data?