kriswiner/MPU9250 #52: Feature request

Gargo, 2016-03-14:
Could you rewrite your code for ch340-based arduino/esp8266? I mean Wemos D1 R1, DCCduino and etc. They are enough popular because of their price but at the same time they can't use avr and stm libraries

kriswiner, 2016-03-14:
It is trivial to run this code with an ESP8266; I have done it. It took me
five minutes to change the few things necessary.

<https://github.com/kriswiner/MPU-9250/issues/52> .
<https://github.com/notifications/beacon/AGY1qn1kY3Hz8xtJaA16zYHLiDzC_Zblks5
ptPrRgaJpZM4Hvx1m.gif>

Gargo, 2016-03-14:
could you commit this code. I tried to rewrite this tutorial:
http://www.lucidarme.me/?p=5057
but I still can't get something similar to correct data

kriswiner, 2016-03-15:
Not sure what you are asking. Try this code: https://github.com/kriswiner/MPU-9250.

Hassan42, 2018-07-07:
What are the changes you have made? 
it would be greatly appreciated if you could provide the edited code.