kriswiner/MPU9250 #53: MPU9250 Static vs. Dynamic Yaw Drift/Artifacts

herbertwest3, 2016-03-20:
Greetings Kris,

I just discovered the treasure trove of your write-ups, and will almost certainly begin using one of your platforms to continue my investigations. I have looked through your work and the questions you have answered, and thought you might be interested in commenting on some problems I am seeing with the 9250. 

I have an iPhone 6 mounted in landscape orientation on a servo-driven panning platform. Attached somewhat rigidly (with tape) to it is an Invensense 9250 EVB. While driving the platform (or not), I collect all raw sensor data plus (supposedly) fused quaternions from both platforms. On the iPhone, I am collecting CMMotion data, and for the EVB i am running the STM402discovery-based SDK.  The
motion is always a pure yaw (aside from mechanical errors in the mounting) over a sweep angle of
about 50 degrees.  I use various oscillation periods, as fast as about 1.5seconds and as slow as 1 minute.

Results are interesting, and very repeatable. Yaw angles recovered from *reported* quaternions are plotted below.

[image]

When the platform is stationary, the Invensense SDK yaw drifts at about 0.25 deg/minute. It maintains
this drift, when stationary, across the nearly two-hour recorded timespan.  That's
great, but the iPhone is even better.

When the platform is moving, especially quickly, things go completely to hell for the SDK. And the iPhone shows not only a yaw drift but also occasional large transients. 

Although its not visible in these plots, I have some strong evidence that when there is sufficient repetitive motion, the iPhone makes a large-scale adjustment every 15 seconds. The SDK does something similar on a 30 second timescale. This is most clearly seen in the plots for roll and pitch, which have (relatively) large changes in their patterns of cross-coupled motion that abruptly appear at these intervals. In some cases it is possible to make the 15-second iPhone interval apparent even on the yaw plot.  An early suspicion was that some software was deciding the solution was unstable and oscillating, and was resetting or re-estimating/re-calibrating. However, there are signs of this large-scale shift in bias at the right side of the plot where rotations only happen every 30 seconds.

(BTW I haven't compared "DMP ON" vs "DMP OFF" modes with the SDK -- its pretty difficult to use --
so i am not possible this is as good as it gets for the Invensense solution; but I have been futzing with the iOs stuff for quite a while)

Have you noticed any similar behavior, and do you know if Madgwick and the other fusion solutions you have investigated will fare better?

There are a couple of other issues with this data.  Toward the center of the plot, the red data from the EVB appears to be leading the blue data by multiple seconds.  This is not true at either end.  That's disturbing -- but I have not verified that I am not dropping serial port packets (and there are no time tags in the current setup).  Actually, the entire red curve had to be expanded by a factor of about 1.297 (and the data shifted a bit) to get these curves to line up.  This suggests that the output sample rate was NOT 100 Hz  but more like 77.1 Hz.  I don't know how that could have happened (will investigate) but clearly the EVB output solution is good enough that the internal reference clock can't be that far off.

I would greatly appreciate any thoughts you have on these matters.

HW

kriswiner, 2016-03-20:
Hi Herbert,

I am not sure I have much to say about this behavior. The DMP only performs
fusion on the accel and gyro and therefore will be subject to the drift you
see. I agree that when the device shows a discontinuity it is likely
adjusting itself to new conditions. This could include recalibration of bias
offsets as well as magnetic anomaly detection in the case of the cellphone.
This is often undesirable and can lead to the jogs you see. Usually there is
a way to turn this off or adjust its behavior, but probably not on a cell
phone. Likewise, these algorithms often have stillness modes which detect
when motion is temporarily stopped for recalibration and to conserve power
by placing the gyro and/or accel in a low power (low data rate) mode which
can also lead to discontinuous behavior.

My recommendation is to 1) make sure the orientation estimation you are
relying on makes use of a well-calibrated magnetometer, and 2) know how the
device s programmed to respond to various situations (stillness mode,
dynamic calibration, etc). I am not sure what your application is but there
are better solutions than the MPU9250 DMP available.

Kris

yaw1
<https://cloud.githubusercontent.com/assets/6385514/13902466/c865cd00-ee06-1
1e5-88eb-e34f9cf896ea.png> 

<https://github.com/notifications/beacon/AGY1qjX51ZmkFPx5pe51VE6sXRe1BKfUks5
pvLIlgaJpZM4H0lOV.gif>

herbertwest3, 2016-03-20:
Kris,

Thanks for the response. This tends to confirm what I suspected -- that I am seeing artifacts of Apple's and DMP's specific fusion solution. I am not specifically interested in DMP, and will immediately focus on other readily-available solutions.  Such as your own contributions, which might exactly suit my needs. 

I was hoping you had occasion to see how your own implementations fared in a similar oscillatory environment.  But I'll check it out myself in any case.

Thanks!

Herb

closed 2016-03-20 by herbertwest3

kriswiner, 2016-03-20:
Glad I was some help. If you have other questions (that maybe I can actually
answer) PM me at onehorse@earthlink.net. In the meantime, I would commend
this <https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki>  to your
attention if you have not already seen it.

 

Kris

 

 

 

 

 

 

<https://github.com/notifications/beacon/AGY1qswo-eq7X1jfm0p_LsPmmtoPZZYJks5
pvMtLgaJpZM4H0lOV.gif>