kriswiner/MPU9250 #54: Calculating correct Yaw

1994james, 2016-03-29:
Hello Kris, 

I really need help calculating the correct yaw of my 9250 sensor. It currently outputs accurate pitch and roll bu the yaw does not seem to be accurate when i pan to the left or right with the device on a flat surface. 

This is my current code --
https://github.com/1994james/9250-code/blob/master/_9250_code.ino

this is the last step i need to complete this, i need the yaw for my pan servo where the sensor is pointing in a +/-90 degree range pan and tilt. Any help is appreciated, running out of time to get it done.

Thanks in advance

kriswiner, 2016-03-30:
I don't have time to look at your code. I will assume you have properly
calibrated your sensors (if you don't know what this means, look here:
https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration ).

Maybe you can use the following MPU9250-specific info:

// Sensors x (y)-axis of the accelerometer/gyro is aligned with the y
(x)-axis of the magnetometer;
  // the magnetometer z-axis (+ down) is misaligned with z-axis (+ up) of
accelerometer and gyro!
  // We have to make some allowance for this orientation mismatch in feeding
the output to the quaternion filter.
  // We will assume that +y accel/gyro is North, then x accel/gyro is East.
So if we want te quaternions properly aligned
  // we need to feed into the madgwick function Ay, Ax, -Az, Gy, Gx, -Gz,
Mx, My, and Mz. But because gravity is by convention
  // positive down, we need to invert the accel data, so we pass -Ay, -Ax,
Az, Gy, Gx, -Gz, Mx, My, and Mz into the Madgwick
  // function to get North along the accel +y-axis, East along the accel
+x-axis, and Down along the accel -z-axis.
  // This orientation choice can be modified to allow any convenient
(non-NED) orientation convention.
  // This is ok by aircraft orientation standards!  
  // Pass gyro rate as rad/s
    MadgwickQuaternionUpdate(-ay, -ax, az, gy*PI/180.0f, gx*PI/180.0f,
-gz*PI/180.0f,  mx,  my, mz);
//  if(passThru)MahonyQuaternionUpdate(-ay, -ax, az, gy*PI/180.0f,
gx*PI/180.0f, -gz*PI/180.0f,  mx,  my, mz);

Kris

https://github.com/1994james/9250-code/blob/master/_9250_code.ino

<https://github.com/notifications/beacon/AGY1qtE1DvExYEQe2rEdR1M6oooV-bDZks5
pyPxsgaJpZM4H6lrI.gif>

1994james, 2016-03-30:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com');>]

I have now sorted the yaw. I also had to + 67 where you had -13.8 for it to
be correct for my location. All I need to do now is to limit it to +/-90
rather than +/-180 and use the yaw and pitch output to control the servos.
Will simple pwm solve this?

Thanks for the reply

kriswiner, 2016-03-30:
Most MCUs have a servo library but PWM driving a fet will also work.

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
https://github.com/notifications/beacon/AGY1qtE1DvExYEQe2rEdR1M6oooV-bDZks5

<https://github.com/notifications/beacon/AGY1qi4G9dGDefzxmx1Qnhl4l4YJG_coks5
pydizgaJpZM4H6lrI.gif>

1994james, 2016-03-30:
Ok, so my Yaw isn't fixed, its very temperamental and it change when pitch/roll changes in certain ways, mostly when pointing down. The center also seems to change position. I've got magcal implemented in my code.

Here are a couple of serial results it produces at rest, this doesn't seem right does it?.

ax = 64.39 ay = 20.51 az = 1002.38 mg
gx = -1.10 gy = 1.13 gz = -3.07 deg/s
mx = 39 my = -15 mz = 644 mG
q0 = 0.98 qx = -0.03 qy = 0.02 qz = -0.20
Yaw, Pitch, Roll: 54.53, 1.37, -3.23

ax = 65.37 ay = 20.63 az = 1000.06 mg
gx = -1.00 gy = 1.09 gz = -3.10 deg/s
mx = 49 my = -11 mz = 644 mG
q0 = 0.97 qx = -0.03 qy = 0.02 qz = -0.23
Yaw, Pitch, Roll: 50.70, 1.04, -3.97

ax = 69.76 ay = 20.26 az = 1003.60 mg
gx = -0.98 gy = 1.05 gz = -3.03 deg/s
mx = 49 my = -3 mz = 638 mG
q0 = 0.98 qx = -0.03 qy = 0.02 qz = -0.22
Yaw, Pitch, Roll: 51.79, 1.24, -3.67

kriswiner, 2016-03-30:
Check the expected magnetic fields in your area. For me I expect 200, -50, and 400 mG for x/y/z, rest. Total magnitude = 450 mG. Your values look high. Did you perform the figure eight to calibrate the mag? Plot Mx vs Mz. Is it spherical centered on the origin after calibration? Without propoer calibration the yaw will be garbage.

1994james, 2016-03-30:
I'll get the stupid question out of the way first lol, how do I find the
expected magnetic fields? I've had a look but can't find it.

I did do the figure of 8, however do you mean do the figure in the same
area over again or try and fit in as many 8's in different directions
during the calib time? I tend to swing it around like an idiot for 10
seconds lol. Thanks

kriswiner, 2016-03-30:
The purpose of the mag calibration is to sample as much of the response
surface as possible. So moving the mag in a literal figure eight all over
the place is just right. But you might need more than ten seconds.

There must be some equivalent of the US Geological service in your country
or Wikipedia that lists the expected magnetic fields in your area.

<https://github.com/notifications/beacon/AGY1ql8IWkj7hZmpJpJClLQu9PphxksLks5
pywHxgaJpZM4H6lrI.gif>

1994james, 2016-03-31:
The only magnetic data I can find for Portsmouth uk is this - 
[image]

kriswiner, 2016-03-31:
http://www.ngdc.noaa.gov/geomag-web/#igrfwmm

1994james, 2016-03-31:
I get the same values as the picture I posted from that link.

kriswiner, 2016-03-31:
Yup

1994james, 2016-03-31:
should these be the values when i am pointing the sensor north and on a flat surface?

1994james, 2016-04-01:
also is there anyway that my interrupt can be sped up, as my servo is currently moving delayed and slow becuase of this. As i would like the movement be as close to real time as possible. Thanks

kriswiner, 2016-04-01:
Faster processor

<https://github.com/notifications/beacon/AGY1qitHVh47Tacdgb0UTcyp_Exu3uamks5
pzGVKgaJpZM4H6lrI.gif>

1994james, 2016-04-01:
Where it says in serial - "rate = 100hz" is that the speed new data is
being aquired?

kriswiner, 2016-04-01:
No, that is the rate of the fusion algorithm. If it is 100 Hz you must be
using a 16 MHz Cortex M0 or an 8-bit AVR processor. On the Teensy this rate
is typically 1000 Hz or more.

https://github.com/notifications/beacon/AGY1qitHVh47Tacdgb0UTcyp_Exu3uamks5

<https://github.com/notifications/beacon/AGY1qrrY7BdVUKcgqH_u24b52Jn8S_Q0ks5
pzGiNgaJpZM4H6lrI.gif>

1994james, 2016-04-01:
Yeah im using an 8 bit AVR.

could i not just add another 0 to the line  -   deltat = ((Now -
lastUpdate)/10000000.0f); // set integration time by time elapsed since
last filter update
to make is 1000Hz?

1994james, 2016-04-01:
Actually doing that has made my servo incredibly slow

kriswiner, 2016-04-01:
Nope, I think you ought to read this before you go any further.

https://github.com/kriswiner/MPU-6050/wiki/Affordable-9-DoF-Sensor-Fusion

https://github.com/notifications/beacon/AGY1qitHVh47Tacdgb0UTcyp_Exu3uamks5
https://github.com/notifications/beacon/AGY1qrrY7BdVUKcgqH_u24b52Jn8S_Q0ks5

<https://github.com/notifications/beacon/AGY1qm5yRJiGVQO_bc1ggDC6OH1X08JYks5
pzG7YgaJpZM4H6lrI.gif>

monstah, 2017-07-13:
> [image]

> so x = 443.37 mG
y = -3.27 mG
z = 198.55 mG
and 484.9 mG for total magnitude

According to the picture, why are the x and z values swapped? 

I got this values for my city and country (Potsdam, Germany)

[image]

so I would expect the following values in mG: 

X/North:    190.667 
Y/East:       -18.5
Z/Down:    450.375 
Total:         489.437 

Is that correct? I am a little bit confused because of the picture above. 

> should these be the values when i am pointing the sensor north and on a flat surface?

I was asking myself the same question. Is that right?

kriswiner, 2017-07-13:
You want to use these values as a sanity check. You are not going to
measure these exactly. In my case, I expect about 50, 200, and 410 for a
total magnitude of ~500 so when I test a new MPU9250, for example, I make
sure that I am reading something close to this after calibration to be sure
the mag is doing what it should.

darklord07, 2017-11-16:
Hello All,
I am working on similar problem (Calculate Yaw,pitch and roll) using MPU9250. 
[MPU_data.txt](https://github.com/kriswiner/MPU9250/files/1478727/MPU_data.txt)
[MPU_Heading.txt](https://github.com/kriswiner/MPU9250/files/1478728/MPU_Heading.txt)
I have configured accelerometer data for +/-16g, Gyro for 2000dps and Magnetometer is configured to output 16 bit data.  Since I am working on resource constrained environment, I have configured IMU to sample data at 100Hz. (Can't go with 512Hz). 

The reference code of Madgwick algorithm can be found here. 
http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/ - C code. 

However, when I pass values to algorithm (with Beta 0.1) I found out that values of yaw,pitch and roll never converged. **Both raw sensor values and heading values are attached for reference .**

Can someone explain the possible cause of this problem? Also in research paper (www.x-io.co.uk/res/doc/madgwick_internal_report.pdf)  Madgwick has explained about calibration senor data from raw data ( section 4.4 ). Can someone shed a light on that too.

brrnd99, 2020-10-20:
@1994james can u send ur coding  on the completed yaw cuz i have a similar project to be done too, much appreciated.