kriswiner/MPU9250 #56: simple program for MPU 9250

GyKovacs89, 2016-04-03:
hi kriswiner,

I saw your programs about the MPU 9250 modul. They are very coplex congatulation.

I started to understand your code, and make it a bit simplier to me. I just need to get the three Euler angles on serial, nothig more.

When i use your claibration:
    magbias[0] = 470.;//+0.;  // User environmental x-axis correction in milliGauss, should be automatically calculated
    magbias[1] = 120.;//+385.;  // User environmental x-axis correction in milliGauss
    magbias[2] = 125.;//-303.;  // User environmental x-axis correction in milliGauss

there are some double axis movement when the sensor is moving on just one axis. So i calibrated it by hand as you see commented, I made the three /xy,xz,yz/ pointcloud and was rperfectly round and they ws on each other but the resut was worse.

Maybe do you have the simple program what I need, or do you have idea why do i have still this problem?

Thanks,
Gyuri

kriswiner, 2016-04-03:
Did you also calibrate the gyro and accel to correct offset?

<https://github.com/notifications/beacon/AGY1qum7xh7vkcqXuAuii0xC--nGZnSaks5
p0BANgaJpZM4H-pIB.gif>

GyKovacs89, 2016-04-03:
i did not, I will try to calirate the accel but how can i calibrate the gyro?

GyKovacs89, 2016-04-03:
I looked for the calibration with hand but I think the vales ae beautiful
accel:

[image]
gyro:

[image]

mag:
[image]

Gyuri

kriswiner, 2016-04-03:
Look here:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

<https://github.com/notifications/beacon/AGY1qhzgtf7bvJnKYAqvOFBrbhORjZBUks5
p0CW2gaJpZM4H-pIB.gif>

GyKovacs89, 2016-04-06:
Hi I have already tried, but too complex to make it simplier for me, i have commented the parts what i don't need, but after drawing the 8s in the air, the result is worse, maybe i have commented too much.

Maybe do you have an older program, where you just got as result the orientation without any other interface, just on serial?

Thanks in advance
Gyuri