kriswiner/MPU9250 #57: interrupt pin working?

matt3036, 2016-04-03:
Hi Kris,
first of all thx for the awesome code, especially the sensor fusion!
I'm using the Drotek breakout board (http://www.drotek.com/shop/en/home/421-mpu9250-gyro-accelerometer-magnetometer.html) and trying to use the interrupt pin connected to an input pin on my Arduino instead of reading the interrupt register like you did.
Unfortunately the interrupt pin is always on low level, so I can never read values from the sensor.
Can you give me a hint on how to configure the registers to get high level on the interrupt pin when the registers are filled.
I'm quite sure the board is ok since I have two of them and the same issue.

Thanks,
Matt

kriswiner, 2016-04-03:
You have to configure the interrupts in the MPU set up to be active high,
push pull. Check with an oscilloscope that it is working as expected.

(http://www.drotek.com/shop/en/home/421-mpu9250-gyro-accelerometer-magnetome
ter.html) and trying to use the interrupt pin connected to an input pin on

<https://github.com/notifications/beacon/AGY1qukqzC_oEdu7yedRxh99npvQOMACks5
p0CGNgaJpZM4H-qPc.gif>

matt3036, 2016-04-03:
I didn't dive into the registers yet, but in your code it says, that they are already active high and push pull:
from line 807 on:

// Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional chips 
  // can join the I2C bus and all can be controlled by the Arduino as master
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);    
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit 0) interrupt

Are those the "wrong" registers for what I want to do?

kriswiner, 2016-04-03:
Looks right to me, can you verify the interrupt is triggering from the
MPU9250?

<https://github.com/notifications/beacon/AGY1qiUcFLJFiaUN5U9pXfONH8C8F3dFks5
p0CRrgaJpZM4H-qPc.gif>

matt3036, 2016-04-10:
At the moment I don't have a oscilloscope, so I have to wait till next week.
Anyway, I have some other issues with the code, so  I will dive in a little deeper and try to understand what it does in more detail.
I'll let you know when I have some results ;).

matt3036, 2016-04-12:
Hi,
the interrupt pin is working now...it was just a connection issue. I wrote a very basic version of the code on my own which doesn't include very much of the initialization your code does. But at least it is generating up to date values and SPI transmission is also working. Understanding all the calibration and setup stuff will take a few more days...

Since I didn't find anything using Google about it, maybe you could answer me another question.
Is it possible to get a timestamp for each value (acc, gyro) from the mpu directly? Of course I could setup an interrupt on the arduino and set the timestamp in an Arduino-ISR, but this doesn't take into account the timespan for sensing->storing in FIFO->set interrupt pin->execute ISR.
Thank's for any ideas! :)

Regards,
Matt

ValooFX, 2016-08-21:
i also struggled to get the code working (it only printed zeros) and i suspected the interrupt.
In case anyone has a similar problem, what did the trick for me was replacing the line 
`attachInterrupt(intPin, myinthandler, RISING);  // define interrupt for INT pin output of MPU9250`
with
`attachInterrupt(digitalPinToInterrupt(intPin), myinthandler, RISING);`
(using an arduino uno)