kriswiner/MPU9250 #58: Could not connect to MPU9250: 0xFF

John-Val, 2016-04-05:
Hi Kris,
I try to use read data from a 9 dof which has written MPY 92/65 on the breakout board. On the back it has written 9250/9255/6500/6555.
So I think I should be able to use your example code. But testing it on two Arduino Uno's and with 3 different breakout boards I keep on getting:

Could not connect to MPU9250: 0xFF.

Both with or without the suggested alterations:
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.

The uno boards have separate SCL SDA pins. I used those but I also used A4 and A5 as in the sketch.
I use the 3.3 V power supply. 

Am I doing something wrong or did I just buy some useless plastic and ceramics?

I hope you can help me out.

kriswiner, 2016-04-05:
If you bought the purple boards from China then yes, they are junk.

Make sure the SDA and SCL lines both show 3V3 with a voltmeter when powered on.

It is likely you don't have the I2C address of the MPU9250 right. It can be either 0x68 or 0x69. If you run an I2C scan, what do you get? Here's how:

`// I2C scan function

void I2Cscan()
{
// scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
    
}

`

John-Val, 2016-04-05:
Dear Kris,
Thank you very much for your extremely quick response. Yes I did buy them in China, but tomorrow I certainly will try your other suggestions.
Sincerly yours,
John Val

John-Val, 2016-04-06:
Hi Kris,
It seems I got it working.
The MPU9255 rc2 is on 0x68
Who Am I should return 73. 
I also forgot to change the intPin to 2.
Many thanks again for your help.

closed 2016-04-06 by John-Val

sakshirawal14, 2017-05-21:
@kriswiner Hi
I am facing the same issue. I tried running the I2Cscan(), but the serial monitor shows no output whatsoever. How do I know if the address is correct in the code?

kriswiner, 2017-05-21:
What is the voltage on SDA and SCL?

sakshirawal14, 2017-05-22:
Hi..I'm using the Arduino Uno and the MPU 9250. I have made the following connections
VDD---3.3V
GND---GND
SCL---A5
SDA---A4

kriswiner, 2017-05-22:
Mpu9259 is 3v3 only; you need to use a logic level converter?

Do you have pullup resistors on the mpu9250 board?

sakshirawal14, 2017-05-22:
It works on 3.3V. Haven't used it before but the forums and posta I have referrred to say that only the above 4 connections are needed for it to work. I'm not using any external pull-ups either.

kriswiner, 2017-05-22:
What is the voltage on sda and scl?

You need pullups.

sakshirawal14, 2017-05-22:
Will the voltage on SDA and SCL not be 3.3 V only? I don't have much hands on experience with thw hardware. How do I use pull-ups?

kriswiner, 2017-05-22:
Connect a 10 kOhm resistor between 3V3 and SDA. Do the same for SCL.

sakshirawal14, 2017-05-22:
Okay..thank you. Will do that and try running the code. Will get back to you then.

sakshirawal14, 2017-05-22:
Okay..so the I2C scanner detects the device at 0x68.
But when I try to run the code, it says "MPU9250 connection failed"
Any idea..why?

kriswiner, 2017-05-22:
Not without more info

BobarTrump, 2017-05-22:
Hi kriswiner, and thank you very much for all what you did :) 

I'm facing a similar problem. It seems it appears when my Arduino Due reset.. When the MPU is found, it works very well and I get serial data for a while, no crash nothing, but when I close the serial monitor, and reopen it (I think it means that Arduino Due is restarted) then ~1 on 3 times, the MPU will not be found. Then I need to restart the Arduino IDE unplug my USB plug again, without warranty it works. 

Here is my wiring diagram, I use 2K resistors, do you think it's enough ? 
[image]

kriswiner, 2017-05-22:
Maybe too much, you have pullups already on the breakout no?

BobarTrump, 2017-05-22:
yes I think this 1,5 k that's what measured on SDA0 and SDA1 (20 and 21)

kriswiner, 2017-05-22:
Looks like you have GND connected to 3V3 in your diagram too.

BobarTrump, 2017-05-22:
oh ! it should be on the another one ? I didn't know about that ..

kriswiner, 2017-05-22:
Huh?

BobarTrump, 2017-05-22:
Excuse me, I didn't understand your sentence the first time. this is a mistake in my diagram but I didn't did it in real (fortunately !)... I tought you were saying me to change of GND ...

BobarTrump, 2017-05-22:
I think I solved my problem, two things : 

first, I removed theses useless pullup resistors, that improved it. 
second, it happened again sometimes so I found that interesting link that described a problem really similar : 
https://github.com/esp8266/Arduino/issues/1025 

so I implemented the solution : 

```
void I2C_recovery() {
 Serial.println("Starting I2C bus recovery");
  delay(2000);
  int SDAPIN = 70;
  int CLKPIN = 71;
  //try i2c bus recovery at 100kHz = 5uS high, 5uS low
  pinMode(SDAPIN, OUTPUT);//keeping SDA high during recovery
  digitalWrite(SDAPIN, HIGH);
  pinMode(CLKPIN, OUTPUT);
  for (int i = 0; i < 10; i++) { //9nth cycle acts as NACK
    digitalWrite(CLKPIN, HIGH);
    delayMicroseconds(5);
    digitalWrite(CLKPIN, LOW);
    delayMicroseconds(5);
  }

  //a STOP signal (SDA from low to high while CLK is high)
  digitalWrite(SDAPIN, LOW);
  delayMicroseconds(5);
  digitalWrite(CLKPIN, HIGH);
  delayMicroseconds(2);
  digitalWrite(SDAPIN, HIGH);
  delayMicroseconds(2);
  //bus status is now : FREE

  Serial.println("bus recovery done, starting scan in 2 secs");
  //return to power up mode
  pinMode(SDAPIN, INPUT);
  pinMode(CLKPIN, INPUT);
  delay(2000);
  //pins + begin advised in https://github.com/esp8266/Arduino/issues/452
  //Wire1.pins(SDAPIN, CLKPIN); //this changes default values for sda and clock as well
  Wire1.begin();
  //only pins: no signal on clk and sda
  //only begin: no signal on clk, no signal on sda

  //no further processing in case of error
  /*
  while(true)
  {
    i2c_scan(); 
  }
  */
}
```

I'm using an Arduino Due with SDA1 and SCL1 that's why I use Wire1 .  

And add it in the begining of my setup like this : 

```
 if (!I2Cscan()) {
    I2C_recovery();
    I2Cscan();
  }
```

I modified the I2C scan so it returns true if a device has been detected, false if this is not the case...

Looks like it helps :) 
```
Scanning...
No I2C devices found
Starting I2C bus recovery
bus recovery done, starting scan in 2 secs
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
```

sakshirawal14, 2017-05-23:
@kriswiner Hi..What info can I help you with?

kriswiner, 2017-05-23:
No idea what your problem is

sakshirawal14, 2017-05-23:
I2Cscanner detects the device at 0x68, but when I try to run my sketch to get raw data from the IMU, it says  "MPU9250 connection failed"
Also, I get 2 warning messages in the terminal window as:
Invalid library found in C:\Users\sakshi\Documents\Arduino\libraries\i2cdevlib-master: C:\Users\sakshi\Documents\Arduino\libraries\i2cdevlib-master
Invalid library found in C:\Users\sakshi\Documents\Arduino\libraries\i2cdevlib-master: C:\Users\sakshi\Documents\Arduino\libraries\i2cdevlib-master

kriswiner, 2017-05-23:
Why use i2cdev? Maybe need to ask them?

sakshirawal14, 2017-05-23:
I tried using your code, but it had some errors. So I just wanted to do a basic raw data read and check first.

kriswiner, 2017-05-23:
OK,

What MCU are you using?

Which code did you try?

sakshirawal14, 2017-05-23:
I'm using the Arduino Uno and trying to get the raw values using I2C

kriswiner, 2017-05-23:
Try here: http://www.lucidarme.me/?p=5057

John-Val, 2017-05-23:
To my knowledge MPU9250 is on 73. Try the MPU9150 library instead since is MPU9150 on 68.

John

kriswiner, 2017-05-23:
You can't use an MPU9150 library for the MPU9250, the register addresses
are different.

If you can get an ACK from the MPU9250 at I2C address 0x68, then what does
the WHO_AM_I register show? Should be 0x71 for the MPU9250 or 0x73 for the
MPU9255.

Patrickyp, 2018-05-23:
Kris can you help me I'm trying to write high to the AD0/SDO pin of my sparkfun 9250 but the address is staying at 0x68 even though I made the pin high.    

[https://i.imgur.com/jaw8rxF.jpg](url)

void setup()
{
    Wire.begin();
    //  TWBR = 12;  // 400 kbit/sec I2C speed
    Serial.begin(38400);
    
    // Set up the interrupt pin, its set as active high, push-pull
    pinMode(intPin, INPUT);
    digitalWrite(intPin, LOW);
    pinMode(myLed, OUTPUT);
    digitalWrite(myLed, HIGH);
    
    pinMode(12, OUTPUT);
    digitalWrite(12, HIGH);
    
    delay(2000);
    I2Cscan();

and returns 

Scanning...
I2C device found at address 0x68  !
done

MPU9250
9-DOF 16-bit
motion sensor
60 ug LSB
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

kriswiner, 2018-05-23:
Is pin 12 connected to ADO or not? Is pin 12 showing 3.3 V?

MitalPattani, 2018-05-29:
@Patrickyp You need to soldier your sj2 centre pad to the right one for and then connect AD0 to 3v3 or gnd for 0×69 or 0×68 respectively.

Patrickyp, 2018-05-29:
@MitalPattani are you talking about the white little rectangle next to the square that looks like it has solder on it?

Patrickyp, 2018-05-30:
https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/#product-reviews

@kriswiner would you recommend buying this as alternative to the sparkfun mpu9250? I'm thinking of trying your version to see if it works well for me.

kriswiner, 2018-05-30:
> product-reviews

This has both the MPU9250 (in a better design than that offered by
Sparkfun) and the EM7180 motion co-processor. With the latter, you can read
quaternions or Euler angles (yaw, pitch, roll) directly from the EM7180 via
I2C register reads . It typically offers 2 degree rms heading accuracy with
proper calibration as well as warm start capability; starting from power on
in a previously calibrated state. It is a relatively low-power solution
providing quaternion updates at the rate of the gyro up to 400 Hz
guaranteed.

Compared to the Sparkfun offering, it is an astounding bargain. So yes, I
would recommend it.

MitalPattani, 2018-05-30:
Yes the silver coloured pads below which sj2 is write. When you desoldier it you will see 3 rectangular pads, you need to connect one on the centre and right.