kriswiner/MPU9250 #59: 1Khz sample Accel and Gyro

AM498, 2016-04-25:
Hi kris,
I have some quick questions
I'm wondering why in your Accel+Gyro configuration you choose:

 // Disable FSYNC and set accelerometer and gyro bandwidth to 44 and 42 Hz, respectively; 
 // DLPF_CFG = bits 2:0 = 010; this sets the sample rate at 1 kHz for both
 // Maximum delay is 4.9 ms which is just over a 200 Hz maximum rate
  writeByte(MPU9250_ADDRESS, CONFIG, 0x2);

200Hz and not 1Khz?
I don't really understand, it means that inside the MPU it sample the datas at 1000Hz but you can read these datas at maximum 200Hz?

For another purpose, if I want to sample my accel and my gyro at 1Khz I just have to change 
 writeByte(MPU9250_ADDRESS, CONFIG, 0x0); right?

Thank you very much

kriswiner, 2016-04-25:
Setting the sample rate for the MPU9250 is complicated and must be done in
two registers. One to set the sample rate and one to set the decimation, in
this case by a factor of 5. This means the accel and gyro are sampled at
1000/5 Hz in my code. I would recommend you consult the data sheet.

Kris

<https://github.com/notifications/beacon/AGY1qpNtqwETjYOaKVy-WzuTj1XKfiboks5
p7TDfgaJpZM4IPYBG.gif>

AM498, 2016-04-26:
Hum well I'm kind of lost with the datasheet:
For this:
// Disable FSYNC and set accelerometer and gyro bandwidth to 44 and 42 Hz, respectively; 
// DLPF_CFG = bits 2:0 = 010; this sets the sample rate at 1 kHz for both
// Maximum delay is 4.9 ms which is just over a 200 Hz maximum rate
writeByte(MPU9250_ADDRESS, CONFIG, 0x2);
we have
[image]

And SMPLRT_DIV just divide SAMPLE_RATE= Internal_Sample_Rate / (1 + SMPLRT_DIV)

So If I want 1Khz accel and Gyro I can just 

writeByte(MPU9250_ADDRESS, CONFIG, 0x0);
means : 
 *          |   ACCELEROMETER    |           GYROSCOPE
 * DLPF_CFG | Bandwidth | Delay  | Bandwidth | Delay  | Sample Rate
 * ---------+-----------+--------+-----------+--------+-------------
 * 0        | 260Hz     | 0ms    | 256Hz     | 0.98ms | 8kHz

accel+gyro sample at 8Khz
and after that I use SMPLRT_DIV =7 to get SAMPLE_RATE= 8Khz/ (1 + 7)

I am right? thank you very much for your precious help

kriswiner, 2016-04-26:
Yes, for the gyro. I think the max rate on the accel is 1 kHz.

tiancaipipi110, 2018-04-07:
Hi Kriswiner, 
do you know how to change the accelerometer sampling rate of MPU6515 to 4000hz? I checked the datasheet, it should be able to do so. But I have little knowledge on coding. Appreciate your help if you would. Thank you!

kriswiner, 2018-04-07:
Configure in register 29, see Table on page 15 of the register map. So set
bit 3 to 1, 0x08...

tiancaipipi110, 2018-04-08:
Thanks Kriswiner!

iotsanjeev, 2020-09-22:
Hello Kriswiner 
Could you please guide us to set the sample rate of  MPU 9250 at  any custom rate like 200Hz.  Please help me it would be so nice of you and how we get error free roll pitch and yaw readings

kriswiner, 2020-09-22:
Sample rate is set in the initialization function. Select 1 kHz rate and
then use SampleRateDivide to divide this by 4 + 1 to get 200 Hz. Please
consult the data sheet and register map.