kriswiner/MPU9250 #60: Could not connect getting 0x1 instead of 0x71

raoabhijith, 2016-04-27:
Hi Kris,

I am currently using your STM32F401 project for my STM32 Nucleo board, and I am getting the below message:

CPU SystemCoreClock is 84000000 Hz
I AM 0x1
I SHOULD BE 0x71
Could not connect to MPU9250:
0x1

When I tried to fix this by changing the condition as 0X1, I got to read the data from the sensor but everything seems to be zeros and nothing is really changing. Please suggest me how to get this working for STM32 Nucleo F401RE. I am using the Mbed compiler.

kriswiner, 2016-04-27:
You don't have either power to the sensor or have the wrong I2C connections.

raoabhijith, 2016-04-28:
I have connected SCL and SDA to A4 and A5 pins asnper your comments in the code. And VCC for the board is connected to 3.3V on the board. Do you think something needs to be changed?

kriswiner, 2016-04-28:
Do you have pullup resistors on SDA and SCL? What is the voltage showing on
SDA and SCL?

0x71 (#60)

<https://github.com/notifications/beacon/AGY1qjaelPsJ58Htx7hgyv21evBKyC9Sks5
p8AvggaJpZM4IRI7-.gif>

raoabhijith, 2016-04-29:
Without the pull up resistors only I am getting close to 3.3V on SDA and SCL pins Kris.. Not sure what the problem could be :(

raoabhijith, 2016-04-29:
My final connections with my STM32 nucleo are like this:
VCC - 3.3V
GND - GND
SDA - A4
SCL - A5
ADO - GND
and the I2C address has been changed to x68 since ADO has been given LOW.. I am using this breakout board: 

[image]

kriswiner, 2016-04-29:
I don't know why it is not working. Make sure you are calling A4 and A5 in
your code, and you could try switching A4 and A5, but the board you are
using looks OK. You could try another MPU9250 board.

Or try another MCU like a Teensy.

0x71 (#60)

image
<https://cloud.githubusercontent.com/assets/18704787/14920364/b00d9580-0e4a-
11e6-964e-0da310759faa.png> 

-
<https://github.com/notifications/beacon/AGY1qjd4Cs0lp5-krFARMfiOWekg_huZks5
p8h-YgaJpZM4IRI7-.gif>

whatnick, 2016-05-03:
On the Nucleo the i2c bus is actually not connected to A4-A5. Instead it is on the other side the last 2 pins. Try hooking up the sensor there.

whatnick, 2016-05-03:
See pinout here - https://developer.mbed.org/platforms/ST-Nucleo-F401RE/#board-pinout. A4-A5 is not linked to i2c.

raoabhijith, 2016-05-03:
Thank you so much whatnick. It worked now I am able to read the sensor data successfully. Thank you Kris for the suggestions.

closed 2016-05-03 by raoabhijith