kriswiner/MPU9250 #61: MPU 9250 in I2C does not return any data

abstruse123, 2016-05-03:
Dear Mr. Kriswiner

Thanks for your project on MPU 9250, I could draw some support form it.

I interfaced MPU 9250 on the GY 9250 board to a PIC MCU through I2C protocol.

I wanted to read the WHO_AM_I register of MPU 9250. 
MPU 9250 does not return any data in response.

kriswiner, 2016-05-03:
Do you have pullup resistors on the MPU9250 I2C lines? What is the voltage reading from SDA/SCL pins? Are you using the correct pins on the PIC for I2C?

abstruse123, 2016-05-05:
Thanks for ur response.
Yes pull up resistors of 10 kohm each are present on SCL and SDA pins.
Getting around 2V on both the pins.

kriswiner, 2016-05-05:
That sounds about right. There must be some other problem. Try a different
breakout or a different MCU to rule out the software.

(#61)

<https://github.com/notifications/beacon/AGY1qtt9QZ7OYg6cYDwMacZ2lj1nEepCks5
p-XpHgaJpZM4IWGGT.gif>

abstruse123, 2016-05-05:
I am using 3.3V supply from a PIC kit to power the MPU 9250 board.
Changed the resistors to 5 kohm each on SCL and SDA lines and now the WHO-AM-I register content is correctly read as 'q' i.e. ASCII value of 0x71.

kriswiner, 2016-05-05:
There are 4.7 kOhm resistors already on the mini board; did you close the
solder jumper on the front to enable them? Or maybe if not this is why you
were having trouble?

(#61)

<https://github.com/notifications/beacon/AGY1qjX0qvsblqTYzi1ZfMZCgIHzVtFPks5
p-YSagaJpZM4IWGGT.gif>

abstruse123, 2016-05-05:
I am using this board  http://www.hotmcu.com/9dof-imu-module-with-mpu9250-p-172.html
It has 10 kohm resistors and there are no jumpers provided on the board.
It works most of the time with this 5 kohm tweak but not very consistently.

kriswiner, 2016-05-05:
You get what you pay for it seems.