kriswiner/MPU9250 #63: The accelerometer offset register

YanJenHuang, 2016-05-26:
Hi @kriswiner ,
Is the accelerometer and gyro offset register's value subtracted by the accel/gyro output?
I have reading the accel offset and get the value (not divided by the scale)below:
-4627.00	-20818.00	0.00
I am curious about what scale it is. Is it ±16g as denoted in Registerr-Map document?
[image]
Thanks~

kriswiner, 2016-05-26:
Yes, they should be subtracted. The accel offset is in whatever units you
choose. For the hardware offset register IIRC it is either 4G or 8 G.

Registerr-Map accel offset
<https://cloud.githubusercontent.com/assets/13307758/15568110/a81f7772-235c-
11e6-9b52-5c8f19a9b74b.png> 

<https://github.com/notifications/beacon/AGY1qg-kJCYmiaP72cn9RiE4ulUi13pYks5
qFVcHgaJpZM4InTkE.gif>

YanJenHuang, 2016-05-26:
Hi @kriswiner ,

If i don't want to calibrate the sensor every time, I have to store the accel bias in my firmware as a constant value right?
After reboot the sensor, I just have to use the accel offset register's value to subtract the bias value and then write the new value into the accel offset register right?

for example:
accel offset read from register:
X1 = -4690.00;	Y1 = 25.00; Z1 = 10752.00;
bias(take the mean value for taking 1000 samples):
X2 = -1032.00;	Y2 = 340.00; Z2= 16940.00;

After subtracted:
X_new = X1-X2; Y_new=Y1-Y2; Z_new=Z1-Z2;

Please tell me if i have misunderstanding!

Thanks~