kriswiner/MPU9250 #64: MPU9250BasicAHRS.ino will not compile in Arduino as Pro Mini?

geosmall, 2016-06-04:
Readme appears to indicate that MPU9250BasicAHRS.ino will compile in Arduino as a Pro Mini target.

Copying MPU9250BasicAHRS.ino to a folder named MPU9250BasicAHRS and compiling for Pro Mini threw errors on missing Adafruit libraries.  Missing libraries were located in GitHub and copied into Libraries folder.

Next, compiling MPU9250BasicAHRS.ino gives the following error:

`
MPU9250BasicAHRS.ino: In function 'void loop()':
MPU9250BasicAHRS:433: error: 'MahonyQuaternionUpdate' was not declared in this scope
'MahonyQuaternionUpdate' was not declared in this scope
`

Is there another library missing that contains the MahonyQuaternionUpdate function?

Thanks,
George

kriswiner, 2016-06-04:
Yes, Look in the same github repository there is a quaternionFilters.ino
file required.

Arduino as Pro Mini? (#64)

pmks5qIY_ggaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qvfQeiAhYfjMGYlWErqZ23bY6VWdks5
qIY_ggaJpZM4IuJMI.gif>

geosmall, 2016-06-04:
Ah, yes thank you Kris.  Copying over quaternionFilters.ino to MPU9250BasicAHRS as well allows project to compile, although I now get an error not enough dynamic memory.  This is on Arduino 1.6.5, it may be a versioning issue as I am only 20 bytes over.  Compiling to ATMega1284P or 256 target gives no errors. 

`Sketch uses 26,984 bytes (87%) of program storage space. Maximum is 30,720 bytes.
Global variables use 2,068 bytes (100%) of dynamic memory, leaving -20 bytes for local variables. Maximum is 2,048 bytes.`

closed 2016-06-04 by geosmall

grassm, 2016-07-21:
geosmall  How did you get around size limitation?  I'm trying this on a Nano /promini.  Compiling for the mega2560 uses 27,362 bytes of program memory and 2,068 bytes of dynamic memory.  The nano has 30,720 and 2,037 bytes respectively.   How did you get the dynamic memory down below the 2,037? 
I'm using 1.6.5.

kriswiner, 2016-07-21:
I probably had to comment out most of the serial debugging and get rid of
the self testing and calibration.

I gave up running on the 3V3 8 MHz pro Mini about the day after I got my
first Teensy. My recommendation is to do the same. It will be the best
$19.80 you ever spend.

If you want even more performance, get a Dragonfly for $10 more. See:

https://www.tindie.com/products/onehorse/dragonfly-stm32l4-breakout-board/

This has 512k flash and either 128k bytes of RAM IIRC.

Both are programmable via an Arduino IDE.

Kris

Arduino as Pro Mini target (#64)

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234384415> ,
3MztFB7ks5qX9-agaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qtPUkaCl_f6XLec_UbZWCxYSEn4Tks5
qX9-agaJpZM4IuJMI.gif>

grassm, 2016-07-21:
Hi Kris, thanks for the quick response.   I replaced all of the Serial.print("....")  with Serial.print(F("......."))  and did the same with the Serial.println(".....").   Now all of the literals are left in program memory.   Dynamic usage is now 1,370 bytes 66%.
I have a tendency to let a lot of magic smoke out so I try to use the Nano and promini as much as I can.
The code is now working on the Nano and I have a couple Nokia displays on order.    I'll order a Teensy also.   With the sensor just sitting on the desk top the Yaw jumps around by several degrees.  Now that it's working, I'll have to play around with it.  I'm new to the IMU/PMUs.  My need is for a cart to track straight within +/- 2" over 50'  which is +/- .19 degrees.    From what I've seen so far The 9250 won't do it for me.    There will be vibration and varying magnetic fields from the motors. Guess I'll just have to play with it a little.

grassm, 2016-07-21:
...and thanks for sharing all of your work.  I can tell you've put a lot of time into this.

kriswiner, 2016-07-21:
Glad you got it working.

Best you'll be able to do is +/- 2 degrees rms absolute heading accuracy.

Sounds like your sensors are not calibrated yet, this will cause yaw
instability. Also, if you're doing fusion on a 8 MHz or 16 MHz AVR, your
fusion rates will be sloooooow. Really not good enough. You'll need a Teensy
and more.

Take a look here for backgraound info on sensor fusion:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki

Kris

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234406565> ,
291dwNqks5qX_ZDgaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qjBccoyLnLCh5slG1d0mX5Af3mTOks5
qX_ZDgaJpZM4IuJMI.gif>

grassm, 2016-07-21:
The 9250 has been running for about an hour.  The yaw varies just under  +/- 2 degrees; that matches what you said.   It's not calibrated.  But +/-2 degrees is a factor of 10 off.  So this won't work for me. Guess I'll have to look into setting up a GPS base station.  From what I understand about that is +/- 2".  May as well learn what I can on the 9250 while I'm at it. I'll check out the link and order a Teensy.

kriswiner, 2016-07-21:
Without calibration you're just getting garbage. I mean with good
calibration the MPU9250 can deliver +/- 4 degree absolute heading accuracy
with a jitter of a few tenths of a degree. Even better when using the
EM7180, there you can achieve +/- 2 degree accuracy and 0.1 degree stability
in yaw for given orientation of the device.

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234416436> ,
Kn7RC8mks5qYAOOgaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qii8eDqZSAV2EZaNg3YSvQnyTb6mks5
qYAOOgaJpZM4IuJMI.gif>

grassm, 2016-07-22:
Just finished reading about the USFS. I may not have been clear, but I don't need to have my vehicle travel a specific bearing WRT true north.  I want it to be able to travel 50' in the relative bearing that it's pointed in.  If the yaw is reading 20 degrees will the USFS track that reading within +/- .2 degrees?  To ask it in another way.... over a 50' distance that would be covered in about 15-20 seconds, would the USFS be able to maintain a +/- .2 accuracy?

kriswiner, 2016-07-22:
If you use the MPU9250 and the sensors are properly calibrated, yes.

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234422247> ,
dVYoQQsks5qYAx8gaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qqaBcr1YoGc1uSkEBHc1bIM82zNHks5
qYAx8gaJpZM4IuJMI.gif>

kriswiner, 2016-07-22:
With a fast MCU, open-source Madgwick sensor fusion, the MPU9250 properly
calibrated you can get +/- 4 degree absolute heading accuracy wit a few
tenths degree jitter.

With the USFS you can get +/- 2 degree absolute heading accuracy with one or
two tenths degree jitter.

You cannot approach this performance with any GPS-based solution I am aware
of.

Do I understand you correctly? Using the MPU9250 (and a Teensy, so I have
the fast update) if properly calibrated, I can get +/- 0.2 degrees? Or is
that with a calibrated USFS from Tindie?

-
<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234423869> ,
cd1rNpyks5qYA9agaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qjrgs8J-KkmVI_N3SmIiawZxxe-Wks5
qYA9agaJpZM4IuJMI.gif>

grassm, 2016-07-22:
Then it's worth it to me to try the USFS and the Teensy.  The Teensy has more of a base so I'll start there and migrate to the Dragonfly.  I have several of the generic Leaf Labs Maple minis STM32F103.  Roger Clark (Australia)  has done a lot with them.   They work well if you don't need to use any of the libraries that were written for the Arduino boards. 

What I've read about the GPS base station setup: Similar to WAAS.  You have a fixed GPS unit as your reference.  The variances at the reference GPS are transmitted to the roving GPS unit.  The roving unit then uses the variance data to adjust it's own location based on the fixed GPS unit.  They have these systems for sale but they cost several grand.   The advertised accuracy is +/- 3 or 4 cm.  That would work great for me.  Just out of my price range.

Just finished reading your BIO/interview.  WOW! Me... retired from 25years as an engineer.  I had done some MCU assembly level programming for the USN on PICs, 80C88s, and Atmel devices.   After I retired I stumbled upon the Arduino!   Fell in love with the simplicity and minor learning curve.  The "adult" tool chains that you talked about have the same effect on me.  Way to much hassle to come up to speed.  

Thanks for all of your work and help that you so generously share with every one! 
Sincerely, Mark

grassm, 2016-07-22:
Ordered the USFS and a Dragonfly!

kriswiner, 2016-07-22:
Appreciate the kind words Mark, and glad to help.

A note on the Dragonfly vs. Teensy. If you want to run fast fusion on the
MCU then there is a big difference between the two. While on an 8 MHz Pro
Mini you might get 200 Hz for the fusion rate, meaning you might only get
one iteration of the Madgwick algorithm for each new gyro data point at 200
Hz ODR, on the Teensy when using interrupts you could get 3 kHz fusion rate,
which means you could interate ten or fifteen times on each new sensor data
set and reach closer to full convergence in the orientation estimation. 

With the Dragonfly, you have asynchronous IO and a single-precision floating
point unit both of which are missing on the Teensy, and with these you can
run the fusion at 60 kHz, yes sixty, twenty times as fast as possible on the
Teensy. This means you can run at much higher sensor sample rates, say 1 kHz
for accel and 4 kHz for gyro and still fully iterate to an asymptotic
orientation solution. This means in practice that the jitter on the heading
even with open-source fusion can be quite low. And the topper is the
Dragonfly uses one-seventh the power of the Teensy for an equivalent task.

You really get your money's worth for the extra $10 you pay for the
Dragonfly!

Kris

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234429775> ,
JdgdWDmks5qYBnEgaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qtxqyMF9X9nnPfaW11pkWAfV69h0ks5
qYBnEgaJpZM4IuJMI.gif>

kriswiner, 2016-07-22:
Good choices! Let me know how I can help you get up to speed on these.

<https://github.com/kriswiner/MPU-9250/issues/64#issuecomment-234430744> ,
PILRaZwks5qYBuRgaJpZM4IuJMI> .
<https://github.com/notifications/beacon/AGY1qurRtncxleD3s7LlMDnXQCV2K9xVks5
qYBuRgaJpZM4IuJMI.gif>