kriswiner/MPU9250 #65: Gyro Offset (Bias)

barakman, 2016-06-19:
I have empirically found that I need to use:

data[0] = (+gyro_bias[0]/4  >> 8) & 0xFF;
data[1] = (+gyro_bias[0]/4)       & 0xFF;
data[2] = (-gyro_bias[1]/4  >> 8) & 0xFF;
data[3] = (-gyro_bias[1]/4)       & 0xFF;
data[4] = (+gyro_bias[2]/4  >> 8) & 0xFF;
data[5] = (+gyro_bias[2]/4) & 0xFF;

This is different from the original code, in those '+' signs (originally '-' signs).

kriswiner, 2016-06-19:
It doesn't make sense to treat one axis differently from the other two. I suspect you have an integer sign conversion problem instead. It doesn't matter for this particular MPU9250 if you use this fix but this coding is guaranteed to give the wrong result with any other.

barakman, 2016-06-19:
Any chance that this my observation is the result of the physical position of the MPU controller?

kriswiner, 2016-06-19:
No.