kriswiner/MPU9250 #66: Compatibility?

chrisspen, 2016-07-09:
I bought [this](https://www.amazon.com/Diymall%C2%AE-Mpu-9250-Nine-axis-Attitude-Acceleration/dp/B00OPNUO9U) MPU9250 board, which I'm trying to calibrate and read data from. I've found some sample code that successfully reads raw values, but I can't find anything to calibrate it or access the DMP functionality.

From your README, you specifically target versions of this board implemented as Teensy add-on shields. Will your code work for this board? If so, which of your ino files are best suited to this board?

kriswiner, 2016-07-09:
> I bought this
> <https://www.amazon.com/Diymall%C2%AE-Mpu-9250-Nine-axis-Attitude-Accelerati
> on/dp/B00OPNUO9U>  MPU9250 board, which I'm trying to calibrate and read

Any of my sketches will work with this board, but you will have to adapt
them to your specific MCU and connections.

7Tks5qUAs5gaJpZM4JIs3h> .
<https://github.com/notifications/beacon/AGY1qrqyMB4yvb5nb11-PNAhLJ1KSjVjks5
qUAs5gaJpZM4JIs3h.gif>

chrisspen, 2016-07-10:
Your code doesn't compile. Do you have a installation guide or dependency list? It seems to rely on a library called "mbed". I'm assuming that corresponds to https://www.mbed.com, but after several minutes of searching, I can find no way to download any code. From the documentation, it brands itself an "OS", leading me to believe it's will add an enormous amount of complexity to my simple Arduino project.

kriswiner, 2016-07-10:
Not sure why yu chose a sample designed to run on the STM32F4 but try this
one instead; with small modification you should be able to get it to work.

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino

<https://github.com/kriswiner/MPU-9250/issues/66#issuecomment-231567009> ,
PZks5qUFirgaJpZM4JIs3h> .
<https://github.com/notifications/beacon/AGY1qsUXmtuBnMq4awUx3nSX5u2dtPs9ks5
qUFirgaJpZM4JIs3h.gif>

chrisspen, 2016-07-10:
That is the one I tried. It complained of an undefined MadgwickQuaternionUpdate(), which I saw is defined several times in your code, so I first tried copying over the file quaternionFilters.ino, but that complained of more undefined variables. So I tried copying over MPU9250.h...but again it complained of the missing mbed.h. I eventually found that copying and pasting the code from quaternionFilters.ino into MPU9250BasicAHRS.ino allows it to compile cleanly.

It now runs, but the self-calibration doesn't work. I left the sensor lie completely flat while it calibrated, but the resulting yaw values are wrong. If I rotate the sensor 90 degrees, it only registers a change of 5 degrees. 

How do you interpret the magnetometer values? When the compass is pointing exactly north, what should the values read?

kriswiner, 2016-07-10:
This one has a proper calibration suite.

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

<https://github.com/kriswiner/MPU-9250/issues/66#issuecomment-231568098> ,
iIks5qUGHFgaJpZM4JIs3h> .
<https://github.com/notifications/beacon/AGY1qmzeyj0CLh4JdeYp22LM4JJnXvnEks5
qUGHFgaJpZM4JIs3h.gif>

chrisspen, 2016-07-10:
That one does run through a calibration procedure...but then gives me zeros for all readings except temperature...

kriswiner, 2016-07-10:
You need to use the interrupt.

<https://github.com/kriswiner/MPU-9250/issues/66#issuecomment-231569300> ,
AOks5qUG01gaJpZM4JIs3h> .
<https://github.com/notifications/beacon/AGY1qvNpAHPDnNOH44DpV6i1gE_sjusHks5
qUG01gaJpZM4JIs3h.gif>