kriswiner/MPU9250 #67: Porting MCU-9250 code to Java on Raspberry Pi

gjwo, 2016-07-13:
My son and I are trying to port your code https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS_t3.ino into Java to run on a Raspberry Pi 3 to act as a navigation system for a model tank. Thanks for the effort you must have put in to produce this, it's a lot of work, I doubt we would have got far without it as the chip documentation is somewhat lacking.  Neither of us have done navigation before though both of us have a computing background, so some I apologise in advance if some of the questions are naive.
The code is communicating with the devices MCU-9250 & HMC5883L,  but we are getting some odd results when calibrating.
Q1 - Should the model be at rest when calibrating or should we be moving it through a figure of eight (or both at different times)?
Q2  Is the faster processor speed likely to cause any issues when reading or writing to the chip, should we add addition delays?
Q3 Having calculated yaw, pitch and roll, do we just integrate to get lower order navigation information or is there more to it than that?

gjwo, 2016-07-13:
Q1 - answer both, move when prompted!

gjwo, 2016-07-13:
See repository https://github.com/MAWoodMain/RPITank/blob/master/src/main/java/sensors/MPU9250/MPU9250.java

Q4 - when running the accelerometer (at rest) the values seem to be jumping around see sample below

```text
Accelerometer accuracy:(% away from factory values)
x: 25.29503969853891%
y: 18.998132257660597%
z: 31.733021614288283%
Gyroscope accuracy:(% away from factory values)
x: 1.750724898603777%
y: 102.08698818901449%
z: 1.7267709867019099%

Accelerometer -1936.0, -104.0, -16823.0
Accelerometer -119.0, -107.0, -16808.0
Accelerometer -1922.0, -103.0, -88.0
Accelerometer -125.0, -118.0, -54.0
Accelerometer -1935.0, -85.0, -34.0
Accelerometer -1939.0, -110.0, -120.0
Accelerometer -1967.0, -107.0, -106.0
Accelerometer -1929.0, -113.0, -104.0
Accelerometer -1967.0, -92.0, -44.0
Accelerometer -108.0, 6778.0, -35.0
Accelerometer -1922.0, -79.0, -116.0
Accelerometer -1943.0, -103.0, -16788.0
Accelerometer -1944.0, -122.0, -20.0
Accelerometer -1977.0, -124.0, -127.0
Accelerometer -1961.0, -114.0, -108.0
Accelerometer -1946.0, -101.0, -122.0
Accelerometer -1950.0, -120.0, -92.0
Accelerometer -1944.0, -117.0, -16772.0
Accelerometer -1980.0, 6780.0, -16796.0
Accelerometer -127.0, -100.0, -104.0
Accelerometer -122.0, -109.0, -16769.0
Accelerometer -127.0, -109.0, -16778.0
Accelerometer -127.0, -126.0, -93.0
```
The x and z values seem particularly wild, do you have any ideas as to why this might happen?

kriswiner, 2016-07-13:
Hi Graham,

If you already have an MPU9250 not sure why you need an HMC5883 too.

Anyway, when calibrating the accel/mag the device needs to be flat on a
table at rest. When calibrating the mag the device needs to move around to
sample as much of the response surface as possible, figure eight is the
usual advice.

The processor speed won't matter but I have heard the RPi has some trouble
with I2C, not sure what to recommend here. The sensors can handle up to 400
kHz I2C speed which should be plenty to get good performance.

The Madgwick (or any other) sensor fusion algorithm is intended to produce
absolute orientation wrt gravity and True North. This is represented
typically by Euler angles. Integrating the Euler angles doesn't really make
any sense. If you want to estimate velocity you would have to 1) remove the
gravity component from the acceleration, 2) integrated the resulting
acceleration to get velocity. From there you could integrate again to get
position but the white noise in the data will rather quickly render large
errors in position estimation. But it is worth a try just to see it for
yourself.

What kind of MPU9250 breakout board are you using?

The jitter in the acceleration with the device at rest should be no more
than +/- 20 millig. I notice you listed raw counts. These need to be
converted to scaled acceleration values (I like milli g so that at rest the
z-axis should read +1000 (device up, gravity down)).

How does the RPi handle two's complement? You might have to modify the
integer casting differently than we do in Arduino.

Kris

(#67)

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS_t3.ino

emks5qVL3jgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qpCP_IDoI1kAsHqVCsjAVOQlCfzjks5
qVL3jgaJpZM4JLRwF.gif>

gjwo, 2016-07-13:
Kris,
Thanks for the response.

The  HMC5883L is a red herring, that was an earlier sensor we were using, and the class for it is still in the code which threw me, my son is interfacing to the device, I picked it up from his code. 

The board is  MPU9250 BMP280 10DOF Acceleration Gyroscope Compass Nine Shaft Sensor Module GY-91 found here https://www.amazon.co.uk/dp/B017VT9TF4/ref=sr_ph_1?ie=UTF8&qid=1468424771&sr=sr-1&keywords=9%20dog - as a result of this conversation we have just realised it has a pressure sensor on it!

The penny has now dropped about acceleration versus orientation, thanks for that.

We did have an issue with twos complement in a different area of code, it wasn't an RPI issue but in Java we had used a 32 bit **integer**, instead of a 16 bit **short**.  The register settings were done using hex byte values, so were not affected by this, and the data was read byte wise manually to check the values were as displayed.

I take your point about scaling but we have rowed back from scaling and offsets until we can see something sensible coming out of the raw data.

Would you normally just use sensor like this for just orientation and heading? For our application (a toy tank indoors) we couldn't use GPS for location so were looking to work out the speed and displacement as part of an inertial navigation system.

Given that this is proving hard, if we are still stuck in a week's time plan B would be to use the BNO055 https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/overview as it appears to offer a simpler interface with built in sensor fusion, is this something you have an opinion on? 

Graham

kriswiner, 2016-07-13:
Yes, I have done comparative testing on the BNO055 and it doesn't perform as
well as an EM7180+MP9250 solution. I sell a lot of BNO055 breakout boards
(https://www.tindie.com/products/onehorse/bno-055-9-axis-motion-sensor-with-
hardware-fusion/), probably due to Bosch's reputation and the lure of easy
orientation but in fact the underlying sensors (especially the accel and
mag) leave something to be desired.

I would recommend this instead:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

but I would recommend you read some of the comparison test accounts I wrote
on github before you decide. You can find it here:

https://github.com/kriswiner/MPU-6050/wiki/Hardware-Sensor-Fusion-Solutions

Kris

Pi (#67)

https://www.amazon.co.uk/dp/B017VT9TF4/ref=sr_ph_1?ie=UTF8&qid=1468424771&sr
=sr-1&keywords=9%20dog - as a result of this conversation we have just

https://learn.adafruit.com/adafruit-bno055-absolute-orientation-sensor/overv
iew as it appears to offer a simpler interface with built in sensor fusion,

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-232407933> ,
5Uks5qVRBNgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qjxLBAP6l0dizWCzoL6M31x2u7Ppks5
qVRBNgaJpZM4JLRwF.gif>

gjwo, 2016-07-13:
Kris,
Thanks for the recommendations above, it looks like @MAWoodMain and I need to persevere with the MP9250 for a little longer.
Did you have a view on what might cause the wild fluctuations visible in Q4 above?

Graham

kriswiner, 2016-07-13:
Crappy Chinese breakout board?

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-232428788> ,
kKks5qVSE3gaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qnXe2FnS1PaNIzz2q2S-rSPVcR4lks5
qVSE3gaJpZM4JLRwF.gif>

gjwo, 2016-07-13:
Kris,
@MAWoodmain  has  just tried your code on an arduino with the same sensor, and the output looks much better, so it looks like errors we have made. We had better get debugging!
Graham

gjwo, 2016-11-02:
Kris,
After multiple bug fixes I am now getting this output (without fusion), the first value is a timestamp

> AccReadingCount 100
>  A:  t: +17646.7012 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17646.9023 x: +0.466 y: -3.811 z: -4.994
>  A:  t: +17647.0020 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17647.1016 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17647.2031 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17647.3008 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17647.4023 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17647.5020 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17647.6016 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17647.7012 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17647.8027 x: +0.467 y: -0.001 z: -4.994
>  A:  t: +17647.9004 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17648.0020 x: +0.467 y: -3.808 z: -4.994
>  A:  t: +17648.1016 x: +0.467 y: -0.001 z: -4.994
>  A:  t: +17648.2012 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17648.3008 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17648.4023 x: +0.466 y: -3.811 z: -4.994
>  A:  t: +17648.5039 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17648.6016 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17648.7031 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17648.8027 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17648.9023 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17649.0020 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17649.1035 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17649.2012 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17649.3027 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17649.4023 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17649.5020 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17649.6016 x: +0.466 y: -3.812 z: -4.994
>  A:  t: +17649.7031 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17649.8008 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17649.9023 x: +0.467 y: -3.813 z: -4.994
>  A:  t: +17650.0020 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17650.1016 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17650.2012 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17650.3027 x: +0.467 y: -0.001 z: -4.994
>  A:  t: +17650.4023 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17650.5020 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17650.6016 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17650.7031 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17650.8008 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17650.9023 x: +0.467 y: -3.808 z: -4.994
>  A:  t: +17651.0020 x: +0.467 y: -3.813 z: -4.994
>  A:  t: +17651.1016 x: +0.467 y: -3.813 z: -4.994
>  A:  t: +17651.2012 x: +0.467 y: -3.808 z: -4.994
>  A:  t: +17651.3027 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17651.4004 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17651.5020 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17651.6035 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17651.7012 x: +0.466 y: -3.808 z: -4.994
>  A:  t: +17651.8027 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17651.9023 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17652.0020 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17652.1016 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17652.2031 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17652.3027 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17652.4023 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17652.5020 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17652.6035 x: +0.466 y: -3.811 z: -4.994
>  A:  t: +17652.7012 x: +0.467 y: -3.809 z: -4.994
>  A:  t: +17652.8027 x: +0.467 y: -3.812 z: -4.994
>  A:  t: +17652.9023 x: +0.467 y: -3.811 z: -4.994
>  A:  t: +17653.0020 x: +0.467 y: -3.810 z: -4.994
>  A:  t: +17653.1016 x: +0.466 y: -0.001 z: -4.994
>  A:  t: +17653.2031 x: +0.466 y: -3.810 z: -4.994
>  A:  t: +17653.3008 x: +0.467 y: -3.807 z: -4.994
[... 213 lines cut ...]
>  M:  t: +17654.7051 x: +116.706 y: -0.281 z: -559.020
>  M:  t: +17654.8047 x: +122.418 y: -3.357 z: -552.869
>  M:  t: +17654.9043 x: +112.422 y: -4.895 z: -551.331
>  M:  t: +17655.0039 x: +110.994 y: +2.795 z: -543.642
>  M:  t: +17655.1055 x: +116.706 y: -9.508 z: -537.490
>  M:  t: +17655.2031 x: +122.418 y: -9.508 z: -546.717
>  M:  t: +17655.3047 x: +110.994 y: -6.432 z: -540.566
>  M:  t: +17655.4043 x: +109.566 y: -1.819 z: -542.104
>  M:  t: +17655.5059 x: +113.850 y: +5.870 z: -552.869
>  M:  t: +17655.6035 x: +113.850 y: -12.584 z: -543.642
>  M:  t: +17655.7051 x: +113.850 y: -0.281 z: -543.642
>  M:  t: +17655.8047 x: +115.278 y: +4.332 z: -545.179
>  M:  t: +17655.9043 x: +122.418 y: -0.281 z: -549.793
>  M:  t: +17656.0039 x: +110.994 y: -0.281 z: -565.171
>  M:  t: +17656.1055 x: +122.418 y: -3.357 z: -552.869
>  M:  t: +17656.2031 x: +106.710 y: +1.257 z: -551.331
>  M:  t: +17656.3047 x: +122.418 y: -6.432 z: -549.793
>  M:  t: +17656.4043 x: +123.846 y: -1.819 z: -545.179
>  M:  t: +17656.5039 x: +122.418 y: -0.281 z: -555.944
>  M:  t: +17656.6035 x: +116.706 y: -0.281 z: -534.415
>  M:  t: +17656.7051 x: +118.134 y: +4.332 z: -542.104
>  M:  t: +17656.8027 x: +113.850 y: -6.432 z: -559.020
>  M:  t: +17656.9063 x: +116.706 y: -9.508 z: -543.642
> 
I don't know what a correct set of readings would look like for a sensor at rest, please could you say what you would expect to see. These reading are without moving during mag calibration.

kriswiner, 2016-11-02:
The accel doesn't seem to be correctly reporting. The gyro might be OK but
the values are quite large for being at rest.

As for the mag, the Earth's magnetic field is n average 500 milliGauss for
you should expect the magnitude to be about 0.5. You can look up the local
magnetic field in your area and compare what you read to it. In my case, I
expect 200 mG, 50 mG, and 400 mG.

Kris

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-257892291> ,
k4RTn1Gks5q6KakgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qqhOvJGEo_fyweTGsipYeu5NO5Ipks5
q6KakgaJpZM4JLRwF.gif>

gjwo, 2016-11-02:
Thanks Kris,
So gyro should tend to 0 with no rotation movement, accel should tend to accel due to gravity which will be split across x,y,z depending on its orientation. Mag will depend on where you are. Have I understood that correctly?

kriswiner, 2016-11-02:
Yes, gyro should read about 0, 0, and 0

Accel should read 0, 0, 1

And mag should read ~100 ~100, !400

Kris

Pi (#67)

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-257910729> ,
0EAkVt2ks5q6LQUgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qj02T_CKneRTPygpX9DNVQ5r879cks5
q6LQUgaJpZM4JLRwF.gif>

gjwo, 2016-11-02:
Here are the latest results with the device clamped upright, There is a lot of debugging stuff at the start, I am still trying to work out why the self test looks odd, although having moved the register update to take it out of self test to the end  of the Self test routine seems to have made things more sensible.
[RPITank log  clamped.txt](https://github.com/kriswiner/MPU-9250/files/567264/RPITank.log.clamped.txt)

gjwo, 2016-11-02:
[RPITank log  clamped.txt](https://github.com/kriswiner/MPU-9250/files/567287/RPITank.log.clamped.txt)

gjwo, 2016-11-02:
Kris,
Please can you see if I am doing anything wrong in the self test function, in particular with setting the registers at the different stages. The average values are over a run of 200 samples, (the individual samples don't change much if at all)
in the data below it is REG_NAME: Value before write -> value after write. The integer arrays are of [x,y,z] shown in decimal then hex.

> Start MPU9250Test main()
> Bus acquired
> selfTest
>           SMPLRT_DIV : 00000100 0x4 -> 00000000 0x0
>               CONFIG : 00000011 0x3 -> 00000010 0x2
>          GYRO_CONFIG : 00011000 0x18 -> 00000000 0x0
>         ACCEL_CONFIG : 00001000 0x8 -> 00000000 0x0
>        ACCEL_CONFIG2 : 00000011 0x3 -> 00000010 0x2
> aAvg average: [-1, -32768, -32768] [0xFFFF, 0x8000, 0x8000]
> gAvg average: [-1, -32768, -32768] [0xFFFF, 0x8000, 0x8000]
>         ACCEL_CONFIG : 00000000 0x0 -> 11100000 0xE0
>          GYRO_CONFIG : 00000000 0x0 -> 11100000 0xE0
> aSTAvg average: [-1, -32768, -32768] [0xFFFF, 0x8000, 0x8000]
> gSTAvg average: [17467, 16961, -92] [0xFFFF, 0x8000, 0x8000]
> Self test Accel bytes: [106, 106, 113] [0x6A, 0x6A, 0x71]
> factoryTrimAcc (float): [7448.1006, 7448.1006, 7985.3726]
> Accelerometer accuracy:(% away from factory values)
> x: 0.0%
> y: 0.0%
> z: 0.0%
> Self test Gyro bytes: [194, 201, 228]
> factoryTrimGyro (float): [7448.1006, 7448.1006, 7985.3726]
> Gyroscope accuracy:(% away from factory values)
> x: 97.704994%
> y: 259.43805%
> z: 130.3094%
>         ACCEL_CONFIG : 11100000 0xE0 -> 00000000 0x0
>          GYRO_CONFIG : 11100000 0xE0 -> 00000000 0x0
> End selfTest
>

gjwo, 2016-11-03:
After a lot of restructuring and fixing issues, I thought I would see what the results were without the bias calculations I get this from the Accelerometer when clamped upright.

>  A:  t: +14549.3760 x: -0.026 y: -0.060 z: +0.983
>  A:  t: +14549.4756 x: -0.026 y: -0.060 z: +0.981
>  A:  t: +14549.5752 x: -0.027 y: -0.061 z: +0.981
>  A:  t: +14549.6758 x: -0.029 y: -0.061 z: +0.982
>  A:  t: +14549.7754 x: -0.026 y: -0.061 z: +0.983
>  A:  t: +14549.8750 x: -0.026 y: -0.061 z: +0.979
>  A:  t: +14549.9756 x: -0.027 y: -0.059 z: -0.015
>  A:  t: +14550.0752 x: -0.027 y: -0.061 z: +0.983
>  A:  t: +14550.1748 x: -0.028 y: -0.060 z: +0.983
>  A:  t: +14550.2754 x: -0.028 y: -0.059 z: +0.981
>  A:  t: +14550.3750 x: -0.028 y: -0.062 z: +0.982

I reckon this looks pretty good apart from the odd glitch, would you say this was an acceptable output? or is the error too high?

kriswiner, 2016-11-03:
Seems about right, offset bias calibration should get the errors down to~10
mg or so, which is just the inherent jitter in the sensor.

Pi (#67)

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-258158866> ,
-YMqa6oks5q6fBrgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qn7WkdruF_tb0l5eNfLQQ1i4DU0Wks5
q6fBrgaJpZM4JLRwF.gif>

gjwo, 2016-11-10:
Kris,
thanks to your help I am making progress! Still working without calibration of Accelerometer  biases. Assuming that the base data is coming out right (big assumption), have now got some data through the Madgwick filter and generated Yaw, Pitch and Roll, not sure how to determine if this is right though. Any comments on the output pointing out blindingly obvious (to you) errors would be much appreciated.
This output is working with 100ms sensor averages. 
[Sensor 20161110v2 log.txt](https://github.com/kriswiner/MPU-9250/files/583195/Sensor.20161110v2.log.txt)
java code has been moved to here [https://github.com/gjwo/RPISensors](url) (for info only)

kriswiner, 2016-11-11:
Looks about right, but you need to calibrate to be sure. Check that when you align an axis with gravity you indeed get /- 1 g.

gjwo, 2016-11-12:
Kris,
with gyro calibration (stationary) and mag calibration (wave in figure of 8 horizontally ) but without acc bias written to registers (same issue as your code I think). When the sensor is stationary I seem to be getting a steady drift on mag x, any ideas as to why this might be?

> Mag Calibration: Wave device in a figure eight until done!
> Mag Calibration: Finished
> A [0059.495] x: +000.111 y: -000.032 z: +000.993 G x: -000.946 y: +000.610 z: +000.427 M x: -008.534 y: -001.228 z: +058.212 | Y,P&R: [0059.645] x: +002.079 y: +000.549 z: -006.863
> A [0059.652] x: +000.112 y: -000.032 z: +000.993 G x: -000.946 y: +000.610 z: +000.427 M x: -008.534 y: -001.228 z: +058.212 | Y,P&R: [0059.655] x: +002.220 y: -005.573 z: -003.828
> A [0059.664] x: +000.112 y: -000.031 z: +000.993 G x: -000.916 y: +000.590 z: +000.407 M x: -009.501 y: -001.272 z: +063.025 | Y,P&R: [0059.671] x: +002.785 y: +001.345 z: -004.775
> A [0059.764] x: +000.112 y: -000.032 z: +000.993 G x: -000.916 y: +000.565 z: +000.412 M x: -010.392 y: -001.420 z: +067.976 | Y,P&R: [0059.767] x: +002.963 y: -005.433 z: -003.717
> A [0059.864] x: +000.112 y: -000.032 z: +000.993 G x: -000.928 y: +000.549 z: +000.403 M x: -011.509 y: -001.501 z: +072.932 | Y,P&R: [0059.868] x: +003.699 y: +001.349 z: -005.348
> A [0059.964] x: +000.112 y: -000.032 z: +000.993 G x: -000.905 y: +000.539 z: +000.387 M x: -012.788 y: -001.118 z: +077.797 | Y,P&R: [0059.969] x: +004.329 y: -005.494 z: -005.230
> A [0060.064] x: +000.112 y: -000.032 z: +000.993 G x: -000.907 y: +000.532 z: +000.384 M x: -013.934 y: -000.945 z: +082.512 | Y,P&R: [0060.067] x: +004.948 y: +001.442 z: -004.778
> A [0060.164] x: +000.112 y: -000.032 z: +000.993 G x: -000.908 y: +000.534 z: +000.381 M x: -015.153 y: -000.585 z: +087.476 | Y,P&R: [0060.167] x: +005.565 y: -005.384 z: -005.306
> A [0060.264] x: +000.112 y: -000.032 z: +000.993 G x: -000.916 y: +000.543 z: +000.380 M x: -016.505 y: -000.598 z: +092.195 | Y,P&R: [0060.267] x: +006.356 y: +001.569 z: -005.593
> A [0060.364] x: +000.112 y: -000.032 z: +000.993 G x: -000.916 y: +000.543 z: +000.372 M x: -017.098 y: -000.208 z: +096.985 | Y,P&R: [0060.367] x: +006.355 y: -004.811 z: -003.113
>

kriswiner, 2016-11-12:
Looks like you are summing a small amount of something each call; I suspect
your math or logic in subtracting the mag offsets is wrong somewhere.

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-260116981> ,
LO20quYks5q9aSKgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qiWX6o3RIpIiBJ3E90uTvLIHiiVqks5
q9aSKgaJpZM4JLRwF.gif>

gjwo, 2016-11-12:
Kris,
you're a star, I was adding in the offsets in the update routines (all of them). Having removed that everything looks very stable (readings are a couple of seconds into normal operation). The yaw pitch and roll values now look flaky given the stable readings elsewhere, any ideas about that? Gravity is +ve rather than -ve would that cause any issues?

> A [0287.303] x: +000.109 y: +000.010 z: +000.994 G x: -150.200 y: -125.040 z: -136.000 M x: -048.560 y: -008.160 z: +358.880 | Y,P&R: [0287.305] x: +076.715 y: -036.171 z: +069.416
> A [0287.403] x: +000.109 y: +000.010 z: +000.994 G x: -150.120 y: -124.960 z: -135.960 M x: -048.240 y: -008.400 z: +358.480 | Y,P&R: [0287.405] x: +060.011 y: -028.036 z: +059.776
> A [0287.503] x: +000.109 y: +000.010 z: +000.994 G x: -150.080 y: -124.840 z: -135.840 M x: -048.160 y: -008.480 z: +358.560 | Y,P&R: [0287.505] x: +043.779 y: -022.591 z: +047.915
> A [0287.603] x: +000.109 y: +000.009 z: +000.994 G x: -150.160 y: -124.680 z: -135.680 M x: -048.080 y: -008.640 z: +358.720 | Y,P&R: [0287.605] x: +027.680 y: -020.231 z: +034.625
> A [0287.703] x: +000.109 y: +000.009 z: +000.994 G x: -150.120 y: -124.560 z: -135.480 M x: -048.000 y: -008.800 z: +358.880 | Y,P&R: [0287.705] x: +011.986 y: -021.094 z: +021.044
> A [0287.803] x: +000.109 y: +000.009 z: +000.994 G x: -150.120 y: -124.640 z: -135.480 M x: -047.680 y: -008.960 z: +358.880 | Y,P&R: [0287.805] x: -003.794 y: -024.775 z: +008.815
> A [0287.903] x: +000.109 y: +000.009 z: +000.994 G x: -150.080 y: -124.640 z: -135.480 M x: -047.840 y: -008.800 z: +358.480 | Y,P&R: [0287.905] x: -019.120 y: -030.480 z: -001.236
> A [0288.003] x: +000.109 y: +000.009 z: +000.994 G x: -150.120 y: -124.720 z: -135.400 M x: -047.640 y: -008.920 z: +358.760 | Y,P&R: [0288.005] x: -034.525 y: -038.257 z: -008.485
> A [0288.103] x: +000.109 y: +000.009 z: +000.994 G x: -150.120 y: -124.640 z: -135.320 M x: -047.000 y: -008.840 z: +358.760 | Y,P&R: [0288.105] x: -051.042 y: -047.560 z: -012.947
> A [0288.203] x: +000.109 y: +000.009 z: +000.994 G x: -150.080 y: -124.640 z: -135.320 M x: -047.280 y: -008.800 z: +358.880 | Y,P&R: [0288.205] x: -070.445 y: -058.106 z: -013.444
> A [0288.303] x: +000.109 y: +000.009 z: +000.994 G x: -150.040 y: -124.640 z: -135.240 M x: -047.320 y: -008.680 z: +358.920 | Y,P&R: [0288.305] x: -098.162 y: -068.611 z: -005.275
>

kriswiner, 2016-11-12:
You have to 1) properly calibrate the sensors and 2) pass the sensor data to
the Madwick/Mahoney filter with the axes properly aligned, 3) run with
appropriate sample rates, bandwidths, and sensor fusion rates.

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-260137907> ,
33eaUFsks5q9f5igaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qiyCo5BfOhyrUJZSvyK_eWFUb8hUks5
q9f5igaJpZM4JLRwF.gif>

gjwo, 2016-11-12:
Kris, as per you advice
1) **properly calibrate the sensors** - I think I am doing that
2) **pass the sensor data to the Madwick/Mahoney filter with the axes properly aligned,** 
I think I did what you instructed here, see code below
```
            	Instruments.setMagnetometer( mpu9250.getLatestGaussianData());
                Instruments.setAccelerometer(mpu9250.getLatestAcceleration());
                Instruments.setGyroscope(mpu9250.getLatestRotationalAcceleration());
                
            	// Examples of calling the filters, READ BEFORE USING!!		!!!
            	// sensors x (y)-axis of the accelerometer is aligned with the y (x)-axis of the magnetometer;
            	// the magnetometer z-axis (+ down) is opposite to z-axis (+ up) of accelerometer and gyro!
            	// We have to make some allowance for this orientation mismatch in feeding the output to the quaternion filter.
            	// For the MPU-9250, we have chosen a magnetic rotation that keeps the sensor forward along the x-axis just like
            	// in the LSM9DS0 sensor. This rotation can be modified to allow any convenient orientation convention.
            	// This is ok by aircraft orientation standards!  
            	// Pass gyro rate as rad/s
            	//  MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f,  my,  mx, mz);

                ajustedGyr = new TimestampedData3f(Instruments.getGyroscope());
                ajustedGyr.setX(Instruments.getGyroscope().getX()*(float)Math.PI/180.0f); //Pass gyro rate as rad/s
                ajustedGyr.setY(Instruments.getGyroscope().getY()*(float)Math.PI/180.0f);
                ajustedGyr.setZ(Instruments.getGyroscope().getZ()*(float)Math.PI/180.0f);
                ajustedMag = new TimestampedData3f(Instruments.getMagnetometer()); //set timestamp and Z
                ajustedMag.setX(Instruments.getMagnetometer().getY()); //swap X and Y, Z stays the same
                ajustedMag.setY(Instruments.getMagnetometer().getX());

                SensorFusion.MadgwickQuaternionUpdate(Instruments.getAccelerometer(),ajustedGyr,ajustedMag,((float)(DELTA_T)/((float)TimestampedData3f.NANOS_PER_SEC)));

```
3) **run with appropriate sample rates, bandwidths, and sensor fusion rates.**
The hardware register settings are the same as yours with respect to rates, bandwidths, SMPLRT_DIV etc, not sure what you mean by the sensor fusion rates - please can you expand on this?

gjwo, 2016-11-12:
Just spotted something, with the Magnetometer there is no support for writing back biases or scaling to the hardware so it is only a correction made by the main program, different to the acc and Gyro.  I may have got this wrong, fixing the code now!

kriswiner, 2016-11-12:
Yes, you need to run with a processor capable of iterating the sensor fusion
algorithm at least four or five times for each new data sample you get. If
you run the sample rates at 200 Hz, the processor must be able to run the
fusion algorithm at 1 kHz or more for a stable result.

The Arduino Pro Mini at 8 MHz cannot, the Raspberry Pi should be able to.

Pi (#67)

 
 

 
 
 

 

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-260139082> ,
qgA-zuhks5q9gM_gaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qtGzWqEfg1jv3imiB8wRO3gn4DWkks5
q9gM_gaJpZM4JLRwF.gif>

gjwo, 2016-11-15:
Hi Kris,
sorry for the delay in replying, I have been fixing a lot of bugs, also calculating deltat and refreshing the display at 2Hz rather than 200Hz. 

Is what you mean by this running MadgwickQuaternionUpdate with the same readings and the deltat of the interval between calls  to Madgwick at five times the frequency if the changes in readings?

I also need to go slug hunting, as the performance isn't good enough to do it yet. The frequency is slowly approaching 153 Hz asymptotically over a 5 minute run, and I need to understand why it isn't 200Hz from the start!

gjwo, 2016-11-15:
Hi Kris,
working on the assumption above, it looks like its working (not quite 1KHz on RPi 3), I am running MadgwickQuaternionUpdate  about every millisecond with new data if it's available, if not with the current data. The results are below, please let me know how this looks to you. This was a no movement run.
[Sensor 20161115v6 log 30sec run.txt](https://github.com/kriswiner/MPU-9250/files/592336/Sensor.20161115v6.log.30sec.run.txt)

I need to think about output now.

kriswiner, 2016-11-15:
The frequency of the Madgiwck calls depends on the processor you are using.
You can but really shouldn't set the fusion rate manually at first. Using
the Teensy (MK20 72 MHz Cortex M4) I can easily get 1 kHz fusion rates, much
more using the STM32L4. a kHz is just about right. 153 Hz won't give you a
stable solution.

Pi (#67)

<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-260609935> ,
XAYg-3rks5q-Y6MgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qs1heKye_fNdHCqhUwOBtb3qx9nlks5
q-Y6MgaJpZM4JLRwF.gif>

gjwo, 2016-11-15:
Kris, see latest comment & log running at approx 1K

gjwo, 2016-11-22:
I think my work here is done!
[image]

gjwo, 2016-11-22:
Thanks for your help Kris.

closed 2016-11-22 by gjwo

kriswiner, 2016-11-22:
Glad you got it working!

Pi (#67)

image
<https://cloud.githubusercontent.com/assets/12109990/20541840/bf54b9de-b0f6-
11e6-93bc-a875c8e23b13.png> 

-
<https://github.com/kriswiner/MPU-9250/issues/67#issuecomment-262364324> ,
beePg1iks5rA1gdgaJpZM4JLRwF> .
<https://github.com/notifications/beacon/AGY1qvlz-zJCOWKCcoMa6MJAHE0Q4ir6ks5
rA1gdgaJpZM4JLRwF.gif>