kriswiner/MPU9250 #68: Setting up two MPU9250s on the same I2C bus using AD0

pchez, 2016-07-16:
Hi Kris,

In page 44 of the register map (https://www.invensense.com/wp-content/uploads/2015/02/MPU-9250-Register-Map.pdf), it says that the default device ID is 0x71. I am indeed able to read 0x71 from the WHO_AM_I register when I pull AD0 low. However, when i pull AD0 high, the WHO_AM_I register reads back 0x6.

How do I set the I2C address to be 0x68 when AD0=0 and 0x69 when AD0=1, as specified in the datasheet? 

Thanks a bunch.

kriswiner, 2016-07-16:
Which breakout are you using?

You should be able to apply 3V3 to ADO and use the I2C address 0x69 to read
the WHO_AM_I register (0x71 always).

Set ADO = 1 and do an I2C scan; what I2C devices can you find on the bus?

using AD0 (#68)

(https://www.invensense.com/wp-content/uploads/2015/02/MPU-9250-Register-Map
.pdf), it says that the default device ID is 0x71. I am indeed able to read

6Vcod37ks5qWEs5gaJpZM4JN6Ci> .
<https://github.com/notifications/beacon/AGY1qsf4t1oLY_m6S4wnW0iXxWur_BOAks5
qWEs5gaJpZM4JN6Ci.gif>

pchez, 2016-07-17:
I am using the DIYMALL breakout.

Thanks for the help! I was able to pull ADO high and read the I2C addresses (0x69 and 0x68) to obtain 0x71.

pchez, 2016-07-22:
Hi Kris,
I'm trying to set up 2 sensors so that my nRF51dk on mbed can read from both sensors over the same I2c bus. Right now, I'm able to get both sensors online and can read from the sensor with I2C address 0x68, but cannot read from the sensor with I2C address 0x69.

This is what I'm using to read from both sensors. 

`// If intPin goes high, all data registers have new data
        MPU9250_ADDRESS = SENSOR1_ADDRESS;
        if(sensor1.readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01)   // On interrupt, check if data ready interrupt
		{
            sensor1.readAccelData(accelCount1);  // Read the x/y/z adc values   
            // Now we'll calculate the accleration value into actual g's
            ax1 = (float)accelCount1[0]*aRes - accelBias1[0];  // get actual g value, this depends on scale being set
            ay1 = (float)accelCount1[1]*aRes - accelBias1[1];   
            az1 = (float)accelCount1[2]*aRes - accelBias1[2];  

            sensor1.readGyroData(gyroCount1);  // Read the x/y/z adc values
            // Calculate the gyro value into actual degrees per second
            gx1 = (float)gyroCount1[0]*gRes - gyroBias1[0];  // get actual gyro value, this depends on scale being set
            gy1 = (float)gyroCount1[1]*gRes - gyroBias1[1];  
            gz1 = (float)gyroCount1[2]*gRes - gyroBias1[2];   

            sensor1.readMagData(magCount);  // Read the x/y/z adc values   
            // Calculate the magnetometer values in milliGauss
            // Include factory calibration per data sheet and user environmental corrections
            mx1 = (float)magCount[0]*mRes*magCalibration[0] - magbias[0];  // get actual magnetometer value, this depends on scale being set
            my1 = (float)magCount[1]*mRes*magCalibration[1] - magbias[1];  
            mz1 = (float)magCount[2]*mRes*magCalibration[2] - magbias[2];   
        }
        
        MPU9250_ADDRESS = SENSOR2_ADDRESS;
        if(sensor2.readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01) // On interrupt, check if data ready interrupt
        {  
            sensor2.readAccelData(accelCount2);  // Read the x/y/z adc values   
            // Now we'll calculate the accleration value into actual g's
            ax2 = (float)accelCount2[0]*aRes - accelBias2[0];  // get actual g value, this depends on scale being set
            ay2 = (float)accelCount2[1]*aRes - accelBias2[1];   
            az2 = (float)accelCount2[2]*aRes - accelBias2[2];  

            sensor2.readGyroData(gyroCount2);  // Read the x/y/z adc values
            // Calculate the gyro value into actual degrees per second
            gx2 = (float)gyroCount2[0]*gRes - gyroBias2[0];  // get actual gyro value, this depends on scale being set
            gy2 = (float)gyroCount2[1]*gRes - gyroBias2[1];  
            gz2 = (float)gyroCount2[2]*gRes - gyroBias2[2];   

            sensor2.readMagData(magCount);  // Read the x/y/z adc values   
            // Calculate the magnetometer values in milliGauss
            // Include factory calibration per data sheet and user environmental corrections
            mx2 = (float)magCount[0]*mRes*magCalibration[0] - magbias[0];  // get actual magnetometer value, this depends on scale being set
            my2 = (float)magCount[1]*mRes*magCalibration[1] - magbias[1];  
            mz2 = (float)magCount[2]*mRes*magCalibration[2] - magbias[2];   
        
        }`

kriswiner, 2016-07-22:
If you scan the bus do you get ACKs from both sensors, one at 0x68 and 0ne
at 0x69?

How are you selecting 0x69?

<https://github.com/kriswiner/MPU-9250/issues/68#issuecomment-234663048> ,
ImgFCFiks5qYTfygaJpZM4JN6Ci> .
<https://github.com/notifications/beacon/AGY1qmbnI2imuXiub6Un5KfJ8Zxjt0Quks5
qYTfygaJpZM4JN6Ci.gif>

pchez, 2016-07-23:
Yes, I get ACKs from both sensors. To select it during the bus scan, I call readBytes with 0x69<<1 as an argument. Then to read from the second sensor, I physically modified MPU9250_ADDRESS to be 0x69<<1 before calling readAccelData and readGyroData. Likewise, to read from the first sensor, I physically modified MPU9250_ADDRESS to be 0x68<<1.

kriswiner, 2016-07-23:
Then I don't quite understand why this doesn't work for you.

bus using AD0 (#68)

<https://github.com/kriswiner/MPU-9250/issues/68#issuecomment-234687075> ,
uEnmoCBks5qYWAHgaJpZM4JN6Ci> .
<https://github.com/notifications/beacon/AGY1qussR9OeNlYr1Fbh3DigLnfzIbTSks5
qYWAHgaJpZM4JN6Ci.gif>

cojo5412, 2016-11-01:
Hey Kris,

I've got this working by using the AD0 pulled high for 0x69 and pulled low on a second unit for 0x68. I'm confused with there only being one magnetometer address? Does this mean that each time I try to access the magnetometer on one IMU, I'm actually pulling mag data from both IMU's somehow? If so, is there a way to disable the magnetometer on only one of the 9250's at startup so when I call the 0x0C magnetometer address I only access one mag and not both? 

Thanks so much!

kriswiner, 2016-11-01:
The way Invensense expects you to do this is to use the FIFO to read the
data, and then you can toggle between both MPU9250 devices.

Another way to do this is to use two different I2C busses, or to use a
TCA9548A I2C multiplexer, or to use GPIO pins to supply power to one so you
can shut it down when you don't want to read it, but yeah, basically if you
just connect two MPU9250 on the same I2C bus without the above you can't
address the mags separately.

bus using AD0 (#68)

<https://github.com/kriswiner/MPU-9250/issues/68#issuecomment-257704641> ,
t5LKXJPks5q56-tgaJpZM4JN6Ci> .
<https://github.com/notifications/beacon/AGY1qoBVuoqwyocb6Ao9sZrwBq_eLo7uks5
q56-tgaJpZM4JN6Ci.gif>

MikeFair, 2016-11-06:
Another way to do this is to use the I2C Master abilities of the MPU to poll its own MAG and feed the data into the 24 bytes of EXT_SENS_DATA registers.  This is separate from the FIFO buffer;  though the data being fed into EXT_SENS_DATA can also be fed into the FIFO Buffer.

Assuming SLAVE0 does the work, the mag data is retrievable from the MPU starting at register 0x49.

I've been playing around with making this work so that I can have 1 MPU polling its own MAG (6 bytes) plus 6-axis data on another MPU (12 bytes - 14 if you snag temp).  Leaving 6 bytes left to capture something else (like the mag data from the second/downstream MPU).

Documentation and information seems to be scarce, so here's what I've learned:

```
(0x38 INT_PIN_CFG:0)  BYPASS_EN = 0   // Disconnects the bypass on EDA/ECL if it was enabled
(0x6A USER_CTL:5)  I2C_MST_EN = 1     // Enables the I2C Master within the MPU

0x24 I2C_MST_CTRL = 0x5D
(I2C_MST_CTRL:7)   MULT_MST_EN = 0     // Unless there are multiple masters hooked up to that bus;
(I2C_MST_CTRL:6)   WAIT_FOR_ES = 1     // Causes the data ready interrupt to wait for the external data
(I2C_MST_CTRL:5)   SLV3_FIFO_EN = 0    // Unless you are using FIFO for Slave3 requested data 
(I2C_MST_CTRL:4)   I2C_P_NSR = 1       // When transitioning slaves; issue a full stop, then a start
(I2C_MST_CTRL:0-3) I2C_MST_CLK = 0xD   // 400kHz -- see table 

0x25 I2C_SLV0_ADDR = 0x80 (b10000000) | 0x0C // High-bit 1 = read; the rest is the device's address
0x26 I2C_SLV0_REG = 0x00    // register to r/w;  0x0C / 0x00 = AK8963 DeviceID (WIA) register
0x27 I2C_SLV0_CTRL = 0x81
(I2C_SLV0_CTRL:7)  I2C_SLV0_EN  = 1     // Enable slave 0
(I2C_SLV0_CTRL:6) I2C_SLV0_SW  = 0      // Don't swap bytes around while reading the data in
(I2C_SLV0_CTRL:5) I2C_SLV0_REG_DIS = 0  // Every txn, first write the register address; then execute
(I2C_SLV0_CTRL:4) I2C_SLV0_GRP = 0      // If swapping, are the words aligned on even or odd addresses
(I2C_SLV0_CTRL:0-3) I2C_SLV0_LEN = 0x1  // How many bytes to read (up to 15); 24 in EXT_SENS_DATA

0x63 I2C_SLV0_DO = 0x00   // If we were to write to the slave 0 device; this is the byte that is sent
```

> External sensor data is written to these registers at the Sample Rate as defined in (Register 25 / SMPLRT_DIV / 0x19).  This access rate can be reduced by using the Slave Delay Enable registers (Register 103 / I2C_MST_DELAY_CTRL / 0x67).
> 
> This reduced access rate is determined by I2C_MST_DLY (Register 52 / I2C_SLV4_CTL:0-4 / 0x34 ). 
> 
 

Once set up, you should see 0x48 (the AK8963's DeviceID) appearing in the EXT_SENS_DATA register 0x49.  

The data in EXT_SENS_DATA doesn't get cleared, so if the device stops responding, the old values get stuck there and you can't really tell (and you can't write to it to clear it yourself).  To ensure the transaction actually happened, read the "NACK" status of each slave along with the actual EXT_SENS_DATA registers (0x36 I2C_MST_STATUS:0-4); 0 = success, 1 = failure.