kriswiner/MPU9250 #70: Yaw miscalculated

Zarbizaure, 2016-08-18:
Hey,
I recently bought a cheap [GY-91 sensor](http://chipskey.en.alibaba.com/product/60439909912-218419143/GY_91_MPU9250_BMP280_10DOF_9_axis_Accelerometer_Gyroscope_Compass_Sensor_Module.html#!), which does wiork well as an acceleromer, gyro and baromter despite its price. However, the yaw value returned by the [MPU9250BasicAHRS.ino](https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino) program is really weird.

First of all, pitch and roll value are correct, and they are not influenced by any change in the other direction. The accuracy is not exeptional (~5degree) but that's enough for what I want to do.

The yaw is apparently (kind of) calibrated with the earth magnetic shield; there is a 180 degree difference between the higer and the lowest value. Between these two angles, the mesured value goes up and down again. It looks like a sinusoïd: see attached file.
[Yaw.xlsx](https://github.com/kriswiner/MPU-9250/files/424446/Yaw.xlsx)

Do you have any idea? I though that the conversion from quaternion to euler angle for yaw was wrong, but I have no idea how to get it correct....
Thanks!

kriswiner, 2016-08-18:
How did you calibrate the sensors?

<http://chipskey.en.alibaba.com/product/60439909912-218419143/GY_91_MPU9250_
BMP280_10DOF_9_axis_Accelerometer_Gyroscope_Compass_Sensor_Module.html#!> ,

like a sinuso�d: see attached file.

zgmTee8ks5qhBr4gaJpZM4JnQyU> .
<https://github.com/notifications/beacon/AGY1qkYv5397GuUqjtJVYQVFGYEDFvB3ks5
qhBr4gaJpZM4JnQyU.gif>

Zarbizaure, 2016-08-18:
I didn't. I have no idea how to calibrate them, but the gyro/accelerometer results are fine and precise. I doubt calibrating would change anything to this sort of sinusoïdal result, but I could still try. What do you do to calibrate it? I guess I'll have to change these lines, but with which values? 
`    magbias[0] = +470.;  // User environmental x-axis correction in milliGauss, should be automatically calculated
    magbias[1] = +120.;  // User environmental x-axis correction in milliGauss
    magbias[2] = +125.;  // User environmental x-axis correction in milliGauss`

kriswiner, 2016-08-18:
Read this:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer
-Calibration

and then copy the calibration functions from here:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

If you don't calibrate your mag you will never get proper heading.

It makes no sense to use the hard coded calibration value from my particular
magnetometer, no wonder you are getting a sinusoid.

this sort of sinuso�dal result, but I could still try. What do you do to

<https://github.com/kriswiner/MPU-9250/issues/70#issuecomment-240872114> ,
sPMaIeLks5qhNeYgaJpZM4JnQyU> .
<https://github.com/notifications/beacon/AGY1qrXACdeacSP6WePPuOukaf-lxAhSks5
qhNeYgaJpZM4JnQyU.gif>

Zarbizaure, 2016-08-18:
Thanks!
It seems better now but I still have some issue.
First of all, only mag scales value are returned; magBias retunr 0.0 for the three value (which I guess is wrong).
Plus, the compass is fine for a moment then start to "drift": the value increase really fast and starts to be totally different.
I guess my magBias matrix is wrong somewhere, I will check...

Zarbizaure, 2016-08-19:
Finaly working! Hurray! With precision close to the degree :-) I used your latest file (with the MS5637 display) and changed some stuff...calibration with a "8" was really wrong, but I make it spin on the z-axis during calibration and it worked perfectly!
Thanks for your amazing work! I think I'll make a library from your edited file (in order to simplify my code), if it does work well may I submit a PR? (it will be suited for serial only though).

kriswiner, 2016-08-19:
If calibration with figure eight motion was wrong it just means you didn't
sample the response surface enough. You can increase the time or increase
the range of your motion. Figure eight is the better motion, otherwise you
will have yaw error when you introduce pitch or roll.

If you want to modify the code I will consider a pull request but it is
unlikely I will accept if there is a large rewrite; I like the code spread
out in one file so I can see everything.

<https://github.com/kriswiner/MPU-9250/issues/70#issuecomment-240894983> ,
Y5FfoTuks5qhPX3gaJpZM4JnQyU> .
<https://github.com/notifications/beacon/AGY1quSq_3AWLxOTnn-8glYpeK6TOuBxks5
qhPX3gaJpZM4JnQyU.gif>

Zarbizaure, 2016-08-19:
I will try with larger movements but my better results are still induced with a mix from "8" withouth z-rotating and z-rotation (because doing both movement at once, considering that my sensor is housed inside an RC plane and that I need to move the whole plane to calibrate, it is not easy).

No, that was just to create new library files for those who want to use it as a library (it easier as part as a larger project). I won't change others files.

ajaymills, 2017-08-11:
Hi, I also have an issue with Yaw and I suspect it is due to miscalibrating. Based on the fact that my plot regions have sort of "two elongated lobes" it's evident I'm not moving it in an optimal way. What should the figure 8 motion look like? It seems like an 8 motion will mainly cover two arbitrary axis or do you have to do multiple 8's in different directions? Do you fully invert the device while going through the motion? 

[image]

kriswiner, 2017-08-11:
You need to sample every point on the response surface many times in your motion.