kriswiner/MPU9250 #72: About MPU9250SelfTest and AK8963_ASAX

ijunglee, 2016-08-26:
Hello, thank you for replying me the e-mail.
I have read your mail about the mag calibration and came to here to see the code. 
I finally understood the procedure of mag calibration!
But I have two more questions as follows:
1. What are the AK8963_ASAX, AK8963_ASAY and AK8963_ASAZ? Should I have to read them and calibrate the mag before I start to do the hard-iron and soft-iron calibration?
2. How about MPU9250SelfTest? Is this function also for calibration and I must to do this for acc and gyro calibration?

Thank you very much!

kriswiner, 2016-08-26:
The self test just tests the proper function of the sensors, it has nothing
to do with calibration. The ASAX values are factory scale values that should
be applied to the three axes to properly scale them.

9MYphRqks5qjjeGgaJpZM4Jto_6> .
<https://github.com/notifications/beacon/AGY1qqzYU7dup7V9Eg7Weqbclc1mIlYzks5
qjjeGgaJpZM4Jto_6.gif>

ijunglee, 2016-08-26:
Thanks for your reply. I have finished the work of mag calibration.
But how do I check the data of magnetometer if it is correct?

kriswiner, 2016-08-26:
Look up your local magnetic field and see if you get it about right. Plot Mx
vs My and Mx vs Mz and see if you get a circular response surface centered
on the origin. Look at Mz when the board is facing up, them compare to Mz
when the board is facing down. Is Mzup = -Mzdowm? Lots of ways to check.

(#72)

<https://github.com/kriswiner/MPU-9250/issues/72#issuecomment-242620611> ,
Qpfe7Avks5qjl8ogaJpZM4Jto_6> .
<https://github.com/notifications/beacon/AGY1qliCmve023mVKNnT1w_a4DJiOK4jks5
qjl8ogaJpZM4Jto_6.gif>

ijunglee, 2016-08-26:
Sorry for so many questions......
This the result when I use the Servomotor rotate the MPU9250 90 degrees back and forward on the horizontal plane and stay in each attitude for 2 seconds  . (I only tried to validate the heading/yaw)
[image]

This is the raw data of the mag data on each axis. The sampling rate is about 50Hz.
[image]

I have calibrate the mag data and use MadgwickAHRS algorithm to do the sensor fusion. I read your code and also change the axis when I pass the raw data to the function like this:
MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);

However, I am not sure if the yaw is correct. Because I always got a peak when I change to one attitude to another. Do you have any idea about how to solve this problem?

Thanks in advance again!

kriswiner, 2016-08-26:
Try:

MadgwickQuaternionUpdate(-ax, ay, az, gxPI/180.0f, -gyPI/180.0f,
-gz*PI/180.0f, my, -mx, -mz);

(#72)

yaw
<https://cloud.githubusercontent.com/assets/12967717/18007175/87cb432c-6bd5-
11e6-9b3f-11ed8cd8cae8.PNG> 

raw
<https://cloud.githubusercontent.com/assets/12967717/18006890/623ea064-6bd4-
11e6-83c5-f7b484c914aa.PNG> 

<https://github.com/kriswiner/MPU-9250/issues/72#issuecomment-242738235> ,
rcUA1foks5qjuzRgaJpZM4Jto_6> .
<https://github.com/notifications/beacon/AGY1qnjMt9TJ7P2a17ij0qho_oa_UCAdks5
qjuzRgaJpZM4Jto_6.gif>