kriswiner/MPU9250 #74: MPU-9250 Self-test Factory Trim Calculation Question

ponny213, 2016-09-15:
I am trying to understand the function "MPU9250SelfTest()" in the code. And i found some inconsistency between the code and the comments, for example:
...
writeByte(MPU9250_ADDRESS, GYRO_CONFIG, 1<<FS);  // Set full scale range for the gyro to 250 dps
...
writeByte(MPU9250_ADDRESS, ACCEL_CONFIG, 1<<FS); // Set full scale range for the accelerometer to 2 g
...
This two lines of code set the full scale range for the gyro to 1000 dps, full scale range for the accelerometer to 8 g. So I am confused how to set full scale range when performing self test?

Then I look up the MPU9250 register map (https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU-9250-Register-Map.pdf). It doesn't give any information about self test. However, the MPU9150 register map (https://strawberry-linux.com/pub/RM-MPU-9150A-00.pdf) has instructions (page 9 and 10) about how to perform self test and calculate factory trim value. 

Based on the instructions on page 9 and 10 of the MPU9150 register map, when performing self-test, the gyroscope full scale range need to be set up as 250 dps, and the accelerometer full scale range need to be set up as 8 g. And the factory trim calculation formula provided by the instructions also seems inconsistent with the code:

factoryTrim[0] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[0] - 1.0) )); // FT[Xa] factory trim calculation
   factoryTrim[1] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[1] - 1.0) )); // FT[Ya] factory trim calculation
   factoryTrim[2] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[2] - 1.0) )); // FT[Za] factory trim calculation
   factoryTrim[3] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[3] - 1.0) )); // FT[Xg] factory trim calculation
   factoryTrim[4] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[4] - 1.0) )); // FT[Yg] factory trim calculation
   factoryTrim[5] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[5] - 1.0) )); // FT[Zg] factory trim calculation

So my question is:
It the MPU9250 self-test same as MPU9150 self-test? If not, where can I find and documentation about MPU9250 self-test and how to understand the code?

kriswiner, 2016-09-15:
" It the MPU9250 self-test same as MPU9150 self-test?"

No, ask Invensense.

Question (#74)

(https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU-9250-Register-Map
.pdf). It doesn't give any information about self test. However, the MPU9150

4JuKxCJks5qqJy7gaJpZM4J9aIR> .
<https://github.com/notifications/beacon/AGY1qplZ_46zr7Aw7fHaZ1v2AGo2jm0nks5
qqJy7gaJpZM4J9aIR.gif>

ponny213, 2016-09-15:
Thanks for your reply Kris. May i ask why did you use this formula to calculate factory trim?
factoryTrim[0] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[0] - 1.0) )); // FT[Xa] factory trim calculation
factoryTrim[1] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[1] - 1.0) )); // FT[Ya] factory trim calculation
factoryTrim[2] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[2] - 1.0) )); // FT[Za] factory trim calculation
factoryTrim[3] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[3] - 1.0) )); // FT[Xg] factory trim calculation
factoryTrim[4] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[4] - 1.0) )); // FT[Yg] factory trim calculation
factoryTrim[5] = (float)(2620/1<<FS)(pow( 1.01 , ((float)selfTest[5] - 1.0) )); // FT[Zg] factory trim calculation

kriswiner, 2016-09-15:
PM me at onehorse@earthlink.net and I will send you the self test
documentation.

Calculation Question (#74)

<https://github.com/kriswiner/MPU-9250/issues/74#issuecomment-247207884> ,
BO78Merks5qqJ__gaJpZM4J9aIR> .
<https://github.com/notifications/beacon/AGY1qslX5Kz1PrKLav-RLId-RTDmh1joks5
qqJ__gaJpZM4J9aIR.gif>

phangialuan, 2017-02-20:
Sir, can you give me the self-test documentation.

kriswiner, 2017-02-20:
Send me an e-mail at tleracorp@gmail.com.

alan-castelli, 2020-04-03:
Hi!!
I wanted to consult too, How did you get those 2600 and did not use what was documented on page 10 and 11 of one of the official documents (title: MPU-6000 / MPU-6050 Register Map and Descriptions)?
Can I ask you for the account you made and what you observed to reach that value?
Thank you!!

kriswiner, 2020-04-03:
Hi Alan,

I am not sure what you mean. Which 2600 are referring to?

Kris

alan-castelli, 2020-04-03:
yes, exatly!!

alan-castelli, 2020-04-03:
the self-test at the end uses a fatory

// Retrieve factory self-test value from self-test code reads
    factoryTrim [0] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [0] - 1.0))); // FT [Xa] factory trim calculation
    factoryTrim [1] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [1] - 1.0))); // FT [Ya] factory trim calculation
    factoryTrim [2] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [2] - 1.0))); // FT [Za] factory trim calculation
    factoryTrim [3] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [3] - 1.0))); // FT [Xg] factory trim calculation
    factoryTrim [4] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [4] - 1.0))); // FT [Yg] factory trim calculation
    factoryTrim [5] = (float) (2620/1 << FS) * (pow (1.01, ((float) selfTest [5] - 1.0))); // FT [Zg] factory trim calculation

why?? How do you calculate it?

alan-castelli, 2020-04-03:
use 2620, sorry
I would like to ask you how you can get it out

kriswiner, 2020-04-03:
Which sketch is this?

alan-castelli, 2020-04-03:
https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino
Line: 978

kriswiner, 2020-04-03:
https://github.com/kriswiner/MPU9250/blob/master/Documents/AN-MPU-9250A-03%20MPU-9250%20Accel%20Gyro%20and%20Compass%20Self-Test%20Implementation%20v1%200_062813.pdf

page 6

alan-castelli, 2020-04-04:
Thank you so much!! that's just it !!