kriswiner/MPU9250 #79: Roll and Pitch changes affect Yaw

marcirsch, 2016-10-04:
Dear Kris!

I am using your mbed library, with an stm32f042, also tried with stm32f446. I have calibrated the accelerometer, gyroscope static, and the magnetometer with the calibration function (and 8 figure). Using bandwidth 41Hz for gyro, 44.8 for accelerometer. Output rate is 200 Hz. Not using fifo.
As you can see on the picture attached, the correlation between yaw and other angles is pretty strong.
I am stuck here, any help is appreciated! 

Thank you for your work on these libraries!
[image]

kriswiner, 2016-10-05:
Are you passing the Madgwick filter the data in the proper order? I mean
accel z = -mag z, etc.

9250
<https://cloud.githubusercontent.com/assets/22622263/19085968/94d4a83c-8a6c-
11e6-9822-0203a2f5db50.JPG> 

-
asIyEdWks5qwpXngaJpZM4KN-5A> .
<https://github.com/notifications/beacon/AGY1qq7700LFFR6-hRcXFAOShuSrLFFiks5
qwpXngaJpZM4KN-5A.gif>

kriswiner, 2016-10-11:
See this discussion:

https://forum.pjrc.com/threads/25965-9-and-10-DoF-LSM9DS0-shield-for-Teensy-3-1?p=117389&viewfull=1#post117389

marcirsch, 2016-10-11:
Hi Kris!

Thank you for your advice! I didn't pay attention to the different orientation of the magnetometer. After changing the arguments, it works! 
One more question: What do you recommend for fast motion capturing? Now Output rate is 1kHz, sending rate is 100Hz, Beta is unchanged. It misses some fast spins. Changing gyro sensitivity would do the thing?

Marci

marcirsch, 2016-10-11:
So I've just found out setting low pass filters' bandwidth a bit higher, around 200, setting gyro sensitivity to 1000 DPS, and Accelerometer to 8gs gives a decent result.

Thank you for your help!
Marci

closed 2016-10-11 by marcirsch

Sanjanawinchester, 2018-07-03:
hey @marcirsch  can you please elaborate exactly how your problem was solved? Because I am facing the same problem and there doesn't seem to be any way out.

Sanjanawinchester, 2018-07-05:
@marcirsch ...I tried that but only sometimes the problem is solved. Only sometimes the pitch doesn't affect the roll and yaw, but many times it does.

Sanjanawinchester, 2018-07-05:
I included the code for calibration of the magnetometer, but how to move the IMU while calibrating. I read to move in figure 8 but exactly how?

kriswiner, 2018-07-05:
Sample the entire 3D response surface

Sanjanawinchester, 2018-07-06:
I did not quite get you. Can you please explain in detail @kriswiner ?

kriswiner, 2018-07-06:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration